mu-editor/mu #2193: ERROR: Could not enter raw REPL.

MakerMexa, 2022-04-24:
### What were you trying to do?

im trying use my esp8266 and see files on the device

### What steps did you take to trigger the issue?

i try reinstalling lasted release of Mu and disable smartscreen filter while Mu it installs but dont works  
[image]

### What did you expect to happen?

can program my board with Mu

### What actually happened?

this is the Mu´s logs:

2022-04-24 09:21:38,881 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.1
2022-04-24 09:21:38,886 - root:270(run) INFO: uname_result(system='Windows', node='3D1', release='10', version='10.0.19044', machine='AMD64', processor='Intel64 Family 6 Model 37 Stepping 5, GenuineIntel')
2022-04-24 09:21:39,044 - root:271(run) INFO: Platform: Windows-10-10.0.19044-SP0
2022-04-24 09:21:39,045 - root:272(run) INFO: Python path: ['C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-57.4.0-py3.8.egg', 'C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-21.1.3-py3.8.egg', 'C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\3DESIGN\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin']
2022-04-24 09:21:39,045 - root:273(run) INFO: Language code: es_MX
2022-04-24 09:21:40,176 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2022-04-24 09:21:41,272 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2022-04-24 09:21:41,278 - mu.virtual_environment:685(ensure_path) INFO: Virtual Environment found at: C:\Users\3DESIGN\AppData\Local\python\mu\mu_venv-38-20220422-171743
2022-04-24 09:21:41,300 - mu.virtual_environment:698(ensure_interpreter) INFO: Interpreter found at: C:\Users\3DESIGN\AppData\Local\python\mu\mu_venv-38-20220422-171743\scripts\python.exe
2022-04-24 09:21:41,304 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\3DESIGN\\AppData\\Local\\python\\mu\\mu_venv-38-20220422-171743\\scripts\\python.exe', '-c', 'import sys; print("%s%s" % sys.version_info[:2])') with kwargs {'shell': True}
2022-04-24 09:21:41,835 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 38
2022-04-24 09:21:41,835 - mu.virtual_environment:730(ensure_interpreter_version) INFO: Both interpreters at version 38
2022-04-24 09:21:41,844 - mu.virtual_environment:759(ensure_pip) INFO: Pip found at: C:\Users\3DESIGN\AppData\Local\python\mu\mu_venv-38-20220422-171743\scripts\pip.exe
2022-04-24 09:21:41,849 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: pgzero
2022-04-24 09:21:41,849 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\3DESIGN\\AppData\\Local\\python\\mu\\mu_venv-38-20220422-171743\\scripts\\python.exe', '-c', 'import pgzero') with kwargs {'shell': True}
2022-04-24 09:21:42,151 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-04-24 09:21:42,151 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: flask
2022-04-24 09:21:42,151 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\3DESIGN\\AppData\\Local\\python\\mu\\mu_venv-38-20220422-171743\\scripts\\python.exe', '-c', 'import flask') with kwargs {'shell': True}
2022-04-24 09:21:43,841 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-04-24 09:21:43,842 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipykernel
2022-04-24 09:21:43,842 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\3DESIGN\\AppData\\Local\\python\\mu\\mu_venv-38-20220422-171743\\scripts\\python.exe', '-c', 'import ipykernel') with kwargs {'shell': True}
2022-04-24 09:21:46,806 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-04-24 09:21:46,806 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipython_genutils
2022-04-24 09:21:46,806 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\3DESIGN\\AppData\\Local\\python\\mu\\mu_venv-38-20220422-171743\\scripts\\python.exe', '-c', 'import ipython_genutils') with kwargs {'shell': True}
2022-04-24 09:21:47,155 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-04-24 09:21:47,155 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: esptool
2022-04-24 09:21:47,156 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\3DESIGN\\AppData\\Local\\python\\mu\\mu_venv-38-20220422-171743\\scripts\\python.exe', '-c', 'import esptool') with kwargs {'shell': True}
2022-04-24 09:21:47,799 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-04-24 09:21:47,800 - mu.virtual_environment:628(ensure_and_create) INFO: Valid virtual environment found at C:\Users\3DESIGN\AppData\Local\python\mu\mu_venv-38-20220422-171743
2022-04-24 09:21:47,804 - mu.settings:169(save) DEBUG: Saving to C:\Users\3DESIGN\AppData\Local\python\mu\venv.json
2022-04-24 09:21:47,896 - mu.logic:768(__init__) INFO: Setting up editor.
2022-04-24 09:21:47,897 - mu.logic:789(__init__) INFO: Log directory: C:\Users\3DESIGN\AppData\Local\python\mu\Logs
2022-04-24 09:21:47,897 - mu.logic:790(__init__) INFO: Data directory: C:\Users\3DESIGN\AppData\Local\python\mu
2022-04-24 09:21:47,898 - mu.logic:804(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2022-04-24 09:21:47,899 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\3DESIGN\mu_code from settings file
2022-04-24 09:21:48,418 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\3DESIGN\mu_code from settings file
2022-04-24 09:21:48,418 - mu.logic:1549(change_mode) INFO: Workspace directory: C:\Users\3DESIGN\mu_code
2022-04-24 09:21:48,418 - mu.logic:878(restore_session) DEBUG: <SessionSettings from C:\Users\3DESIGN\AppData\Local\python\mu\session.json>
2022-04-24 09:21:48,438 - mu.logic:900(restore_session) INFO: Loaded files.
2022-04-24 09:21:48,438 - mu.logic:906(restore_session) INFO: User defined environment variables: {}
2022-04-24 09:21:48,438 - mu.logic:911(restore_session) INFO: Minify scripts on micro:bit? False
2022-04-24 09:21:48,587 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\3DESIGN\mu_code from settings file
2022-04-24 09:21:48,587 - mu.logic:1549(change_mode) INFO: Workspace directory: C:\Users\3DESIGN\mu_code
2022-04-24 09:21:50,055 - mu.logic:962(restore_session) INFO: Starting with blank file.
2022-04-24 09:22:07,019 - mu.logic:747(check_usb) INFO: esp device connected on port: COM3(VID: 0x1A86, PID: 0x7523, manufacturer: 'wch.cn')
2022-04-24 09:22:11,798 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\3DESIGN\mu_code from settings file
2022-04-24 09:22:11,837 - mu.modes.esp:214(toggle_files) INFO: Toggle filesystem on.
2022-04-24 09:22:13,897 - mu.modes.base:686(ls) ERROR: Could not enter raw REPL.
Traceback (most recent call last):
[... 152 lines cut ...]
Pygments          2.11.2
pyserial          3.5
python-dateutil   2.8.2
pywin32           303
pyzmq             22.3.0
reedsolo          1.5.4
setuptools        60.9.3
six               1.16.0
stack-data        0.2.0
tomli             2.0.1
tornado           6.1
traitlets         5.1.1
typing_extensions 4.1.1
wcwidth           0.2.5
Werkzeug          2.0.3
wheel             0.37.1
2022-04-24 10:01:02,518 - mu.virtual_environment:931(installed_packages) INFO: []

### Operating System Version

windows 10

### Mu Version

1.1.1

vogrosty, 2022-07-25:
Hello!
I have same problem. 

Board: ESP12F (ESP8266), was working, but now has same issue
[issue_log.txt](https://github.com/mu-editor/mu/files/9180032/issue_log.txt)
.

vogrosty, 2022-07-25:
The issue disappears if:
1. Activate REPL mode, by clicking "REPL" button
2. Print some commands in repl, for ex. "1+2"
3. Deactivate REPL mode
4. Activate Files mode.

And then all works.

vogrosty, 2022-07-25:
Seems it happens when you make some mistake in modules.

To fix it I reflash firmware with ESP development tools.