mu-editor/mu #2197: Crash Report

bicyu, 2022-04-29:
**REMEMBER TO ATTACH YOUR LOG FILE**

Hello, re the copy of log, sorry but I have done this more than a dozen times and I can not find that file. HOME and BIC are no problems, but .cache does not exist, neither does Mu nor log. 

Running search shows nothing at the below address either.

P**_lease attach your log file to the bug report. Your log file can be found here: /home/bic/.cache/mu/log/mu.log_**

Maybe you can guide me to where exactly the thing is hiding please. I am using Zorin 15.3OS. I have Python 3.6 pre installed and am working from: [automate the boring stuff](url) and[ How to Create Python Virtual Environment on Ubuntu & Debian](url)

 As you can see I have everything (eventually) working but when I press the REPL button the wretched thing crashes with the error message: 

TypeError: __init__() missing 1 required positional argument: 'context'

There is scant info on search (Bing) re that message, what little I can find seems not to relate to Ubuntu or it is written with too much technical language and knowledge assumed.

I am not as you guess a programmer by trade, but if you can give me some clue or hint as to how I can trace and fix this I would have a fighting chance.
Cheers
p

Version: 1.1.1
Locale: en_GB
Platform: Linux 5.4.0-109-generic #123~18.04.1-Ubuntu SMP Fri Apr 8 09:48:52 UTC 2022 x86_64

```
Traceback (most recent call last):
  File "/home/bic/.local/lib/python3.6/site-packages/mu/modes/python3.py", line 371, in on_kernel_start
    self.view.add_jupyter_repl(kernel_manager, kernel_client)
  File "/home/bic/.local/lib/python3.6/site-packages/mu/interface/main.py", line 670, in add_jupyter_repl
    kernel_client.start_channels()
  File "/home/bic/.local/lib/python3.6/site-packages/qtconsole/kernel_mixins.py", line 47, in start_channels
    super(QtKernelClientMixin, self).start_channels(*args, **kw)
  File "/home/bic/.local/lib/python3.6/site-packages/jupyter_client/threaded.py", line 248, in start_channels
    super().start_channels(shell, iopub, stdin, hb, control)
  File "/home/bic/.local/lib/python3.6/site-packages/jupyter_client/client.py", line 295, in start_channels
    self.hb_channel.start()
  File "/home/bic/.local/lib/python3.6/site-packages/jupyter_client/client.py", line 364, in hb_channel
    self._hb_channel = self.hb_channel_class(self.context, self.session, url)
  File "/home/bic/.local/lib/python3.6/site-packages/qtconsole/util.py", line 65, in __new__
    QClass.__init__(inst)
TypeError: __init__() missing 1 required positional argument: 'context'

```****

bicyu, 2022-04-29:
Maybe I should clarify, the only Mu is found at
/home/bic/Python/venv/lib/python3.6/site-packages but it does not have any log file.
Cheers
P

bicyu, 2022-04-29:
Sorry to do this piece meal, I am still working on it myself.
This is the log file from Mu BEFORE I press REPL- if that is any help?

2022-04-29 12:47:20,232 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.1
2022-04-29 12:47:20,233 - root:270(run) INFO: uname_result(system='Linux', node='bic-System-Product-Name', release='5.4.0-109-generic', version='#123~18.04.1-Ubuntu SMP Fri Apr 8 09:48:52 UTC 2022', machine='x86_64', processor='x86_64')
2022-04-29 12:47:20,235 - root:271(run) INFO: Platform: Linux-5.4.0-109-generic-x86_64-with-Zorin-15-bionic
2022-04-29 12:47:20,235 - root:272(run) INFO: Python path: ['/home/bic/Python/venv/bin', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/bic/Python/venv/lib/python3.6/site-packages', '/home/bic/Python/venv/lib/python3.6/site-packages/IPython/extensions']
2022-04-29 12:47:20,235 - root:273(run) INFO: Language code: en_GB
2022-04-29 12:47:20,236 - mu.settings:220(load) WARNING: No settings file found at /home/bic/.local/share/mu/settings.json; skipping
2022-04-29 12:47:20,515 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2022-04-29 12:47:20,516 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2022-04-29 12:47:20,517 - mu.virtual_environment:685(ensure_path) INFO: Virtual Environment found at: /home/bic/.local/share/mu/mu_venv-36-20220428-165544
2022-04-29 12:47:20,518 - mu.virtual_environment:698(ensure_interpreter) INFO: Interpreter found at: /home/bic/.local/share/mu/mu_venv-36-20220428-165544/bin/python
2022-04-29 12:47:20,518 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/home/bic/.local/share/mu/mu_venv-36-20220428-165544/bin/python', '-c', 'import sys; print("%s%s" % sys.version_info[:2])') with kwargs {'shell': False}
2022-04-29 12:47:20,603 - mu.virtual_environment:444(run_subprocess) DEBUG: Process returned 0; output: 36
2022-04-29 12:47:20,603 - mu.virtual_environment:730(ensure_interpreter_version) INFO: Both interpreters at version 36
2022-04-29 12:47:20,603 - mu.virtual_environment:759(ensure_pip) INFO: Pip found at: /home/bic/.local/share/mu/mu_venv-36-20220428-165544/bin/pip
2022-04-29 12:47:20,604 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: pgzero
2022-04-29 12:47:20,604 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/home/bic/.local/share/mu/mu_venv-36-20220428-165544/bin/python', '-c', 'import pgzero') with kwargs {'shell': False}
2022-04-29 12:47:20,658 - mu.virtual_environment:444(run_subprocess) DEBUG: Process returned 0; output: 
2022-04-29 12:47:20,659 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: flask
2022-04-29 12:47:20,659 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/home/bic/.local/share/mu/mu_venv-36-20220428-165544/bin/python', '-c', 'import flask') with kwargs {'shell': False}
2022-04-29 12:47:21,604 - mu.virtual_environment:444(run_subprocess) DEBUG: Process returned 0; output: 
2022-04-29 12:47:21,604 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipykernel
2022-04-29 12:47:21,605 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/home/bic/.local/share/mu/mu_venv-36-20220428-165544/bin/python', '-c', 'import ipykernel') with kwargs {'shell': False}
2022-04-29 12:47:23,102 - mu.virtual_environment:444(run_subprocess) DEBUG: Process returned 0; output: 
2022-04-29 12:47:23,102 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipython_genutils
2022-04-29 12:47:23,102 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/home/bic/.local/share/mu/mu_venv-36-20220428-165544/bin/python', '-c', 'import ipython_genutils') with kwargs {'shell': False}
2022-04-29 12:47:23,153 - mu.virtual_environment:444(run_subprocess) DEBUG: Process returned 0; output: 
2022-04-29 12:47:23,154 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: esptool
2022-04-29 12:47:23,154 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/home/bic/.local/share/mu/mu_venv-36-20220428-165544/bin/python', '-c', 'import esptool') with kwargs {'shell': False}
2022-04-29 12:47:23,298 - mu.virtual_environment:444(run_subprocess) DEBUG: Process returned 0; output: 
2022-04-29 12:47:23,299 - mu.virtual_environment:628(ensure_and_create) INFO: Valid virtual environment found at /home/bic/.local/share/mu/mu_venv-36-20220428-165544
2022-04-29 12:47:23,299 - mu.settings:169(save) DEBUG: Saving to /home/bic/.local/share/mu/venv.json
2022-04-29 12:47:23,321 - mu.logic:768(__init__) INFO: Setting up editor.
2022-04-29 12:47:23,322 - mu.logic:789(__init__) INFO: Log directory: /home/bic/.cache/mu/log
2022-04-29 12:47:23,323 - mu.logic:790(__init__) INFO: Data directory: /home/bic/.local/share/mu
2022-04-29 12:47:23,323 - mu.logic:804(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2022-04-29 12:47:23,323 - mu.modes.base:63(get_default_workspace) INFO: Using workspace /home/bic/mu_code from settings file
2022-04-29 12:47:23,393 - mu.modes.base:63(get_default_workspace) INFO: Using workspace /home/bic/mu_code from settings file
2022-04-29 12:47:23,393 - mu.logic:1549(change_mode) INFO: Workspace directory: /home/bic/mu_code
2022-04-29 12:47:23,393 - mu.logic:878(restore_session) DEBUG: <SessionSettings from /home/bic/.local/share/mu/session.json>
[... cut ...]
ptyprocess         0.7.0
pycparser          2.21
pygame             2.1.2
Pygments           2.12.0
pyserial           3.5
python-dateutil    2.8.2
pyzmq              22.3.0
reedsolo           1.5.4
setuptools         59.6.0
six                1.16.0
tornado            6.1
traitlets          4.3.3
typing_extensions  4.1.1
wcwidth            0.2.5
Werkzeug           2.0.3
wheel              0.37.1
zipp               3.6.0
2022-04-29 14:56:43,697 - mu.virtual_environment:931(installed_packages) INFO: []

carlosperate, 2022-05-17:
Thanks for the report @bicyu, the log is very useful as well.

I think the problem is related to this one: https://github.com/mu-editor/mu/issues/1831

Could you install Mu in a virtual environment and see if the problem persist there? It's not recommended to install Mu with pip in the system Python installation, as it can create conflicts with existing installed packages.

I personally quite like using [pipx](https://pypa.github.io/pipx/), as that takes care of creating a virtual environment and adding Mu "to path" so that it can be executed by running `mu-editor` in a terminal.

bicyu, 2022-05-19:
Hello Carlos, thanks for your reply.
I should have noted in my first message, I am not tech savvy - most of what yu wrote does not mean much to me.

However, i did see that error #1831 seems to talk about C drive, and as being a Linux user I can't relate that.
You ask about setting up a virtual environment.
Do you mean a separate VE for MU, as in separate from the VE I set up for Python (see the link in my first post)

> It's not recommended to install Mu with pip in the system Python installation, as it can create conflicts with existing installed packages.

Uhhm how can I check to see that is what I am doing?
If so, is it reversible?

> I personally quite like using [pipx](https://pypa.github.io/pipx/), as that tak

I will check this and see what it means, how to do it etc.

Thanks for your time
Cheers
P

bicyu, 2022-05-19:
Hello Carlos, thanks for your reply.
I should have noted in my first message, I am not tech savvy - most of what you wrote does not mean much to me.

However, i did see that error #1831 seems to talk about C drive, and as being a Linux user I can't relate that.
You ask about setting up a virtual environment.
Do you mean a separate VE for MU, as in separate from the VE I set up for Python (see the link in my first post)

> It's not recommended to install Mu with pip in the system Python installation, as it can create conflicts with existing installed packages.

Uhhm how can I check to see that is what I am doing?
If so, is it reversible?

> I personally quite like using [pipx](https://pypa.github.io/pipx/), as that tak

I will check this and see what it means, how to do it etc.

Thanks for your time
Cheers
P

carlosperate, 2022-05-19:
Hi @bicyu

I didn't realised you had already set up a separate virtualenv, thanks for clarifying. Do you use the same virtual environment to install all your Python pacakges?
I would recommed to create a new one just for Mu. Hopefully that should resolve the problem you are having running the Mu Editor.

Then, once Mu is working, if there are any pacakges you might need to run your projects, you can install them inside Mu following these instructions: https://codewith.mu/en/tutorials/1.1/pypi

bicyu, 2022-05-27:
Hello Carlos, sorry for the delay, life etc.
Python is in its own VENV folder in my HOME -Mu code is in a separate folder - also in HOME.
I guessed it was a virtual environment but..?

Anyway, I was searching for "**_how to instal Mu in a virtual environment in linux"_** not much joy but I [did find this](https://codewith.mu/en/howto/1.1/install_linux#:~:text=Step%201%20-%20Download%20the%20AppImage%20Download%20the,permissions%20group%20%28usually%20the%20dialout%20or%20uucp%20groups%29.).

Before I go ahead and waste my time, is it a viable solution do you think?
Cheers
P