mu-editor/mu #2198: Ubuntu 22.04

SashGraf, 2022-04-29:
### What were you trying to do?

start mu-editor in terminal on Ubuntu 22.04

### What steps did you take to trigger the issue?

sudo apt-get install mu-editor

mu-editor 

### What did you expect to happen?

mu-editor should start

### What actually happened?

nothing

### Operating System Version

Ubuntu 22.04

### Mu Version

1.0.3

### Editor Log

Starting Mu 1.0.3
2022-04-29 22:39:19,424 - root:113(run) INFO: uname_result(system='Linux', node='kristina-All-Series', release='5.15.0-27-generic', version='#28-Ubuntu SMP Thu Apr 14 04:55:28 UTC 2022', machine='x86_64')
2022-04-29 22:39:19,424 - root:114(run) INFO: Python path: ['/usr/share/mu-editor', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/usr/local/lib/python3.10/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3/dist-packages/IPython/extensions']
2022-04-29 22:39:19,424 - root:115(run) INFO: Language code: de_DE
2022-04-29 22:39:19,517 - mu.logic:543(__init__) INFO: Setting up editor.
2022-04-29 22:39:19,517 - mu.logic:562(__init__) INFO: Settings path: /home/pavel/.local/share/mu/settings.json
2022-04-29 22:39:19,517 - mu.logic:563(__init__) INFO: Session path: /home/pavel/.local/share/mu/session.json
2022-04-29 22:39:19,517 - mu.logic:564(__init__) INFO: Log directory: /home/pavel/.cache/mu/log
2022-04-29 22:39:19,517 - mu.logic:565(__init__) INFO: Data directory: /home/pavel/.local/share/mu
2022-04-29 22:39:19,529 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2022-04-29 22:39:19,577 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/usr/bin/mu-editor", line 33, in <module>
    sys.exit(load_entry_point('mu-editor==1.0.3', 'console_scripts', 'mu-editor')())
  File "/usr/share/mu-editor/mu/app.py", line 148, in run
    editor_window.setup(editor.debug_toggle_breakpoint, editor.theme)
  File "/usr/share/mu-editor/mu/interface/main.py", line 825, in setup
    self.autosize_window()
  File "/usr/share/mu-editor/mu/interface/main.py", line 774, in autosize_window
    self.move((screen.width() - size.width()) / 2,
TypeError: arguments did not match any overloaded call:
  move(self, QPoint): argument 1 has unexpected type 'float'
  move(self, int, int): argument 1 has unexpected type 'float'

mariusmotea, 2022-05-05:
Same problem 

```
 File "/usr/share/mu-editor/mu/interface/main.py", line 774, in autosize_window
    self.move((screen.width() - size.width()) / 2,
TypeError: arguments did not match any overloaded call:
  move(self, QPoint): argument 1 has unexpected type 'float'
  move(self, int, int): argument 1 has unexpected type 'float'
```

carlosperate, 2022-05-17:
Thanks for the report @SashGraf and @MarPiSora and thanks for the workaround @mariusmotea!

I've added a little bit of detail to the problem in https://github.com/mu-editor/mu/issues/2228, hopefully the workarounds listed there might be able to help you.

As we are tracking the problem over there I'll close this issue, but please feel free to reopen if that doesn't work or you have any other questions.

closed 2022-05-17 by carlosperate

Infoisntfreenotheresaheftyfuckingfee, 2022-07-10:
> As workaround i edit the main.py file and made these changes:
> 
> ```
>     def autosize_window(self):
>         """
>         Makes the editor 80% of the width*height of the screen and centres it.
>         """
>         screen = QDesktopWidget().screenGeometry()
>         w = int(screen.width() * 0.8)
>         h = int(screen.height() * 0.8)
>         self.resize(w, h)
>         size = self.geometry()
>         #self.move((screen.width() - size.width()) / 2,
>         #          (screen.height() - size.height()) / 2)
>         self.move(500,500)
> ```

If you are a dumb noob like me who got here via google then just know this is the dumbed down version that worked for me if you are in Ubuntu 22.04:

open a terminal. 
$ cd /
$ cd usr/share/mu-editor/mu/interface
$ sudo xdg-open main.py

This opens the file in a text editor.  Then go down to around line 774 and delete the two lines in the above that have the "#" in front (I left the hashtags in and got a weird error) then add "self.move(500,500) and save (if you didn't open with sudo command then you wont be able to save).

Thank you soo much mariusmotea!!!  I was trying to get this installed as I have been using mu editor.  I am in way over my head but your directions got me to succefully open the program.

My instructions above are probably more busy then they have to be but it worked for me.

Thank you again to this community.

mrtnocean, 2022-08-23:
I’m new to Linux, I got the file open but how do I edit it?