mu-editor/mu #2201: trying to install mu in linux using pip fails

dieseltravis, 2022-05-01:
### What were you trying to do?

install mu via pip3

### What steps did you take to trigger the issue?

```
python3 -m venv mu-venv
source mu-venv/bin/activate
# via 🐍 v3.10.4 (mu-venv)
pip3 install mu-editor
```

### What did you expect to happen?

mu-editor installed

### What actually happened?

Output:
```
Collecting mu-editor
  Using cached mu_editor-1.0.3-py3-none-any.whl (2.0 MB)
Collecting qtconsole==4.3.1
  Using cached qtconsole-4.3.1-py2.py3-none-any.whl (108 kB)
Collecting gpiozero>=1.4.1
  Using cached gpiozero-1.6.2-py2.py3-none-any.whl (148 kB)
Collecting matplotlib==2.2.2
  Using cached matplotlib-2.2.2.tar.gz (37.3 MB)
  Preparing metadata (setup.py) ... done
Collecting Pillow>=5.2.0
  Using cached Pillow-9.1.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (4.3 MB)
Collecting PyQtChart==5.14.0
  Using cached PyQtChart-5.14.0-5.14.2-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.2 MB)
Collecting pyserial==3.4
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting pgzero==1.2
  Using cached pgzero-1.2-py3-none-any.whl (69 kB)
Collecting guizero>=1.1.0
  Using cached guizero-1.3.0-py3-none-any.whl (44 kB)
Collecting qscintilla==2.11.4
  Using cached QScintilla-2.11.4-1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.9 MB)
Collecting semver>=2.8.0
  Using cached semver-2.13.0-py2.py3-none-any.whl (12 kB)
Collecting pigpio>=1.40.post1
  Using cached pigpio-1.78-py2.py3-none-any.whl (39 kB)
Collecting requests>=2.19.1
  Using cached requests-2.27.1-py2.py3-none-any.whl (63 kB)
Collecting pyqt5==5.14.1
  Using cached PyQt5-5.14.1-5.14.1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (63.5 MB)
Collecting nudatus>=0.0.3
  Using cached nudatus-0.0.5-py3-none-any.whl (5.3 kB)
Collecting pyflakes==2.0.0
  Using cached pyflakes-2.0.0-py2.py3-none-any.whl (53 kB)
Collecting pycodestyle==2.4.0
  Using cached pycodestyle-2.4.0-py2.py3-none-any.whl (62 kB)
Collecting appdirs>=1.4.3
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting numpy>=1.7.1
  Using cached numpy-1.22.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (16.8 MB)
Collecting cycler>=0.10
  Using cached cycler-0.11.0-py3-none-any.whl (6.4 kB)
Collecting pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1
  Using cached pyparsing-3.0.8-py3-none-any.whl (98 kB)
Collecting python-dateutil>=2.1
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Collecting pytz
  Using cached pytz-2022.1-py2.py3-none-any.whl (503 kB)
Collecting six>=1.10
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting kiwisolver>=1.0.1
  Using cached kiwisolver-1.4.2-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (1.6 MB)
Collecting pygame<2.0,>=1.9.2
  Using cached pygame-1.9.6.tar.gz (3.2 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [18 lines of output]
      
      
      WARNING, No "Setup" File Exists, Running "buildconfig/config.py"
      Using UNIX configuration...
      
      Missing dependencies
      
      Hunting dependencies...
      SDL     : found 1.2.15
      FONT    : not found
      IMAGE   : not found
      MIXER   : not found
      PNG     : found
      JPEG    : not found
      SCRAP   : found
      PORTMIDI: not found
      PORTTIME: not found
      FREETYPE: found 24.2.18
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

### Operating System Version

Pop!_OS 22.04 LTS

### Mu Version

mu_editor-1.0.3-py3-none-any.whl

### Other Info

Framework Laptop

Pop!_OS 22.04 LTS jammy x86-64
Linux pop-os 5.16.19-76051619-generic #202204081339~1649696161~22.04~091f44b SMP PREEMPT Mon Apr 11 17 x86_64 x86_64 x86_64 GNU/Linux

Python 3.10.4
pip 22.0.4

dieseltravis, 2022-05-05:
I'm now getting a different error after manually installing the dependencies for pygame listed here:
https://www.pygame.org/wiki/CompileUbuntu?parent=#Python%203.x

```
> pip3 install mu-editor
Collecting mu-editor
  Using cached mu_editor-1.0.3-py3-none-any.whl (2.0 MB)
Collecting semver>=2.8.0
  Using cached semver-2.13.0-py2.py3-none-any.whl (12 kB)
Collecting guizero>=1.1.0
  Using cached guizero-1.3.0-py3-none-any.whl (44 kB)
Collecting requests>=2.19.1
  Using cached requests-2.27.1-py2.py3-none-any.whl (63 kB)
Requirement already satisfied: pyserial==3.4 in ./.asdf/installs/python/3.10.4/lib/python3.10/site-packages (from mu-editor) (3.4)
Collecting pyqt5==5.14.1
  Using cached PyQt5-5.14.1-5.14.1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (63.5 MB)
Collecting pgzero==1.2
  Using cached pgzero-1.2-py3-none-any.whl (69 kB)
Collecting PyQtChart==5.14.0
  Using cached PyQtChart-5.14.0-5.14.2-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.2 MB)
Collecting Pillow>=5.2.0
  Using cached Pillow-9.1.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (4.3 MB)
Collecting nudatus>=0.0.3
  Using cached nudatus-0.0.5-py3-none-any.whl (5.3 kB)
Collecting pigpio>=1.40.post1
  Using cached pigpio-1.78-py2.py3-none-any.whl (39 kB)
Collecting pyflakes==2.0.0
  Using cached pyflakes-2.0.0-py2.py3-none-any.whl (53 kB)
Collecting gpiozero>=1.4.1
  Using cached gpiozero-1.6.2-py2.py3-none-any.whl (148 kB)
Collecting appdirs>=1.4.3
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting matplotlib==2.2.2
  Using cached matplotlib-2.2.2.tar.gz (37.3 MB)
  Preparing metadata (setup.py) ... done
Collecting qtconsole==4.3.1
  Using cached qtconsole-4.3.1-py2.py3-none-any.whl (108 kB)
Collecting qscintilla==2.11.4
  Using cached QScintilla-2.11.4-1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.9 MB)
Collecting pycodestyle==2.4.0
  Using cached pycodestyle-2.4.0-py2.py3-none-any.whl (62 kB)
Collecting numpy>=1.7.1
  Using cached numpy-1.22.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (16.8 MB)
Collecting cycler>=0.10
  Using cached cycler-0.11.0-py3-none-any.whl (6.4 kB)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in ./.asdf/installs/python/3.10.4/lib/python3.10/site-packages (from matplotlib==2.2.2->mu-editor) (3.0.8)
Collecting python-dateutil>=2.1
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Requirement already satisfied: pytz in ./.asdf/installs/python/3.10.4/lib/python3.10/site-packages (from matplotlib==2.2.2->mu-editor) (2022.1)
Collecting six>=1.10
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting kiwisolver>=1.0.1
  Using cached kiwisolver-1.4.2-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (1.6 MB)
Collecting pygame<2.0,>=1.9.2
  Using cached pygame-1.9.6.tar.gz (3.2 MB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: PyQt5-sip<13,>=12.7 in ./.asdf/installs/python/3.10.4/lib/python3.10/site-packages (from pyqt5==5.14.1->mu-editor) (12.10.1)
Collecting jupyter-client>=4.1
  Using cached jupyter_client-7.3.0-py3-none-any.whl (130 kB)
Collecting pygments
  Using cached Pygments-2.12.0-py3-none-any.whl (1.1 MB)
Collecting ipython-genutils
  Using cached ipython_genutils-0.2.0-py2.py3-none-any.whl (26 kB)
Collecting traitlets
  Using cached traitlets-5.1.1-py3-none-any.whl (102 kB)
Collecting jupyter-core
  Using cached jupyter_core-4.10.0-py3-none-any.whl (87 kB)
Collecting ipykernel>=4.1
  Using cached ipykernel-6.13.0-py3-none-any.whl (131 kB)
Collecting colorzero
[... 580 lines cut ...]
      To contribute to pygame development see:
          https://www.pygame.org/contribute.html
      ---
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  WARNING: No metadata found in ./.asdf/installs/python/3.10.4/lib/python3.10/site-packages
  Rolling back uninstall of pygame
  Moving to /home/travis/.asdf/installs/python/3.10.4/include/python3.10/pygame/
   from /home/travis/.asdf/installs/python/3.10.4/include/python3.10/~ygame
  Moving to /home/travis/.asdf/installs/python/3.10.4/lib/python3.10/site-packages/pygame-2.1.2.dist-info/
   from /home/travis/.asdf/installs/python/3.10.4/lib/python3.10/site-packages/~ygame-2.1.2.dist-info
  Moving to /home/travis/.asdf/installs/python/3.10.4/lib/python3.10/site-packages/pygame.libs/
   from /home/travis/.asdf/installs/python/3.10.4/lib/python3.10/site-packages/~ygame.libs
  Moving to /home/travis/.asdf/installs/python/3.10.4/lib/python3.10/site-packages/pygame/
   from /home/travis/.asdf/installs/python/3.10.4/lib/python3.10/site-packages/~ygame
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pygame

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

dieseltravis, 2022-05-06:
I thought maybe `asdf` was causing issues, so I uninstalled that but I still get errors related to `pygame`.

```
> pip3 install mu-editor
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: mu-editor in /usr/local/lib/python3.10/dist-packages (1.0.3)
Requirement already satisfied: pyqt5==5.14.1 in /usr/local/lib/python3.10/dist-packages (from mu-editor) (5.14.1)
Requirement already satisfied: pyserial==3.4 in ./.local/lib/python3.10/site-packages (from mu-editor) (3.4)
Requirement already satisfied: matplotlib==2.2.2 in ./.local/lib/python3.10/site-packages (from mu-editor) (2.2.2)
Requirement already satisfied: nudatus>=0.0.3 in ./.local/lib/python3.10/site-packages (from mu-editor) (0.0.5)
Requirement already satisfied: pyflakes==2.0.0 in /usr/local/lib/python3.10/dist-packages (from mu-editor) (2.0.0)
Requirement already satisfied: PyQtChart==5.14.0 in /usr/local/lib/python3.10/dist-packages (from mu-editor) (5.14.0)
Requirement already satisfied: gpiozero>=1.4.1 in /usr/local/lib/python3.10/dist-packages (from mu-editor) (1.6.2)
Requirement already satisfied: pycodestyle==2.4.0 in /usr/local/lib/python3.10/dist-packages (from mu-editor) (2.4.0)
Collecting pgzero==1.2
  Using cached pgzero-1.2-py3-none-any.whl (69 kB)
Requirement already satisfied: semver>=2.8.0 in ./.local/lib/python3.10/site-packages (from mu-editor) (2.13.0)
Requirement already satisfied: guizero>=1.1.0 in ./.local/lib/python3.10/site-packages (from mu-editor) (1.3.0)
Requirement already satisfied: Pillow>=5.2.0 in ./.local/lib/python3.10/site-packages (from mu-editor) (9.1.0)
Requirement already satisfied: pigpio>=1.40.post1 in /usr/local/lib/python3.10/dist-packages (from mu-editor) (1.78)
Requirement already satisfied: requests>=2.19.1 in /usr/lib/python3/dist-packages (from mu-editor) (2.25.1)
Requirement already satisfied: appdirs>=1.4.3 in ./.local/lib/python3.10/site-packages (from mu-editor) (1.4.4)
Requirement already satisfied: qtconsole==4.3.1 in /usr/local/lib/python3.10/dist-packages (from mu-editor) (4.3.1)
Requirement already satisfied: qscintilla==2.11.4 in /usr/local/lib/python3.10/dist-packages (from mu-editor) (2.11.4)
Requirement already satisfied: numpy>=1.7.1 in ./.local/lib/python3.10/site-packages (from matplotlib==2.2.2->mu-editor) (1.22.3)
Requirement already satisfied: kiwisolver>=1.0.1 in ./.local/lib/python3.10/site-packages (from matplotlib==2.2.2->mu-editor) (1.4.2)
Requirement already satisfied: pytz in ./.local/lib/python3.10/site-packages (from matplotlib==2.2.2->mu-editor) (2022.1)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in ./.local/lib/python3.10/site-packages (from matplotlib==2.2.2->mu-editor) (3.0.8)
Requirement already satisfied: six>=1.10 in ./.local/lib/python3.10/site-packages (from matplotlib==2.2.2->mu-editor) (1.16.0)
Requirement already satisfied: cycler>=0.10 in ./.local/lib/python3.10/site-packages (from matplotlib==2.2.2->mu-editor) (0.11.0)
Requirement already satisfied: python-dateutil>=2.1 in ./.local/lib/python3.10/site-packages (from matplotlib==2.2.2->mu-editor) (2.8.2)
Collecting pygame<2.0,>=1.9.2
  Using cached pygame-1.9.6.tar.gz (3.2 MB)
  Preparing metadata (setup.py) ... done
Collecting PyQt5-sip<13,>=12.7
  Using cached PyQt5_sip-12.10.1-cp310-cp310-manylinux1_x86_64.whl (352 kB)
Collecting jupyter-client>=4.1
  Using cached jupyter_client-7.3.0-py3-none-any.whl (130 kB)
Requirement already satisfied: pygments in /usr/lib/python3/dist-packages (from qtconsole==4.3.1->mu-editor) (2.11.2)
Collecting traitlets
  Using cached traitlets-5.1.1-py3-none-any.whl (102 kB)
Collecting jupyter-core
  Using cached jupyter_core-4.10.0-py3-none-any.whl (87 kB)
Collecting ipykernel>=4.1
  Using cached ipykernel-6.13.0-py3-none-any.whl (131 kB)
Collecting ipython-genutils
  Using cached ipython_genutils-0.2.0-py2.py3-none-any.whl (26 kB)
Requirement already satisfied: colorzero in /usr/local/lib/python3.10/dist-packages (from gpiozero>=1.4.1->mu-editor) (2.0)
Requirement already satisfied: packaging in ./.local/lib/python3.10/site-packages (from ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (21.3)
Collecting tornado>=6.1
  Using cached tornado-6.1-cp310-cp310-linux_x86_64.whl
Collecting nest-asyncio
  Using cached nest_asyncio-1.5.5-py3-none-any.whl (5.2 kB)
Collecting matplotlib-inline>=0.1
  Using cached matplotlib_inline-0.1.3-py3-none-any.whl (8.2 kB)
Collecting ipython>=7.23.1
  Using cached ipython-8.3.0-py3-none-any.whl (750 kB)
Requirement already satisfied: debugpy>=1.0 in /usr/local/lib/python3.10/dist-packages (from ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (1.6.0)
Collecting psutil
  Using cached psutil-5.9.0-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (281 kB)
Collecting pyzmq>=22.3
  Using cached pyzmq-22.3.0-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (1.1 MB)
Collecting entrypoints
  Using cached entrypoints-0.4-py3-none-any.whl (5.3 kB)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from colorzero->gpiozero>=1.4.1->mu-editor) (59.6.0)
Requirement already satisfied: pexpect>4.3 in /usr/lib/python3/dist-packages (from ipython>=7.23.1->ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (4.8.0)
Collecting backcall
  Using cached backcall-0.2.0-py2.py3-none-any.whl (11 kB)
Collecting jedi>=0.16
  Using cached jedi-0.18.1-py2.py3-none-any.whl (1.6 MB)
[... 1069 lines cut ...]
[... cut ...]
   from /home/travis/.local/lib/python3.10/site-packages/~ygame-2.1.2.dist-info
  Moving to /home/travis/.local/lib/python3.10/site-packages/pygame.libs/
   from /home/travis/.local/lib/python3.10/site-packages/~ygame.libs
  Moving to /home/travis/.local/lib/python3.10/site-packages/pygame/
   from /home/travis/.local/lib/python3.10/site-packages/~ygame
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pygame

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

dieseltravis, 2022-05-07:
I tried manually installing pygame==1.9.2 through pygame==1.9.6 and none of them succeed but the latest version of pygame 2.1.2 seems to install fine.

carlosperate, 2022-05-17:
I believe the problem encounter here would be this one: https://github.com/mu-editor/mu/issues/2224

Basically due to some dependencies, Mu is currently not yet compatible with Python 3.9+, so we started adding a max python version to the package metada with the v1.1 betas, so when `pip install mu-editor` is run with a newer Python version pip goes back through the `mu-editor` releases until it finds a release that doesn't have this limitation configured.

In this case it goes back to the v1.0.3 release, which is much older is likely even less compatible with newer Python versions. It might be possible to get it running by building some of the dependencies from source, but that can take quite a bit of effort.

As a workaround the section "Install Mu via PyPI and pipx" in https://github.com/mu-editor/mu/issues/2228 has steps indicating how to install an older Python version and Mu using pipx, which I've tested in Ubuntu 22.04 and should work.

dieseltravis, 2022-05-20:
Installing mu-editor with the **Install Mu via PyPI and pipx** section here worked for me:
https://github.com/mu-editor/mu/issues/2228

closed 2022-05-20 by dieseltravis