mu-editor/mu #2202: Mu Won't Start After Installation

sklogo7, 2022-05-01:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_GH
Platform: Windows 10 10.0.21996 AMD64

```
b\site-packages\mu\virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "C:\Users\DELL CORE i5 8TH GEN\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 885, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\DELL CORE i5 8TH GEN\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\DELL CORE i5 8TH GEN\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\DELL CORE i5 8TH GEN\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\DELL CORE i5 8TH GEN\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\DELL CORE i5 8TH GEN\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 259:

```

bnfgnn, 2022-05-02:
Piattaforma: Ubunto versione 22.04 LTS
Programma_ mu-editor 1.0.1
Il programma installato da Ubunto Software non si avvia dopo l'installazione.
L'icona è presente tra i programmi ma l'editor non si avvia.
Nessun messaggio di errore viene visualizzato.

carlosperate, 2022-05-10:
Thanks for the report @sklogo7!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

antoninguyen84, 2022-05-12:
[image]

antoninguyen84, 2022-05-12:
How can I solve this?

carlosperate, 2022-05-12:
Hi @antoninguyen84 could you open a new issue with error message and attach the log? (The log is located in the location written at the bottom of the screenshot you've provided).
That will help us separate different problem and be able to look into your problem. Thanks!

bnfgnn, 2022-05-12:
Salve, scusate se scrivo in italiano ma il mio inglese non me lo permette.
Ho risolto il problema con MU reinstallando il sistema operativo (UBUNTU 22.04).
Ora funziona come prima che si bloccasse.
Scusate il disturbo e grazie per l’interessamento.
Cordiali saluti.

Giovanni.

antoninguyen84, 2022-05-12:
How can I solve this problem ?
[image]

carlosperate, 2022-05-12:
Thanks for the solution @bnfgnn!

I'll attach here a Google translation from your message:

> Hi, sorry if I write in Italian but my English does not allow me.
I solved the problem with MU by reinstalling the operating system (UBUNTU 22.04).
Now it works as before it crashed.
Sorry for the inconvenience and thanks for your interest.
Best regards.

carlosperate, 2022-06-06:
Hi @sklogo7, are you still experiencing this issue? Would you be able to attach the log from `C:\Users\DELL CORE i5 8TH GEN\AppData\Local\python\mu\Logs\mu.log`