mu-editor/mu #2211: while installing mu-editor on linux pop os, i get errors "metadata-generation-failed" and "error: subprocess-exited-with-error      × python setup.py egg_info did not run successfully."

GBG7, 2022-05-08:
### What were you trying to do?

I was trying to install mu editor on linux using the guide on the website,

### What steps did you take to trigger the issue?

I simply attempted to type `pip3 install mu-editor` in the terminal. 

### What did you expect to happen?

I expected mu-editor to be installed normally, and allow me to launch mu-editor from terminal

### What actually happened?

I got errors "`metadata-generation-failed"` and "`error: subprocess-exited-with-error      × python setup.py egg_info did not run successfully."`

The error says "note: This is an issue with the package mentioned above, not pip.". Just in case though, i reinstalled pip, and ensured that im running python 3.10.4. I've tried to update everything i see, but it's not working and giving me tons of trouble.

### Operating System Version

Pop!_OS 22.04 LTS x86_64 

### Mu Version

mu won't install, so idk the version. Latest version that the linux guide installs I suppose

### Other Info

I've posted this question on stackoverflow, and a user suggested  that I "try running with `--use-deprecated=backtrack-on-build-failures` for example `pip install pygames --use-deprecated=backtrack-on-build-failures`" or run this command before doing my installation guide; "`sudo apt-get install python-dev libsdl-image1.2-dev libsdl-mixer1.2-dev libsdl-ttf2.0-dev libsdl1.2-dev libsmpeg-dev python-numpy subversion libportmidi-dev ffmpeg libswscale-dev libavformat-dev libavcodec-dev libfreetype6-dev Before running your commands`"

These didn't work either. step 1. Gave me an error saying couldn't find a version that satisfies the requirement pygames. 

step 2 said that python-dev was missing and that "`the following packages replace it: python2-dev:i386 python2:i386 python2-dev python2 python-dev-is-python3`" however, while installing mu-editor again, i get another new error saying "`python setup.py egg_info did not run successfully.`" and "error: metadata-generation-failed" AGAIN.

### Editor Log

Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: mu-editor in ./.local/lib/python3.10/site-packages (1.0.3)
Requirement already satisfied: pyserial==3.4 in ./.local/lib/python3.10/site-packages (from mu-editor) (3.4)
Requirement already satisfied: qscintilla==2.11.4 in ./.local/lib/python3.10/site-packages (from mu-editor) (2.11.4)
Requirement already satisfied: pigpio>=1.40.post1 in ./.local/lib/python3.10/site-packages (from mu-editor) (1.78)
Requirement already satisfied: requests>=2.19.1 in /usr/lib/python3/dist-packages (from mu-editor) (2.25.1)
Requirement already satisfied: appdirs>=1.4.3 in ./.local/lib/python3.10/site-packages (from mu-editor) (1.4.4)
Requirement already satisfied: pyflakes==2.0.0 in ./.local/lib/python3.10/site-packages (from mu-editor) (2.0.0)
Requirement already satisfied: gpiozero>=1.4.1 in ./.local/lib/python3.10/site-packages (from mu-editor) (1.6.2)
Requirement already satisfied: semver>=2.8.0 in ./.local/lib/python3.10/site-packages (from mu-editor) (2.13.0)
Requirement already satisfied: matplotlib==2.2.2 in ./.local/lib/python3.10/site-packages (from mu-editor) (2.2.2)
Requirement already satisfied: PyQtChart==5.14.0 in ./.local/lib/python3.10/site-packages (from mu-editor) (5.14.0)
Requirement already satisfied: guizero>=1.1.0 in ./.local/lib/python3.10/site-packages (from mu-editor) (1.3.0)
Requirement already satisfied: nudatus>=0.0.3 in ./.local/lib/python3.10/site-packages (from mu-editor) (0.0.5)
Requirement already satisfied: Pillow>=5.2.0 in /usr/lib/python3/dist-packages (from mu-editor) (9.0.1)
Requirement already satisfied: pycodestyle==2.4.0 in ./.local/lib/python3.10/site-packages (from mu-editor) (2.4.0)
Collecting pgzero==1.2
  Using cached pgzero-1.2-py3-none-any.whl (69 kB)
Collecting pyqt5==5.14.1
  Using cached PyQt5-5.14.1-5.14.1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (63.5 MB)
Requirement already satisfied: qtconsole==4.3.1 in ./.local/lib/python3.10/site-packages (from mu-editor) (4.3.1)
Collecting python-dateutil>=2.1
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Requirement already satisfied: six>=1.10 in /usr/lib/python3/dist-packages (from matplotlib==2.2.2->mu-editor) (1.16.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /usr/lib/python3/dist-packages (from matplotlib==2.2.2->mu-editor) (2.4.7)
Collecting kiwisolver>=1.0.1
  Using cached kiwisolver-1.4.2-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (1.6 MB)
Requirement already satisfied: numpy>=1.7.1 in ./.local/lib/python3.10/site-packages (from matplotlib==2.2.2->mu-editor) (1.22.3)
Collecting cycler>=0.10
  Using cached cycler-0.11.0-py3-none-any.whl (6.4 kB)
Requirement already satisfied: pytz in /usr/lib/python3/dist-packages (from matplotlib==2.2.2->mu-editor) (2022.1)
Collecting pygame<2.0,>=1.9.2
  Using cached pygame-1.9.6.tar.gz (3.2 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [12 lines of output]
      
      
      WARNING, No "Setup" File Exists, Running "buildconfig/config.py"
      Using UNIX configuration...
      
      /bin/sh: 1: sdl-config: not found
      /bin/sh: 1: sdl-config: not found
      /bin/sh: 1: sdl-config: not found
      Unable to run "sdl-config". Please make sure a development version of SDL is installed.
      
      Hunting dependencies...
      WARNING: "sdl-config" failed!
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

dieseltravis, 2022-05-09:
I am having the same issue (#2201). I think that it's because mu-editor is using an older version of pygame that won't work in 22.04 but I'm not sure.

GBG7, 2022-05-09:
> I am having the same issue (#2201). I think that it's because mu-editor is using an older version of pygame that won't work in 22.04 but I'm not sure.

I remember using mu editor right after updating my pop os for a bit. But after restarting my pc, it stopped working. I supposed you might be right. 
Would pygame have to update in order to work with 22.04? or would it be the other way around

GBG7, 2022-05-09:
update: I think the issue's with unroll. i need to manually install setup.py or something like that.

dieseltravis, 2022-05-18:
@carlosperate that workaround works for me, thanks!

carlosperate, 2022-05-18:
That's great, thanks for the update @dieseltravis!

closed 2022-05-25 by carlosperate