mu-editor/mu #2219: Crash Report

hypo5, 2022-05-12:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_GB
Platform: Windows 10 10.0.19042 AMD64

```
Traceback (most recent call last):
  File "C:\Program Files\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Program Files\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Program Files\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Program Files\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 771, in create
    self.create_venv()
  File "C:\Program Files\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Program Files\Mu Editor\Python\python.exe at C:\Users\tommy.thompson\AppData\Local\python\mu\mu_venv-38-20220512-100130
OSError: [Errno 28] No space left on device

```

carlosperate, 2022-05-12:
Hi @hypo5, thanks for the report!

Looking at the error it seems like you've run out of space in your C:\ drive. Could you free up some space and then try again?

carlosperate, 2022-06-08:
Hi @hypo5, are you still experiencing this problem? Was the issue disk space?