mu-editor/mu #2222: Crash Report

Tinkering-Townsperson, 2022-05-12:
**REMEMBER TO ATTACH YOUR LOG FILE**
[mu.log](https://github.com/mu-editor/mu/files/8683159/mu.log)

Version: 1.1.1
Locale: en_CA
Platform: Windows 10 10.0.19043 AMD64

```
Traceback (most recent call last):
  File "C:\Users\AfterNoonPM\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\python3.py", line 123, in stop_kernel
    self.repl_kernel_client.stop_channels()
  File "C:\Users\AfterNoonPM\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\qtconsole\kernel_mixins.py", line 54, in stop_channels
    super(QtKernelClientMixin, self).stop_channels()
  File "C:\Users\AfterNoonPM\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\jupyter_client\threaded.py", line 249, in stop_channels
    super().stop_channels()
  File "C:\Users\AfterNoonPM\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\jupyter_client\client.py", line 124, in stop_channels
    if self.shell_channel.is_alive():
AttributeError: 'QtKernelClient' object has no attribute 'shell_channel'

```

carlosperate, 2022-05-16:
Thanks for the report and the log @Tinkering-Townsperson!

I had a quick look at the source code of the jupyter_client dependency and I coudn't find a quick way to determine the issue without knowing the version of the other dependencies.

When Mu is running, could you copy the log from the settings window (the cog at the bottom right corner) and paste it here? When copied from that window the log should contain a list with the version of the dependencies.

carlosperate, 2022-07-05:
Hi @Tinkering-Townsperson, are you able to send us the log as shown in my last message? It'll be really useful for us to debug this issue.