mu-editor/mu #2223: Docs: AppImage additional instructions needed for Ubuntu 22.04

carlosperate, 2022-05-13:


carlosperate, 2022-05-13:
Interestingly about the `QT_QPA_PLATFORM=wayland`  en var, running Mu from source code and these three configurations seem to work (even if they print warnings on the terminal):

```
$ python run.py

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
libGL error: pci id for fd 23: 1ab8:0010, driver (null)
pci id for fd 24: 1ab8:0010, driver (null)
```

```
$ QT_QPA_PLATFORM=wayland python run.py 

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
QSocketNotifier: Can only be used with threads started with QThread
libEGL warning: pci id for fd 13: 1ab8:0010, driver (null)

pci id for fd 14: 1ab8:0010, driver (null)
```

```
$ QT_QPA_PLATFORM=xcb python run.py 

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
libGL error: pci id for fd 23: 1ab8:0010, driver (null)
pci id for fd 24: 1ab8:0010, driver (null)

```

```
$ env -u WAYLAND_DISPLAY  python run.py 

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
libGL error: pci id for fd 22: 1ab8:0010, driver (null)
pci id for fd 23: 1ab8:0010, driver (null)
```

tmontes, 2022-07-16:
Docs have been updated to include some distribution-specific details (see https://github.com/mu-editor/mu-editor.github.io/pull/70).

carlosperate, 2022-07-16:
That's great, thanks @tmontes! 🎉

closed 2022-07-16 by carlosperate