mu-editor/mu #2225: Crash Report

d1nes, 2022-05-15:
[[**REMEMBER]([url](url))](C:\Users\PC1\AppData\Local\python\mu\Logs\mu.log) TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\PC1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\PC1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\PC1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "C:\Users\PC1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in ensure
    self.ensure_interpreter_version()
  File "C:\Users\PC1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 732, in ensure_interpreter_version
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Mu interpreter at version 38; venv interpreter at version .

``
[mu.log](https://github.com/mu-editor/mu/files/8694674/mu.log)
`

carlosperate, 2022-05-16:
Thanks for the crash report and the logs @d1nes!

Could you run these commands in a command prompt and paste the result here?
I'm not sure why the venv python interpreter is not returning the Python version:

```
"C:\Users\PC1\AppData\Local\python\mu\mu_venv-38-20220515-113052.FAILED\scripts\python.exe" -c 'import sys; print("%s%s" % sys.version_info[:2])'
``` 

```
"C:\Users\PC1\AppData\Local\python\mu\mu_venv-38-20220515-113052.FAILED\scripts\python.exe" -c "import sys; print(sys.version_info)"
```

carlosperate, 2022-06-10:
Hi @d1nes 

Would you be able to run the commands from my previous message? It would be really useful for us to debug this issue.