mu-editor/mu #2228: Debian package issue in Ubuntu 22.04

carlosperate, 2022-05-16:


carlosperate, 2022-05-17:
@knowledgejunkie just to let you know the Debian package for Mu v1.0.3 has this issue with Python 3.10, which is the default python3 in Ubuntu 22.04. Although I believe Debian Bullseye is at 3.9, so it probably doesn't throw this exception there.

knowledgejunkie, 2022-05-17:
@carlosperate Thank you for the heads up and the patch - I've just uploaded an updated version of mu-editor 1.0.3 to Debian unstable which resolves this issue.

I'm looking at packaging mu-editor 1.1.1 in the next few weeks, along with the new micro:bit firmware (I only have a v1 device for testing)

carlosperate, 2022-05-22:
Thanks Nick!
I could also test it with a micro:bit V2 in a Virtual Machine if there were some easy steps to install a test deb (or configure the test deb channel).

Mu v1.1.1 might be a bit trickier to package, so let us know if you have any questions. Basically Mu now creates a virtual environment to install user-specific packages and run the user scripts isolated from the Python installation that runs Mu itself.
In this area, one thing that is worth keep in an eye for is this one: https://github.com/mu-editor/mu/issues/1634#issuecomment-1102458486

cpina, 2022-07-23:
> @knowledgejunkie just to let you know the Debian package for Mu v1.0.3 has this issue with Python 3.10, which is the default python3 in Ubuntu 22.04. Although I believe Debian Bullseye is at 3.9, so it probably doesn't throw this exception there.

I would like to confirm that:

- Debian Bullseye 11.4 with its mu-editor version 1.0.3+dfsg-2 and Python 3.9: all good "out of the box"
- Lubuntu Jammy Jellyfish 22.04 with its mu-editor version : I had to apply the patch from "Patch the source code"

knowledgejunkie, 2022-08-11:
@cpina Thank you for your findings.

The Python 3.10-loving 1.0.3+dfsg-4 version of mu-editor, which was uploaded to Debian in May, is available in the Ubuntu 22.10 Kinetic Kudo repository: https://packages.ubuntu.com/kinetic/mu-editor.

Note than on Debian, mu-editor et al have been removed from testing (bookworm) due to a dependency issue with jekyll. They'll reappear automatically once that has been resolved. They are still available in unstable.

I hope to get around to packaging the current release soon. Thanks @carlosperate for your offer of answering questions!