mu-editor/mu #2232: Crash Report

Brahmoz, 2022-05-17:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_US
Platform: Windows 10 10.0.22000 AMD64

```
s\mu\logic.py", line 721, in check_usb
    detected = mode.find_devices(with_logging=False)
  File "C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 423, in find_devices
    device = self.compatible_board(port)
  File "C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\circuitpython.py", line 295, in compatible_board
    for comport in circuitpython_serial.repl_comports():
  File "C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 69, in repl_comports
    for port in comports()
  File "C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 49, in comports
    ports = _list_ports_windows.comports()
  File "C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 458, in comports
    return list(iterate_comports())
  File "C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 342, in iterate_comports
    info = list_ports_common.ListPortInfo(port_name_buffer.value, skip_link_detection=True)
TypeError: __init__() got an unexpected keyword argument 'skip_link_detection'

``
[mu.log](https://github.com/mu-editor/mu/files/8706686/mu.log)
`

Brahmoz, 2022-05-17:
I am new to Mu trying to learn coding. This is happening on Windows 11 for a while. Could you pls tell me a solution?

carlosperate, 2022-05-17:
Thanks for the report and the log @Brahmoz!

The issue raise is saying that the constructor for `pyserial.list_ports_common.ListPortInfo()` doesn't have the `skip_link_detection` keyword argument, but as part of the Windows installer you should have PySerial v3.5, which definitely has that argument:
https://github.com/pyserial/pyserial/blob/v3.5/serial/tools/list_ports_common.py#L38

The only way this could happen is if somehow your Mu installation has an older PySerial version, or maybe using a PySerial for a different Python installation.

Could you run the following commands in a command prompt and paste here the results?

```
"C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
```

```
"C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\python.exe" -I -m pip list --verbose
```

Brahmoz, 2022-05-17:
Microsoft Windows [Version 10.0.22000.675]
(c) Microsoft Corporation. All rights reserved.

C:\Users\tuttu>C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
'C:\Users\tuttu\AppData\Local\Programs\Mu' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\tuttu>\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
The system cannot find the path specified.

C:\Users\tuttu>"C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
Package                Version Location                                                                                             Installer
---------------------- ------- ---------------------------------------------------------------------------------------------------- ---------
adafruit-board-toolkit 1.1.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
appdirs                1.4.3   c:\users\tuttu\appdata\roaming\python\python38\site-packages                                         pip
asttokens              2.0.5   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
backcall               0.2.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
black                  21.12b0 c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
click                  8.0.4   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
colorama               0.4.4   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
decorator              5.1.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
distlib                0.3.0   c:\users\tuttu\appdata\roaming\python\python38\site-packages                             
ecdsa                  0.13.3  c:\users\tuttu\appdata\roaming\python\python38\site-packages                                         pip
esptool                2.7     c:\users\tuttu\appdata\roaming\python\python38\site-packages                             
executing              0.8.2   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
filelock               3.0.12  c:\users\tuttu\appdata\roaming\python\python38\site-packages                                         pip
flake8                 4.0.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
ipykernel              5.5.6   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
ipython                8.0.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
ipython-genutils       0.2.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
jedi                   0.18.1  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
jupyter-client         6.1.12  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
jupyter-core           4.9.2   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
matplotlib-inline      0.1.3   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
mccabe                 0.6.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
mu-editor              1.1.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                 
mypy-extensions        0.4.3   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
nudatus                0.0.5   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
packaging              21.3    c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
parso                  0.8.3   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pathspec               0.9.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pickleshare            0.7.5   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pip                    19.3.1  c:\users\tuttu\appdata\roaming\python\python38\site-packages                                         pip
platformdirs           2.5.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
prompt-toolkit         3.0.28  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pure-eval              0.2.2   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pyaes                  1.6.1   c:\users\tuttu\appdata\roaming\python\python38\site-packages                             
[... cut ...]
  Found link https://files.pythonhosted.org/packages/f3/77/23152f90de45957b59591c34dcb39b78194eb67d088d4f8799e9aa9726c4/pip-22.1-py3-none-any.whl#sha256=802e797fb741be1c2d475533d4ea951957e4940091422bd4a24848a7ac95609d (from https://pypi.org/simple/pip/) (requires-python:>=3.7), version: 22.1
  Found link https://files.pythonhosted.org/packages/99/bb/696e256f4f445809f25efd4e4ce42ff99664dc089cafa1e097d5fec7fc33/pip-22.1.tar.gz#sha256=2debf847016cfe643fa1512e2d781d3ca9e5c878ba0652583842d50cc2bcc605 (from https://pypi.org/simple/pip/) (requires-python:>=3.7), version: 22.1
Given no hashes to check 193 links for project 'pip': discarding no candidates

carlosperate, 2022-05-17:
Thanks @Brahmoz, that's really useful.
I believe that's only the first command, could you also the second one and paste the result here?

```
"C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\python.exe" -I -m pip list --verbose
```

Thanks!

Brahmoz, 2022-05-17:
While running second line on CMD getting this

C:\Users\tuttu>C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\python.exe -I -m pip list --verbose
'C:\Users\tuttu\AppData\Local\Programs\Mu' is not recognized as an internal or external command,
operable program or batch file.

Brahmoz, 2022-05-17:
[mu.log](https://github.com/mu-editor/mu/files/8707039/mu.log)

carlosperate, 2022-05-17:
You have to copy and paste it exactly as it is, with the quote marks:
```
"C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\python.exe" -I -m pip list --verbose
```

Brahmoz, 2022-05-17:
C:\Users\tuttu>"C:\Users\tuttu\AppData\Local\Programs\Mu Editor\Python\python.exe" -I -m pip list --verbose
Package                Version Location                                                                                             Installer
---------------------- ------- ---------------------------------------------------------------------------------------------------- ---------
adafruit-board-toolkit 1.1.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
appdirs                1.4.4   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
asttokens              2.0.5   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
backcall               0.2.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
black                  21.12b0 c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
click                  8.0.4   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
colorama               0.4.4   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
decorator              5.1.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
distlib                0.3.4   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
executing              0.8.2   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
filelock               3.6.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
flake8                 4.0.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
ipykernel              5.5.6   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
ipython                8.0.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
ipython-genutils       0.2.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
jedi                   0.18.1  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
jupyter-client         6.1.12  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
jupyter-core           4.9.2   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
matplotlib-inline      0.1.3   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
mccabe                 0.6.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
mu-editor              1.1.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages
mypy-extensions        0.4.3   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
nudatus                0.0.5   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
packaging              21.3    c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
parso                  0.8.3   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pathspec               0.9.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pickleshare            0.7.5   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pip                    21.1.3  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages\pip-21.1.3-py3.8.egg
platformdirs           2.5.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
prompt-toolkit         3.0.28  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pure-eval              0.2.2   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pycodestyle            2.8.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pyflakes               2.4.0   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
Pygments               2.11.2  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pyparsing              3.0.7   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
PyQt5                  5.13.2  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
PyQt5-sip              12.9.1  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
[... cut ...]
typing-extensions      4.1.1   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
virtualenv             20.13.1 c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
wcwidth                0.2.5   c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip
wheel                  0.37.1  c:\users\tuttu\appdata\local\programs\mu editor\python\lib\site-packages                             pip

carlosperate, 2022-05-17:
Thanks so much @Brahmoz, based on that I think the problem is that Mu is accidentally picking up packages from a different Python installation.

Would you be able to test the windows installer from this page and see if that fixes the issue? https://github.com/mu-editor/mu/actions/runs/2299649753

You should be able to download it from the Artifacts section:
[image]

Thanks!

Brahmoz, 2022-05-17:
Thank you soo much for your suggestion. New installation running fine now

carlosperate, 2022-05-17:
That's great, thanks for the help with the debugging and the testing!
As we got it to work I'll close this, and the next Mu release will include the fix and should work out of the box :)

closed 2022-05-17 by carlosperate