mu-editor/mu #2234: Crash Report

TimmyMcPickles, 2022-05-19:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_US
Platform: Linux 5.4.0-110-generic #124-Ubuntu SMP Thu Apr 14 19:46:19 UTC 2022 x86_64

```
lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/tmp/.mount_Mu_EdiZRKLhR/usr/lib/python3.8/runpy.py", line 144, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/tmp/.mount_Mu_EdiZRKLhR/usr/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/ethan/.local/lib/python3.8/site-packages/ipykernel/__init__.py", line 5, in <module>
    from .connect import *  # noqa
  File "/home/ethan/.local/lib/python3.8/site-packages/ipykernel/connect.py", line 11, in <module>
    import jupyter_client
  File "/home/ethan/.local/lib/python3.8/site-packages/jupyter_client/__init__.py", line 11, in <module>
    from .manager import AsyncKernelManager  # noqa
  File "/home/ethan/.local/lib/python3.8/site-packages/jupyter_client/manager.py", line 32, in <module>
    from .provisioning import KernelProvisionerBase
  File "/home/ethan/.local/lib/python3.8/site-packages/jupyter_client/provisioning/__init__.py", line 1, in <module>
    from .factory import KernelProvisionerFactory  # noqa
  File "/home/ethan/.local/lib/python3.8/site-packages/jupyter_client/provisioning/factory.py", line 11, in <module>
    from entrypoints import EntryPoint
ModuleNotFoundError: No module named 'entrypoints'

```

carlosperate, 2022-05-19:
Thanks for the report @TimmyMcPickles!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

TimmyMcPickles, 2022-05-19:
> Thanks for the report @TimmyMcPickles! Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

I'm pretty new to Linux, so it might be a bit before I can find that file, but I'll try

TimmyMcPickles, 2022-05-19:
[mu.log](https://github.com/mu-editor/mu/files/8734291/mu.log)
@carlosperate

carlosperate, 2022-05-20:
Thanks for the log @TimmyMcPickles, was this ussing the AppImage? Or how did you install Mu?

TimmyMcPickles, 2022-05-30:
> @TimmyMcPickles if you used the AppImage (and only in that case), could you run this test version to see if it fixes your issue?
> 
> https://github.com/mu-editor/mu/actions/runs/2371920809
> 
> You can download it at the bottom of the page (for the link to work you need to be logged into GitHub): [image]

Yes, this seems to have fixed the issue. Thank you so much!!