mu-editor/mu #2235: Appimage crashes trying to find libcrypt.so.1

MasterGeekMX, 2022-05-19:
### What actually happened?

The editor never launches. if one tries and runs directly the appimage from the terminal the following error comes out:

```
./Mu_Editor-1.1.1-x86_64.AppImage
/tmp/.mount_Mu_Edi35H7hF/usr/bin/python3.8: error while loading shared libraries: libcrypt.so.1: cannot open shared object file: No such file or directory
```

no log file was generated

### Operating System Version

Arch Linux with Python 3.10.4

### Mu Version

1.1.1

MasterGeekMX, 2022-05-19:
UPDATE: bouncing around search results, I found that installing libxcrypt-compat may work.

So I did `sudo pacman -S libxcrypt-compat` and it worked...

but now it throws a different error:

```
No settings file found at /home/mgmx/.local/share/mu/venv.json; skipping
[appimage-binfmt-bypass/bin] ERROR: child exited with code 11
```

carlosperate, 2022-05-20:
Thanks for the report and the debugging @MasterGeekMX! 

@tmontes
Looking at the AppImage excludes list, `libcrypt.so.1` was there originally, but it's been commented out 3 year ago:
https://github.com/AppImage/pkg2appimage/blob/a9c85b7e61a3a883f4a35c41c5decb5af88b6b5d/excludelist#L13

So it should have been picked by the tool? Does pup need to include a list of additional libraries to include?

tmontes, 2022-07-16:
Hey @MasterGeekMX, here's an update directly from the EuroPython 2022 sprints... @hoh joined us and we tried running Mu's AppImage on his laptop, also running Arch Linux.

So far we could make it run under Wayland, but still hitting crashes on X11.