mu-editor/mu #2249: Mu Editor Crashing on open - Surface Go

r3dh0t, 2022-05-26:
### What were you trying to do?

Open MU

### What steps did you take to trigger the issue?

Opening Mu Editor on Surface Go laptops, crashes on open for about 40% of students. We tried restarting laptops, uninstall/installing, running as admin account etc. Works for my Year 10-12 students using i5 Dells and Lenovos

### Operating System Version

Windows 10

### Editor Log

`Traceback (most recent call last):
  File "C:\Users\eswee6\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\eswee6\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\eswee6\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\eswee6\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 771, in create
    self.create_venv()
  File "C:\Users\eswee6\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\eswee6\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\eswee6\AppData\Local\python\mu\mu_venv-38-20220526-142537
`

carlosperate, 2022-05-31:
Thanks for the report @r3dh0t!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\eswee6\AppData\Local\python\mu\Logs\mu.log`

r3dh0t, 2022-06-09:
@carlosperate 

Not sur ehow to attach 
`2022-06-09 14:02:21,758 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.1
2022-06-09 14:02:21,759 - root:270(run) INFO: uname_result(system='Windows', node='019531283051', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 9, GenuineIntel')
2022-06-09 14:02:21,764 - root:271(run) INFO: Platform: Windows-10-10.0.19041-SP0
2022-06-09 14:02:21,764 - root:272(run) INFO: Python path: ['C:\\Users\\landerson19\\Downloads', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-57.4.0-py3.8.egg', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-21.1.3-py3.8.egg', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin']
2022-06-09 14:02:21,765 - root:273(run) INFO: Language code: en_AU
2022-06-09 14:02:21,766 - mu.settings:220(load) WARNING: No settings file found at C:\Users\landerson19\AppData\Local\python\mu\settings.json; skipping
2022-06-09 14:02:21,767 - mu.settings:220(load) WARNING: No settings file found at C:\Users\landerson19\AppData\Local\python\mu\session.json; skipping
2022-06-09 14:02:23,089 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2022-06-09 14:02:27,173 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2022-06-09 14:02:27,173 - mu.virtual_environment:781(create_venv) INFO: Creating virtualenv: C:\Users\landerson19\AppData\Local\python\mu\mu_venv-38-20220609-140135
2022-06-09 14:02:27,213 - mu.virtual_environment:782(create_venv) INFO: Virtualenv name: mu_venv-38-20220609-140135
2022-06-09 14:02:27,223 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-I', '-m', 'virtualenv', '-p', 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\landerson19\\AppData\\Local\\python\\mu\\mu_venv-38-20220609-140135') with kwargs {'env': {'ALLUSERSPROFILE': 'C:\\ProgramData', 'APPDATA': 'C:\\Users\\landerson19\\AppData\\Roaming', 'COMMONPROGRAMFILES': 'C:\\Program Files\\Common Files', 'COMMONPROGRAMFILES(X86)': 'C:\\Program Files (x86)\\Common Files', 'COMMONPROGRAMW6432': 'C:\\Program Files\\Common Files', 'COMPUTERNAME': '019531283051', 'COMSPEC': 'C:\\windows\\system32\\cmd.exe', 'DRIVERDATA': 'C:\\Windows\\System32\\Drivers\\DriverData', 'FPS_BROWSER_APP_PROFILE_STRING': 'Internet Explorer', 'FPS_BROWSER_USER_PROFILE_STRING': 'Default', 'HOMEDRIVE': 'C:', 'HOMEPATH': '\\Users\\landerson19', 'LOCALAPPDATA': 'C:\\Users\\landerson19\\AppData\\Local', 'LOGONSERVER': '\\\\PNLDC025', 'NUMBER_OF_PROCESSORS': '4', 'ONEDRIVE': 'C:\\Users\\landerson19\\OneDrive - Brisbane Catholic Education', 'ONEDRIVECOMMERCIAL': 'C:\\Users\\landerson19\\OneDrive - Brisbane Catholic Education', 'OS': 'Windows_NT', 'PATH': 'C:\\Users\\landerson19\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\PyQt5\\Qt\\bin;C:\\windows\\system32;C:\\windows;C:\\windows\\System32\\Wbem;C:\\windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\windows\\System32\\OpenSSH\\;C:\\Users\\landerson19\\AppData\\Local\\Microsoft\\WindowsApps;;C:\\Users\\landerson19\\AppData\\Local\\Programs\\Microsoft VS Code\\bin', 'PATHEXT': '.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC', 'PROCESSOR_ARCHITECTURE': 'AMD64', 'PROCESSOR_IDENTIFIER': 'Intel64 Family 6 Model 142 Stepping 9, GenuineIntel', 'PROCESSOR_LEVEL': '6', 'PROCESSOR_REVISION': '8e09', 'PROGRAMDATA': 'C:\\ProgramData', 'PROGRAMFILES': 'C:\\Program Files', 'PROGRAMFILES(X86)': 'C:\\Program Files (x86)', 'PROGRAMW6432': 'C:\\Program Files', 'PSMODULEPATH': 'C:\\Program Files\\WindowsPowerShell\\Modules;C:\\windows\\system32\\WindowsPowerShell\\v1.0\\Modules', 'PUBLIC': 'C:\\Users\\Public', 'SESSIONNAME': 'Console', 'SYSTEMDRIVE': 'C:', 'SYSTEMROOT': 'C:\\windows', 'TEMP': 'C:\\Users\\LANDER~1\\AppData\\Local\\Temp', 'TMP': 'C:\\Users\\LANDER~1\\AppData\\Local\\Temp', 'USERDNSDOMAIN': 'BCE.CATHOLIC.EDU.AU', 'USERDOMAIN': 'BCE', 'USERDOMAIN_ROAMINGPROFILE': 'BCE', 'USERNAME': 'landerson19', 'USERPROFILE': 'C:\\Users\\landerson19', 'WINDIR': 'C:\\windows', 'QT_API': 'pyqt5', 'QT_AUTO_SCREEN_SCALE_FACTOR': '1', 'QT_MAC_WANTS_LAYER': '1'}}
2022-06-09 14:02:49,329 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-06-09 14:02:49,329 - mu.virtual_environment:797(create_venv) INFO: Created virtual environment using C:\Users\landerson19\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\landerson19\AppData\Local\python\mu\mu_venv-38-20220609-140135
2022-06-09 14:02:49,350 - mu.virtual_environment:873(install_baseline_packages) INFO: Installing baseline packages.
[... cut ...]
  File "C:\Users\landerson19\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 1311, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
2022-06-09 14:10:18,302 - mu.settings:169(save) DEBUG: Saving to C:\Users\landerson19\AppData\Local\python\mu\session.json
2022-06-09 14:10:18,304 - mu.settings:169(save) DEBUG: Saving to C:\Users\landerson19\AppData\Local\python\mu\venv.json
`

carlosperate, 2022-06-09:
Thank you @r3dh0t, that's really useful!

Looking at the log the situation is a bit weird, it seems like the error it's saying it cannot find the `C:\Users\landerson19\AppData\Local\python\mu\mu_venv-38-20220609-140135\scripts\python.exe`, but it was detected as being there a few seconds before that error, it has been able to use the same virtual environment `pip.exe` without issues 🤔

Could you run this in a command prompt to see what files are in the virtual env folder?

```
dir C:\Users\landerson19\AppData\Local\python\mu\mu_venv-38-20220609-140135\scripts\
```

r3dh0t, 2022-06-16:
@carlosperate 

Another student ended up having theirs just start working, not sure why. They had several folder versions with FAILED and but today jsut worked.

landerson19 student just had the one folder 38-2022-06-09 and this was the contents:

`Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----         9/06/2022   2:06 PM                __pycache__
-a----         9/06/2022   2:02 PM           2194 activate
-a----         9/06/2022   2:02 PM           1034 activate.bat
-a----         9/06/2022   2:02 PM           3071 activate.fish
-a----         9/06/2022   2:02 PM           1387 activate.nu
-a----         9/06/2022   2:02 PM           1758 activate.ps1
-a----         9/06/2022   2:02 PM           1193 activate_this.py
-a----         9/06/2022   2:03 PM         106907 black-primer.exe
-a----         9/06/2022   2:03 PM         106912 black.exe
-a----         9/06/2022   2:03 PM         106913 blackd.exe
-a----         9/06/2022   2:02 PM            510 deactivate.bat
-a----         9/06/2022   2:02 PM            333 deactivate.nu
-a----         9/06/2022   2:03 PM         106901 espefuse.exe
-a----         9/06/2022   2:03 PM         106902 espsecure.exe
-a----         9/06/2022   2:03 PM         106900 esptool.exe
-a----         9/06/2022   2:05 PM         106908 f2py.exe
-a----         9/06/2022   2:03 PM         106900 flask.exe
-a----         9/06/2022   2:04 PM         106916 ipython.exe
-a----         9/06/2022   2:04 PM         106916 ipython3.exe
-a----         9/06/2022   2:04 PM         106915 jupyter-kernel.exe
-a----         9/06/2022   2:04 PM         106953 jupyter-kernelspec.exe
-a----         9/06/2022   2:04 PM         106911 jupyter-migrate.exe
-a----         9/06/2022   2:04 PM         106932 jupyter-run.exe
-a----         9/06/2022   2:04 PM         106916 jupyter-troubleshoot.exe
-a----         9/06/2022   2:04 PM         106911 jupyter.exe
-a----         9/06/2022   2:05 PM         106904 pgzrun.exe
-a----         9/06/2022   2:02 PM         115937 pip-3.8.exe
-a----         9/06/2022   2:02 PM         115937 pip.exe
-a----         9/06/2022   2:02 PM         115937 pip3.8.exe
-a----         9/06/2022   2:02 PM         115937 pip3.exe
-a----         9/06/2022   2:02 PM             24 pydoc.bat
-a----         9/06/2022   2:05 PM         106907 pygmentize.exe
-a----         9/06/2022   2:06 PM         106912 pyserial-miniterm.exe
-a----         9/06/2022   2:06 PM         106914 pyserial-ports.exe
-a----         9/06/2022   2:02 PM         544064 python.exe
-a----         9/06/2022   2:02 PM         543040 pythonw.exe
-a----         9/06/2022   2:06 PM          26854 pywin32_postinstall.py
-a----         9/06/2022   2:06 PM           3521 pywin32_testall.py
-a----         9/06/2022   2:02 PM         115924 wheel-3.8.exe
-a----         9/06/2022   2:02 PM         115924 wheel.exe
-a----         9/06/2022   2:02 PM         115924 wheel3.8.exe
-a----         9/06/2022   2:02 PM         115924 wheel3.exe
`

carlosperate, 2022-06-17:
Thanks for the update @r3dh0t !
Was the `mu_venv-38-20220609-140135` perhaps named `mu_venv-38-20220609-140135.FAILED`?

Would you be able to run this command as well? i'd be good to see what are all the folder names:
```
dir C:\Users\landerson19\AppData\Local\python\mu\
```

carlosperate, 2022-07-05:
Hi @r3dh0t, are you able to run the command from my last message? it'll help us debug the issue.

profasare, 2023-01-23:
I have a student having this same issue right now. I've asked for their computer specs (and will reply once I have it), but from my digging this is the offending line: 

Running ('C:\\Users\\landerson19\\AppData\\Local\\python\\mu\\mu_venv-38-20220609-140135\\scripts\\python.exe', '-c', 'import sys; print("%s%s" % sys.version_info[:2])') with kwargs {'shell': True}

I actually opened the specific python interpreter for the virtual environment that failed and ran that line, it produces nothing (empty string?), which is what causes the rest of the process to crash. The sys.version_info seems to be an object that is not array-like and hence the subscript doesn't produce the required output. I don't know why it happens on this particular machine only.