mu-editor/mu #2251: Crash Report

kumarsalugu, 2022-05-28:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_IN
Platform: Linux 5.4.0-113-generic #127-Ubuntu SMP Wed May 18 14:30:56 UTC 2022 x86_64

```
Traceback (most recent call last):
  File "/home/dharani/.local/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/dharani/.local/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/dharani/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/dharani/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/dharani/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/dharani/.local/share/mu/mu_venv-38-20220528-175958
STDERR: /usr/bin/python3: No module named virtualenv

```

kumarsalugu, 2022-05-28:
Unable to create a virtual environment

carlosperate, 2022-05-31:
Hi @kumarsalugu, thanks for the report.

I believe this issue might be https://github.com/mu-editor/mu/issues/1954#issuecomment-1001258055.

Possible workarounds: 
- Install the `python3-virtualenv` package using apt (or your OS package manager)
- Install Mu in a virtual environment, as that is the recommended way and it should work much better.
- I would also personally recommend to use pipx, as it's great for installing Python-based applications: https://pypa.github.io/pipx/
    - pipx will automatically install Python applications in their own isolated virtual environment, and add the executable to the PATH so that it can be easily launched like any other app.