mu-editor/mu #2270: Mu Editor 1.1.1 crashes when CircuitPython device connected

aoyamaprod, 2022-06-22:
### What were you trying to do?

Connect CircuitPython device to a USB port

### What steps did you take to trigger the issue?

1. Start Up Mu Editor 1.1.1
2. Select the CircuitPython mode
3. Connect CircuitPython devices (tried Adafruit CLUE and Seeed XIAO BLE)
4. Mu Editor crashes and returned the log below

### What did you expect to happen?

Recognize the CircuitPython device

### What actually happened?

Mu Editor crashes and returned the error

### Operating System Version

Windows 10 Pro 21H1

### Mu Version

1.1.1

### Editor Log

Starting Mu 1.1.1
2022-06-23 08:11:45,412 - root:270(run) INFO: uname_result(system='Windows', node='Elias', release='10', version='10.0.19043', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
2022-06-23 08:11:45,460 - root:271(run) INFO: Platform: Windows-10-10.0.19043-SP0
2022-06-23 08:11:45,460 - root:272(run) INFO: Python path: ['C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\yojis\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-57.4.0-py3.8.egg', 'C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-21.1.3-py3.8.egg', 'C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\yojis\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin']
2022-06-23 08:11:45,460 - root:273(run) INFO: Language code: ja_JP
2022-06-23 08:11:45,745 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2022-06-23 08:11:45,746 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2022-06-23 08:11:45,747 - mu.virtual_environment:685(ensure_path) INFO: Virtual Environment found at: C:\Users\yojis\AppData\Local\python\mu\mu_venv-38-20220623-075318
2022-06-23 08:11:45,747 - mu.virtual_environment:698(ensure_interpreter) INFO: Interpreter found at: C:\Users\yojis\AppData\Local\python\mu\mu_venv-38-20220623-075318\scripts\python.exe
2022-06-23 08:11:45,747 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\yojis\\AppData\\Local\\python\\mu\\mu_venv-38-20220623-075318\\scripts\\python.exe', '-c', 'import sys; print("%s%s" % sys.version_info[:2])') with kwargs {'shell': True}
2022-06-23 08:11:46,245 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 38
2022-06-23 08:11:46,246 - mu.virtual_environment:730(ensure_interpreter_version) INFO: Both interpreters at version 38
2022-06-23 08:11:46,251 - mu.virtual_environment:759(ensure_pip) INFO: Pip found at: C:\Users\yojis\AppData\Local\python\mu\mu_venv-38-20220623-075318\scripts\pip.exe
2022-06-23 08:11:46,256 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: pgzero
2022-06-23 08:11:46,256 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\yojis\\AppData\\Local\\python\\mu\\mu_venv-38-20220623-075318\\scripts\\python.exe', '-c', 'import pgzero') with kwargs {'shell': True}
2022-06-23 08:11:46,383 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-06-23 08:11:46,383 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: flask
2022-06-23 08:11:46,383 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\yojis\\AppData\\Local\\python\\mu\\mu_venv-38-20220623-075318\\scripts\\python.exe', '-c', 'import flask') with kwargs {'shell': True}
2022-06-23 08:11:46,967 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-06-23 08:11:46,968 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipykernel
2022-06-23 08:11:46,968 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\yojis\\AppData\\Local\\python\\mu\\mu_venv-38-20220623-075318\\scripts\\python.exe', '-c', 'import ipykernel') with kwargs {'shell': True}
2022-06-23 08:11:47,860 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-06-23 08:11:47,860 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipython_genutils
2022-06-23 08:11:47,860 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\yojis\\AppData\\Local\\python\\mu\\mu_venv-38-20220623-075318\\scripts\\python.exe', '-c', 'import ipython_genutils') with kwargs {'shell': True}
2022-06-23 08:11:47,980 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-06-23 08:11:47,980 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: esptool
2022-06-23 08:11:47,980 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\yojis\\AppData\\Local\\python\\mu\\mu_venv-38-20220623-075318\\scripts\\python.exe', '-c', 'import esptool') with kwargs {'shell': True}
2022-06-23 08:11:48,148 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-06-23 08:11:48,148 - mu.virtual_environment:628(ensure_and_create) INFO: Valid virtual environment found at C:\Users\yojis\AppData\Local\python\mu\mu_venv-38-20220623-075318
2022-06-23 08:11:48,149 - mu.settings:169(save) DEBUG: Saving to C:\Users\yojis\AppData\Local\python\mu\venv.json
2022-06-23 08:11:48,186 - mu.logic:768(__init__) INFO: Setting up editor.
2022-06-23 08:11:48,187 - mu.logic:789(__init__) INFO: Log directory: C:\Users\yojis\AppData\Local\python\mu\Logs
2022-06-23 08:11:48,187 - mu.logic:790(__init__) INFO: Data directory: C:\Users\yojis\AppData\Local\python\mu
2022-06-23 08:11:48,187 - mu.logic:804(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2022-06-23 08:11:48,187 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\yojis\mu_code from settings file
2022-06-23 08:11:48,326 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\yojis\mu_code from settings file
2022-06-23 08:11:48,326 - mu.logic:1549(change_mode) INFO: Workspace directory: C:\Users\yojis\mu_code
2022-06-23 08:11:48,326 - mu.logic:878(restore_session) DEBUG: <SessionSettings from C:\Users\yojis\AppData\Local\python\mu\session.json>
2022-06-23 08:11:48,338 - mu.logic:1496(select_mode) INFO: Showing available modes: ['python', 'circuitpython', 'microbit', 'esp', 'web', 'pyboard', 'debugger', 'pygamezero', 'lego', 'pico']
2022-06-23 08:12:14,337 - mu.logic:1503(select_mode) INFO: New mode selected: circuitpython
2022-06-23 08:12:14,678 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\yojis\mu_code from settings file
[... 5 lines cut ...]
2022-06-23 08:12:21,455 - mu.logic:900(restore_session) INFO: Loaded files.
2022-06-23 08:12:21,455 - mu.logic:906(restore_session) INFO: User defined environment variables: {}
2022-06-23 08:12:21,455 - mu.logic:911(restore_session) INFO: Minify scripts on micro:bit? False
2022-06-23 08:12:21,566 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\yojis\mu_code from settings file
2022-06-23 08:12:21,566 - mu.logic:1549(change_mode) INFO: Workspace directory: C:\Users\yojis\mu_code
2022-06-23 08:12:22,189 - mu.logic:962(restore_session) INFO: Starting with blank file.
2022-06-23 08:12:37,414 - root:171(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\yojis\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 721, in check_usb
    detected = mode.find_devices(with_logging=False)
  File "C:\Users\yojis\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 423, in find_devices
    device = self.compatible_board(port)
  File "C:\Users\yojis\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\circuitpython.py", line 295, in compatible_board
    for comport in circuitpython_serial.repl_comports():
[... cut ...]
  File "C:\Users\yojis\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 342, in iterate_comports
    info = list_ports_common.ListPortInfo(port_name_buffer.value, skip_link_detection=True)
TypeError: __init__() got an unexpected keyword argument 'skip_link_detection'
2022-06-23 08:12:37,881 - mu.settings:169(save) DEBUG: Saving to C:\Users\yojis\AppData\Local\python\mu\session.json
2022-06-23 08:12:37,882 - mu.settings:169(save) DEBUG: Saving to C:\Users\yojis\AppData\Local\python\mu\venv.json

carlosperate, 2022-06-23:
Hi @aoyamaprod,

Would you be able to try windows installer from this page? https://github.com/mu-editor/mu/actions/runs/2545517559

You can download it from the Artifacts section at the bottom of the page (make sure you are logged in GitHub for the download link to work):
[image]

I think the updates in the latest test build could resolve the issue you might have with the last release. It would be really useful for us if you could test it and confirm if it works, thanks!

aoyamaprod, 2022-06-23:
The new build you provided works perfectly!
Now Mu Editor does not crash after connecting the CircuitPython Boards (tried Adafruit CLUE and Seeed XIAO BLE). Thank you!!

carlosperate, 2022-06-23:
Great, thanks for the update! I'll close this as resolved 👍

closed 2022-06-23 by carlosperate