mu-editor/mu #2271: Web Mode docs: Add documentation about Python Anywhere uploads

carlosperate, 2022-06-24:
### What steps did you take to trigger the issue?

Also include a note about the rate limiting being IP based: https://github.com/mu-editor/mu/pull/2264#issuecomment-1164781615

carlosperate, 2022-06-24:
Actually, looks like we don't have any docs for the web mode at all.

The Web tutorial link here 404s as it doesn't have an artcile:
- https://codewith.mu/en/tutorials/1.1/
- https://codewith.mu/en/tutorials/1.1/web
- https://github.com/mu-editor/mu-editor.github.io/tree/master/en/tutorials/1.1

We might also need to update this page to add the new modes:
- https://github.com/mu-editor/mu-editor.github.io/blob/master/en/tutorials/1.1/modes.md

prcutler, 2022-07-27:
I don't know Flask personally, but I'd be happy to partner or assist someone if they are interested in working on a Web tutorial for Flask.

carlosperate, 2022-07-28:
Thanks @prcutler!
The web mode has been developmed mainly by @ntoll I believe, who is now on holiday for about a week.
I'd be happy to review any content, although I'm not too versed in Flash or this mode, I can try to respond to any questions you might have.

And, once again, thanks for all your contributions @prcutler, they are really appreciated and will help a lot of Mu users!