mu-editor/mu #2274: Mu editor on newer Macs such as Dec 2021 16" Mac Book Pro (M1x)

mlmiller751, 2022-06-25:
### What were you trying to do?

Install Mu

### What steps did you take to trigger the issue?

Standard installer from .dmg file. Try to drag the app into the applications folder

### What did you expect to happen?

mu.app would appear in Applications folder

### What actually happened?

Error pops up: You can’t copy the item “Mu Editor.app” because its name is too long or includes characters that are invalid on the destination volume.

### Operating System Version

MacOS 12.4

### Mu Version

1.1.1

### Other Info

It is possible to run the app direction from the .dmg file but not possible to copy "Mu Editor.app" onto the newer file system of this type of Mac.

carlosperate, 2022-06-26:
@ntoll I can't remember, but I think you also tried this in an M1?
I only have intel macbooks to test.

mlmiller751, 2022-06-26:
I am trying the following workaround.

1. Install Mu on an external USB stick (FAT format)
2. Test on M1x Mac with Pico.

My prediction is that it will work, meaning the underlying problem has to do with the APFS file system native to the M1x Macs.

I will let you know shortly. It's a slow copy!

mlmiller751, 2022-06-26:
Sounds right. If I’m unavailable for a bit, I’m sorting through Monterey issues on my obsolete MBP.😩

mlmiller751, 2022-06-27:
Also no joy. Hence, it is a M1x issue possibly AS WELL AS an APFS issue. It doesn't like the file format and it doesn't like the code either!

I posted another report, this time with mu.log and a screenshot of my hardware config  (from Apple System Info).

My workaround is that I am now able to use other editors such as IDLE or Emacs with the "screen" command from Terminal to access print/input activity.

I was hoping to teach a University course using the Pico, possibly with Mu Editor for CircuitPython, but probably I'll tell them to use IDLE instead if they have a newer Mac. I have not tested the Mu editor on Windows 11 or on older Macs yet (but Mike has had success on an older Mac). At one point, I had gotten it to work sufficiently to see that it has some nice plotting capability, but I am no longer able to get that far with launching it. It is very odd that it worked that one time (that was launching from the disk image).

It is also incredibly slow trying to start up, which I attribute to the Mac being incredibly paranoid about security. This wouldn't be so bad if it were a 1-time thing after initial installation. But I think launching it from a read-only disk image is why it takes a very long time every time (and now just doesn't ever finish launching at all). I actually end up having to do a cold boot of my Mac just to restore normal operation after a couple of tries launching Mu. I believe it is holding onto some system resources after it fails, such as PTYs.

Regards, Mark

mlmiller751, 2022-06-27:
> I'm getting a new M1 Mac tomorrow. I'll try to recreate the problems you've encountered and will ping the relevant folks in the Mu core team with the expertise to help.

Wonderful! Thank you!

ntoll, 2022-06-29:
So, I've just tested things with my M1 Mac. Everything just works and I encountered no problems at all. :frowning_face: I downloaded the Mac installer from our codewith.mu website and it all worked first time, as expected.

While this is a relief for Mu, it doesn't help with your problem. 

I wonder, what is specific about your setup that means Mu can't start. Is your laptop managed in any way by a third party (for instance your employer), who may restrict the applications that can be run on the device..? Is Rosetta installed on the device (thus allowing you to run Intel binaries on M1 devices)..? I wonder if there are any system logs (not sure where) that will reveal what's interfering with Mu. I'm mostly a Linux user so I'm not sure where you'd find such things.

Thoughts?

carlosperate, 2022-06-29:
There is a bit more info in https://github.com/mu-editor/crash-reports/issues/202

carlosperate, 2022-06-29:
That shouldn't be a problem though, right? I'd assume it'd be the same as copying the "Mu Editor.app" bundle into another directory and launching it from there? Which I've done quite a lot to keep multiple Mu versions around.

It's worth a try, launching from the mounted dmg in a computer where we know Mu works, in case it does produce weird results, but I think in @mlmiller751's computer launching from anywhere is causing problems.

mlmiller751, 2022-06-29:
And I only tried that because it fails to drag-install into Applications folder on APFS file system.

Mark L. Miller, Ph.D.
http://learningtech.org/
San Carlos, CA 94070
***@***.***
650-598-0105 x203

ntoll, 2022-06-30:
So, my Mac is a standard 2021 M1pro 16" MBP that has been maxxed out too... (32GB memory, 1Tb HD etc). I didn't have to give Mu any permissions (my user has admin privileges), and my laptop is actually owned by my employer so they have added security settings and virus protection over which I have no control. I can confirm the filesystem is APFS. I also have a completely up to date and fully patched OSX Monterey version 12.4.

It feels to me like we have pretty much the same setup, so I wonder what else might be different to cause the behaviour you describe..? Like I said, it all works flawlessly for me (which isn't at all helpful for this situation).

Thoughts?

mlmiller751, 2022-07-01:
First, thanks for the follow-through. I've pasted in an abbreviation version of my hardware config below. I am on the latest released Monterey 12.4. My Pico works fine in all other respects.
I can edit the files on it using IDLE, Emacs, or other editors. I can open a screen from terminal to see output from print("whatever"). So the only problem is Mu itself.

My user is a full administrator. I even looked at copying the file from the Terminal into Applications, with sudo. No joy. It tries, runs for a while, and then fails.

I need Mu to work because my students at Northeastern University in the fall will be told to use Mu as their initial editor. Some will be on Windows, some on new Macs, some on older Macs.
I cannot really even expect them to have PowerShell on Windows, much less Linux subsystem for Windows; so I need to use other editors that may not work the same way on Windows (or without providing access for print output). For those with older Macs, we are OK. For any with newer ones, we have a problem. And I cannot demonstrate instructions or examples using my own Mac!

It is encouraging that it works for you, on a nearly identical configuration; but baffling that, despite our similar systems, it does not work for me.

Thanks for any other ideas/suggestions you can think of.

Regards, Mark

Hardware Overview:

  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro18,2
  Chip:	Apple M1 Max
  Total Number of Cores:	10 (8 performance and 2 efficiency)
  Memory:	64 GB
  System Firmware Version:	7459.121.3
  OS Loader Version:	7459.121.3
  Serial Number (system):	MR40X379FP
  Hardware UUID:	9C2679AB-1B82-5047-A064-644ED037F2AA
  Provisioning UDID:	00006001-000C19162204801E
  Activation Lock Status:	Enabled

Data:

  Free:	3.36 TB (3,363,431,899,136 bytes)
  Capacity:	4 TB (3,996,329,328,640 bytes)
  Mount Point:	/System/Volumes/Data
  File System:	APFS
  Writable:	Yes
  Ignore Ownership:	No
  BSD Name:	disk3s5
  Volume UUID:	3004839C-9B8B-4B2A-B0E3-F15FFC71C787
  Physical Drive:
  Device Name:	APPLE SSD AP4096R
  Media Name:	AppleAPFSMedia
  Medium Type:	SSD
  Protocol:	Apple Fabric
  Internal:	Yes
  Partition Map Type:	Unknown
  S.M.A.R.T. Status:	Verified

Mark L. Miller, Ph.D.
Executive Director
Learningtech.org <http://learningtech.org/>
***@***.*** ***@***.***>
650-598-0105 x203

"Prof Mark"
Khoury College, Align Program
Northeastern University, SF Bay Area
***@***.*** ***@***.***>

mlmiller751, 2022-07-01:
Here is the only thing I have noticed that might be helpful. It is looking for a slightly more recent version of setuptools than I see on my own machine.
Is there a pip install I need to do before I can use Mu Editor from the .dmg? Or maybe tell me where to find the exact version you rely on?

FileNotFoundError: [Errno 2] No such file or directory: '/Users/mlmiller/Library/Application Support/virtualenv/wheel/house/setuptools-62.3.4-py3-none-any.whl'
But what I actually find in the Application Support folder is this:

cd /Users/mlmiller/Library/Application\ Support/virtualenv/wheel/house 
***@***.*** house % ls
pip-21.3.1-py3-none-any.whl
pip-21.3.1-py3-none-any.whl.lock
pip-22.0.3-py3-none-any.whl
setuptools-60.2.0-py3-none-any.whl
setuptools-60.2.0-py3-none-any.whl.lock
setuptools-60.6.0-py3-none-any.whl
wheel-0.37.1-py2.py3-none-any.whl
wheel-0.37.1-py2.py3-none-any.whl.lock
***@***.*** house %