mu-editor/mu #2275: Changing location of data folder

OldManTechWorld, 2022-06-25:
Hi all...
We are using MU Editor in our school and love it apart from one aspect.
As we have multiple students use each machine, on log off it deletes their profile.
We are therefore finding, therefore that MU Editor has to re-create everything each time which take an age on some of our (older) machines.
Is there a way to get it to read/write from another folder rather than the user’s profile folder ?

tjguk, 2022-06-25:
@OldManTechWorld depending on what "deletes their profile" means, the following might or might not help.

When it starts up, Mu will look for a file located at `%LOCALAPPDATA%\python\mu\settings.json`. (By default that file doesn't exist, so don't be surprised if you look and there's nothing there). 

That file is a JSON file and, for the purposes of this question, could contain something like the following:

`{ "workspace" : "c:/temp/mu-workspace" }`

Now if deleting the profile also involves deleting the contents of that directory then clearly we're back we started. We do have a plan afoot to allow command-line switches (https://github.com/mu-editor/mu/issues/693) but it needs someone to contribute the code.

Let me know if the `settings.json` file does work for you. If not, I'll see if we can come up with an alternative

tjguk, 2022-06-25:
BTW I clearly need to document the "settings" setup: I had to go and revisit the code to remind myself before writing that reply. And I wrote that code in the first place!

tjguk, 2022-06-25:
Sorry -- one more piece of info: the settings file does honour environment variables. So, if it helped, that setting could read, eg:

`{ "workspace" : "%USERPROFILE%/mu-workspace" }`

dmail00, 2022-07-21:
I wish that I found this issue when making a portable version of Mu. I instead went the route of making start.bat file which sets up some files and enviroment variables, then making the following changes to Mu source.
logic.py
LOG_DIR = os.environ.get("MU_LOG_DIR")

config.py
DATA_DIR = os.environ.get("MU_DATA_DIR")
VENV_DIR = os.path.join(os.environ.get("MU_VENV_DIR"), VENV_NAME)
HOME_DIRECTORY = os.environ.get("MU_HOME_DIR")