mu-editor/mu #2277: Microbit: Import error

mrunal093, 2022-06-27:
### What were you trying to do?

Calling a function from a user defined micropython file.

### What steps did you take to trigger the issue?

I first move the robotbitV4.py file (which has the functions) using the file system. I then try to import it in the main file and call the functions in robotbitV4.py. Here is the main.py : 
from microbit import *
import robobitV4 as r
display.show("2")
# port definitions.
M1A = 0x1
M1B = 0x2
M2A = 0x3
M2B = 0x4

# Motor drive test

r.setup()     # Set up the robobit
while True:
    display.show("S")   # show the loop started
    r.motor(M1A,100)    # max speed clockwise
    sleep(2000)         # pause to see change in motor speed
    display.show("R")   # display R for reverse
    r.motor(M1A,-100 )  # request max speed in reverse)
    sleep(2000)         # pause to see change in motor speed
    display.show("E")   # show at the end of loop
    sleep(1000)

**_**_### And HERE is the robobitV4.py_**_**

# ROBOTBIT version 3 to help alleviate memory issues in microbit version 1
from microbit import *
from time import sleep
from time import sleep_us
from machine import time_pulse_us
import math, ustruct

def setup():
# startup process - initialize i2c connection with motor driver
    i2c.init()
    i2c.write(0x40,bytearray([0x00, 0x00]))        #self.i2c.send(bytearray([0x00, 0x00]), self.address)
    set_all_pwm(0,0)
    i2c.write(0x40,bytearray([0x01, 0x04]))    # self.i2c.send(bytearray([0x01, 0x04]), self.address)
    i2c.write(0x40,bytearray([0x00, 0x01]))    #self.i2c.send(bytearray([0x00, 0x01]), self.address)
    sleep(0.005)  # wait for oscillator
    i2c.write(0x40, bytearray([0x00])) # write register we want to read from first
    mode1 = i2c.read(0x40,1)[0]
    mode1 = 0x00 & ~0x10
    i2c.write(0x40,bytearray([0x00, 0x00]))
    sleep(0.005)  # Wait for ascillator
    set_pwm_freq(50)
#    print("pwm data ",set_pwm(S1,None,None))

def set_all_pwm(on, off):
#  Sets all PWM channels."""
#    i2c.send(bytearray([0xFA, on & 0xFF]), self.address)
    i2c.write(0x40,bytearray([0xFA, on & 0xFF]))
#    i2c.send(bytearray([0xFB, on >> 8]), self.address)
    i2c.write(0x40,bytearray([0xFB, on >> 8]))  # Original Had error
#    i2c.send(bytearray([0xFC, off & 0xFF]), self.address)
    i2c.write(0x40,bytearray([0xFC, off & 0xFF]))
#    i2c.send(bytearray([0xFD, off >> 8]), self.address)
    i2c.write(0x40,bytearray([0xFD, off >> 8]))

def set_pwm_freq(freq_hz):
#Set the PWM frequency to the provided value in hertz."""
    prescaleval = 25000000.0    # 25MHz
    prescaleval /= 4096.0       # 12-bit
    prescaleval /= float(freq_hz)
[... 634 lines cut ...]
pgzero            1.2.1
pickleshare       0.7.5
pip               22.0.3
platformdirs      2.5.1
prompt-toolkit    3.0.28
ptyprocess        0.7.0
pure-eval         0.2.2
pycparser         2.21
pygame            2.1.2
Pygments          2.11.2
pyserial          3.5
python-dateutil   2.8.2
pyzmq             22.3.0
reedsolo          1.5.4
setuptools        60.9.3
six               1.16.0
stack-data        0.2.0
tomli             2.0.1
tornado           6.1
traitlets         5.1.1
typing_extensions 4.1.1
wcwidth           0.2.5
Werkzeug          2.0.3
wheel             0.37.1
2022-06-27 18:08:43,230 - mu.virtual_environment:931(installed_packages) INFO: []

carlosperate, 2022-06-28:
Hi @mrunal093, thanks for the report!

If I understood correctly this is your current setup and flashing method:
- A custom hex file added to the micro:bit settings
- First you add a module via the "Files" panel
- Then you click "Flash"
- When the programme runs in the micro:bit it fails at the point it tries to import the library that was transfered via "Files"

Is this correct?

If this is the case, it's because when a custom MicroPython is added to the settings Mu doesn't have a way to know if the MicroPython running on your micro:bit is the same as the hex file specified. For this reason it has to flash MicroPython every time the "Flash" button is pressed, and then it sends `main.py`.
Flashing MicroPython means erasing all the micro:bit flash contents, so the library you've previously transfered is erased.

In this case what you need to do is first click "Flash" and then transfer any library via the "Files" panel.

carlosperate, 2022-07-05:
Hi @mrunal093, did the instructions from my last message help?