mu-editor/mu #2282: Can't get Mu first run to complete for user - may be linked to redirected AppData

altecsole, 2022-07-12:
### What were you trying to do?

Trying to complete first run for student user

### What steps did you take to trigger the issue?

I installed MU using the .msi with the ALLUSERS=1 switch.

I created a start menu shortcut for "C:\Program Files (x86)\Mu Editor\Python\pythonw.exe" -m mu

The initial configuration starts, but fails.

We do have folder redirection in place to redirect user AppData to their home folders on the server, so this may be an issue? 

Also, on a clean installation I did try installing Pyton 3.8 and then installing Mu using the instructions at https://codewith.mu/en/howto/1.1/install_with_python, but that failed too with an access denied error when trying to write to the redirected AppData folder. I hoped that installing the .msi on a clean install would be a better option.

I did also look to try the portable app version, but the download link points to the .msi version?

### What did you expect to happen?

I hoped that the initial setup would complete for the user

### What actually happened?

Setup failed to complete.

### Operating System Version

Windows 10 v21H2 64-bit

### Mu Version

1.1.1

### Editor Log

2022-07-12 10:54:01,488 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.1
2022-07-12 10:54:01,488 - root:270(run) INFO: uname_result(system='Windows', node='W10-Software', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 79 Stepping 1, GenuineIntel')
2022-07-12 10:54:02,785 - root:271(run) INFO: Platform: Windows-10-10.0.19041-SP0
2022-07-12 10:54:02,785 - root:272(run) INFO: Python path: ['C:\\Program Files (x86)\\Mu Editor\\Python', 'C:\\Program Files (x86)\\Mu Editor\\Python\\python38.zip', 'C:\\Program Files (x86)\\Mu Editor\\Python\\DLLs', 'C:\\Program Files (x86)\\Mu Editor\\Python\\lib', 'C:\\Program Files (x86)\\Mu Editor\\Python', 'C:\\Program Files (x86)\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Program Files (x86)\\Mu Editor\\Python\\lib\\site-packages\\setuptools-57.4.0-py3.8.egg', 'C:\\Program Files (x86)\\Mu Editor\\Python\\lib\\site-packages\\pip-21.1.3-py3.8.egg', 'C:\\Program Files (x86)\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Program Files (x86)\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Program Files (x86)\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin']
2022-07-12 10:54:02,785 - root:273(run) INFO: Language code: en_GB
2022-07-12 10:54:02,785 - mu.settings:220(load) WARNING: No settings file found at C:\Users\testyear7\AppData\Local\python\mu\settings.json; skipping
2022-07-12 10:54:02,785 - mu.settings:220(load) WARNING: No settings file found at C:\Users\testyear7\AppData\Local\python\mu\session.json; skipping
2022-07-12 10:54:03,300 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2022-07-12 10:54:04,816 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2022-07-12 10:54:04,816 - mu.virtual_environment:781(create_venv) INFO: Creating virtualenv: C:\Users\testyear7\AppData\Local\python\mu\mu_venv-38-20220712-105242
2022-07-12 10:54:04,832 - mu.virtual_environment:782(create_venv) INFO: Virtualenv name: mu_venv-38-20220712-105242
2022-07-12 10:54:04,832 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Program Files (x86)\\Mu Editor\\Python\\pythonw.exe', '-I', '-m', 'virtualenv', '-p', 'C:\\Program Files (x86)\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\testyear7\\AppData\\Local\\python\\mu\\mu_venv-38-20220712-105242') with kwargs {'env': {'ALLUSERSPROFILE': 'C:\\ProgramData', 'APPDATA': '\\\\ds-fs01\\home$\\test\\testyear7\\AppData', 'COMMONPROGRAMFILES': 'C:\\Program Files\\Common Files', 'COMMONPROGRAMFILES(X86)': 'C:\\Program Files (x86)\\Common Files', 'COMMONPROGRAMW6432': 'C:\\Program Files\\Common Files', 'COMPUTERNAME': 'W10-SOFTWARE', 'COMSPEC': 'C:\\WINDOWS\\system32\\cmd.exe', 'DRIVERDATA': 'C:\\Windows\\System32\\Drivers\\DriverData', 'FPS_BROWSER_APP_PROFILE_STRING': 'Internet Explorer', 'FPS_BROWSER_USER_PROFILE_STRING': 'Default', 'HOMEDRIVE': 'H:', 'HOMEPATH': '\\', 'HOMESHARE': '\\\\ds-fs01\\home$\\test\\TestYear7', 'LOCALAPPDATA': 'C:\\Users\\testyear7\\AppData\\Local', 'LOGONSERVER': '\\\\DS-DC01', 'NUMBER_OF_PROCESSORS': '4', 'ONEDRIVE': 'C:\\Users\\testyear7\\OneDrive', 'OS': 'Windows_NT', 'PATH': 'C:\\Program Files (x86)\\Mu Editor\\Python\\lib\\site-packages\\PyQt5\\Qt\\bin;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Users\\testyear7\\AppData\\Local\\Microsoft\\WindowsApps;', 'PATHEXT': '.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC', 'PROCESSOR_ARCHITECTURE': 'AMD64', 'PROCESSOR_IDENTIFIER': 'Intel64 Family 6 Model 79 Stepping 1, GenuineIntel', 'PROCESSOR_LEVEL': '6', 'PROCESSOR_REVISION': '4f01', 'PROGRAMDATA': 'C:\\ProgramData', 'PROGRAMFILES': 'C:\\Program Files', 'PROGRAMFILES(X86)': 'C:\\Program Files (x86)', 'PROGRAMW6432': 'C:\\Program Files', 'PSMODULEPATH': 'C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules', 'PUBLIC': 'C:\\Users\\Public', 'SESSIONNAME': 'Console', 'SYSTEMDRIVE': 'C:', 'SYSTEMROOT': 'C:\\WINDOWS', 'TEMP': 'C:\\Users\\TESTYE~1\\AppData\\Local\\Temp', 'TMP': 'C:\\Users\\TESTYE~1\\AppData\\Local\\Temp', 'UATDATA': 'C:\\WINDOWS\\CCM\\UATData\\D9F8C395-CAB8-491d-B8AC-179A1FE1BE77', 'USERDNSDOMAIN': 'DALLAM.LOCAL', 'USERDOMAIN': 'DALLAM', 'USERDOMAIN_ROAMINGPROFILE': 'DALLAM', 'USERNAME': 'testyear7', 'USERPROFILE': 'C:\\Users\\testyear7', 'WINDIR': 'C:\\WINDOWS', 'QT_API': 'pyqt5', 'QT_AUTO_SCREEN_SCALE_FACTOR': '1', 'QT_MAC_WANTS_LAYER': '1'}}
2022-07-12 10:54:28,582 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-07-12 10:54:28,582 - mu.virtual_environment:797(create_venv) INFO: Created virtual environment using C:\Program Files (x86)\Mu Editor\Python\pythonw.exe at C:\Users\testyear7\AppData\Local\python\mu\mu_venv-38-20220712-105242
2022-07-12 10:54:28,582 - mu.virtual_environment:873(install_baseline_packages) INFO: Installing baseline packages.
2022-07-12 10:54:28,597 - mu.virtual_environment:880(install_baseline_packages) INFO: Expecting zipped wheels at C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\wheels\1.1.1.zip
2022-07-12 10:54:29,441 - mu.virtual_environment:855(install_from_zipped_wheels) INFO: About to install from wheel: asttokens-2.0.5-py2.py3-none-any.whl
2022-07-12 10:54:29,441 - mu.virtual_environment:114(run_blocking) INFO: About to run blocking C:\Users\testyear7\AppData\Local\python\mu\mu_venv-38-20220712-105242\scripts\pip.exe with args ['install', '--disable-pip-version-check', '--no-deps', '--no-index', 'C:\\Users\\TESTYE~1\\AppData\\Local\\Temp\\tmp5eyw_3q5\\asttokens-2.0.5-py2.py3-none-any.whl'] and envvars {}
2022-07-12 10:57:29,758 - mu.virtual_environment:164(wait) DEBUG: Finished: False; exitStatus 0; exitCode 0
2022-07-12 10:57:29,758 - mu.virtual_environment:176(wait) ERROR: 
2022-07-12 10:57:30,055 - mu.virtual_environment:643(ensure_and_create) ERROR: Process did not terminate normally:

2022-07-12 10:57:30,070 - mu.virtual_environment:540(quarantine_venv) INFO: Quarantined C:\Users\testyear7\AppData\Local\python\mu\mu_venv-38-20220712-105242 as C:\Users\testyear7\AppData\Local\python\mu\mu_venv-38-20220712-105242.FAILED
2022-07-12 10:57:30,070 - mu.virtual_environment:460(relocate) DEBUG: Relocating to C:\Users\testyear7\AppData\Local\python\mu\mu_venv-38-20220712-105730
2022-07-12 10:57:30,070 - mu.virtual_environment:476(relocate) DEBUG: Virtual environment set up mu_venv-38-20220712-105730 at C:\Users\testyear7\AppData\Local\python\mu\mu_venv-38-20220712-105730
2022-07-12 10:57:30,070 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #2.
2022-07-12 10:57:30,070 - mu.virtual_environment:781(create_venv) INFO: Creating virtualenv: C:\Users\testyear7\AppData\Local\python\mu\mu_venv-38-20220712-105730
2022-07-12 10:57:30,070 - mu.virtual_environment:782(create_venv) INFO: Virtualenv name: mu_venv-38-20220712-105730
2022-07-12 10:57:30,070 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Program Files (x86)\\Mu Editor\\Python\\pythonw.exe', '-I', '-m', 'virtualenv', '-p', 'C:\\Program Files (x86)\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\testyear7\\AppData\\Local\\python\\mu\\mu_venv-38-20220712-105730') with kwargs {'env': {'ALLUSERSPROFILE': 'C:\\Pro [... cut ...]
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 177, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process did not terminate normally:

2022-07-12 11:01:21,065 - mu.settings:169(save) DEBUG: Saving to C:\Users\testyear7\AppData\Local\python\mu\session.json
2022-07-12 11:01:21,081 - mu.settings:169(save) DEBUG: Saving to C:\Users\testyear7\AppData\Local\python\mu\venv.json

tonybaloney, 2022-07-16:
This is a timeout error during the `pip install` process. The error report doesn't say it is a timeout (yet), but the likely cause is a slow network, disk or an interruption on your network connection

Related to #2296 and #2196

altecsole, 2022-08-02:
I did some more testing, including monitoring our web filter to check for any content being blocked. Everything is showing as allowed, so I think the problem must be due to us using redirected App Data and the setup failing to write the the folder. I think we're going to have to leave this, and look at alternatives for September. Many thanks for your help though, it was much appreciated.

PR #2296 (2023-06-24): Double pip timeout each time it times out
Fixes #2196
Fixes #2282
closed 2023-06-24 by tjguk