mu-editor/mu #2285: eh, tried running appimages, then went to compile and run from source....

DrBlackross, 2022-07-14:
### What were you trying to do?

Trying to get Seeed XIAO to blink with latest CircuitPython 7.3.1 (only works with kb version'ish)

### What steps did you take to trigger the issue?

latest appimage 

~/Downloads$ ./Mu_Editor-1.1.1-x86_64.AppImage
No settings file found at /home/drblackross/.local/share/mu/venv.json; skipping
Segmentation fault (core dumped)

then following https://mu.readthedocs.io/en/latest/ decided to clone and compile 

~/Desktop/Mu/mu$ python -m pip install --upgrade pip
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pip in /home/drblackross/.local/lib/python3.10/site-packages (22.1.2)

~/Desktop/Mu/mu$ pip install -e ".[dev]"
ERROR: Package 'mu-editor' requires a different Python: 3.10.4 not in '<3.9,>=3.5'

### What did you expect to happen?

expected it to work

### What actually happened?

i've flashed from 5.0 (stables).uf2 up to adafruit-circuitpython-seeeduino_xiao-en_US-20220714-3515003.uf2.... after version 7.3.0 something changed.
led = DigitalInOut(board.D13) no longer worked, even on adafruits site https://wiki.seeedstudio.com/Seeeduino-XIAO-CircuitPython/ it says D13, then later changes to board.LED (doesn't work).. 
thought it was the way i was writing the python, editor used, etc. so i tried gedit, atom, visualstudio (with circuitpython extension loaded), PyCharm, and figured i goto the source and try to use the Mu-Editor. rewrote the code.py and moved the board.D13 to board.D2 and hooked a scope up to the pad, it works. But, D13 doesn't... LOL, which is the source of my frustration.....

.# load the Led on pin13 for something maybe
led = DigitalInOut(board.D13)
led.direction = Direction.OUTPUT
while len(pulses) == 2:                                               //pulseio is loaded (i think), but even mirroring an input signal on D1 to out on D13 didn't work
	led.value = True
	time.sleep(0.5)
	led.value = False
	time.sleep(0.5)

That's about it, other than the 4 different versions of XIAO in https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/

Seeed_XIAO_nRF52840_Sense
seeeduino_xiao
seeeduino_xiao_kb
seeeduino_xiao_rp2040

i don't know which one to flash other than the luck i had with 5.0 and some seeeduino_xiao_kb versions that would work with Mu-editor or circuitpython in general, but would like to at least running with the current version on the original version of the XIAO.

### Operating System Version

Ubuntu 22.04 LTS

### Mu Version

latest

### Other Info

Python 3.10.4
pip 22.1.2 from /home/drblackross/.local/lib/python3.10/site-packages/pip (python 3.10)

### Editor Log

output from mu.log 

2022-07-14 14:38:28,077 - root:271(run) INFO: 

-----------------

[... 446 lines cut ...]
.     ............................................................................ 82.8/82.8 kB 1.7 MB/s eta 0:00:00
  Downloading Click-7.0-py2.py3-none-any.whl (81 kB)
.     ............................................................................ 81.3/81.3 kB 1.2 MB/s eta 0:00:00
  Downloading click-6.7-py2.py3-none-any.whl (71 kB)
.     ............................................................................ 71.2/71.2 kB 1.7 MB/s eta 0:00:00
  Downloading click-6.6-py2.py3-none-any.whl (71 kB)
.     ............................................................................ 71.0/71.0 kB 943.3 kB/s eta 0:00:00
Collecting black<22.1.0,>=19.10b0
  Downloading black-19.10b0-py36-none-any.whl (97 kB)
.     ............................................................................ 97.5/97.5 kB 2.0 MB/s eta 0:00:00
INFO: pip is looking at multiple versions of appdirs to determine which version is compatible with other requirements. This could take a while.
Collecting appdirs>=1.4.3
  Downloading appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
INFO: pip is looking at multiple versions of <Python from Requires-Python> to determine which version is compatible with other requirements. This could take a while.
INFO: pip is looking at multiple versions of adafruit-board-toolkit to determine which version is compatible with other requirements. This could take a while.
INFO: pip is looking at multiple versions of qtconsole to determine which version is compatible with other requirements. This could take a while.
INFO: pip is looking at multiple versions of qscintilla to determine which version is compatible with other requirements. This could take a while.
INFO: pip is looking at multiple versions of pyqtchart to determine which version is compatible with other requirements. This could take a while.
INFO: pip is looking at multiple versions of pyqt5 to determine which version is compatible with other requirements. This could take a while.
INFO: pip is looking at multiple versions of mu-editor[dev] to determine which version is compatible with other requirements. This could take a while.
ERROR: Package 'mu-editor' requires a different Python: 3.10.4 not in '<3.9,>=3.5'
drblackross@BEAST:~/Desktop/Mu/mu$ python -version
Unknown option: -e
usage: python [option] ... [-c cmd | -m mod | file | -] [arg] ...
Try `python -h' for more information.

tmontes, 2022-07-17:
Hey @DrBlackross, thanks for your report...

We have just updated the docs on running the Mu AppImage file on Ubuntu 22.04...
It boils down to:

Installing a system dependency before hand:
```
sudo apt install libfuse2
```

Launching Mu in the CLI with:
```
QT_QPA_PLATFORM=wayland ./Downloads/Mu_Editor-1.1.1-x86_64.AppImage
```

Please let us know if this helps running Mu on your system.

xuancong84, 2023-04-26:
I also encountered the same segmentation fault. And I tried every `QT_QPA_PLATFORM` option, i.e., eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb. None is working. Either segmentation fault, or plugin not found, or nothing pops up.
```
QT_QPA_PLATFORM=wayland /opt/MuEditor
Failed to create wl_display (No such file or directory)
qt.qpa.plugin: Could not load the Qt platform plugin "wayland" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted
```

My system is MX-Linux running KDE plasma desktop. When I `apt-get install mu-editor`, it can launch, but it does not support micropython plugin. So it is quite broken.