mu-editor/mu #2291: Utils: Create a script to convert Python stub files into Qscintilla autocompletion files

carlosperate, 2022-07-16:


Jeffrey04, 2022-07-17:
so the goal is to turn
```
def panic() -> None:
    """
    Put micro:bit in panic() mode and display an unhappy face.
    Press the reset button to exit panic() mode.
    """
    pass
```

into the following?

```
_("panic() \nPut micro:bit in panic() mode and display an unhappy face. Press the reset button to exit panic() mode.")
```

What about instance methods?

```
class _Accelerometer(object):
    def get_x(self) -> int:
        """
        Return micro:bit's tilt (X acceleration) in milli-g's.
        """
        pass
```

carlosperate, 2022-07-17:
> so the goal is to turn
>
> ```
> def panic() -> None:
>     """
>     Put micro:bit in panic() mode and display an unhappy face.
>     Press the reset button to exit panic() mode.
>     """
>     pass
> ```
>
> into [...]

Yes 👍 

> What about instance methods?
> 
> ```
> class _Accelerometer(object):
>     def get_x(self) -> int:
>         """
>         Return micro:bit's tilt (X acceleration) in milli-g's.
>         """
>         pass
> ```

For the micro:bit API, classes for built in board features have a single instance by default. So the user cannot initialise a new instance of the micro:bit accelerometer class. In this case it looks like the API just follows a similar pattern a module functions:
https://github.com/mu-editor/mu/blob/9fe47acf97096649d94cd26b379247f73e74d6ee/mu/modes/api/microbit.py#L76

For thinsg like MicroPython PyBoard I2C calsses (which you can initilisate) it looks very similar:
https://github.com/mu-editor/mu/blob/9fe47acf97096649d94cd26b379247f73e74d6ee/mu/modes/api/pyboard.py#L175

Jeffrey04, 2022-07-17:
let me attempt to submit a PR then