mu-editor/mu #2293: Windows Store Python causes

tonybaloney, 2022-07-16:
### What were you trying to do?

Launch from a virtual environment

### What steps did you take to trigger the issue?

1. Install Python 3.8 from the Windows store
2. Create a virtual environment
3. Start mu

### Operating System Version

Windows 11

### Mu Version

main (1.1.1)

tonybaloney, 2022-07-16:
Related: https://github.com/platformdirs/platformdirs/issues/90

Jeffrey04, 2022-07-17:
I also needed to do this to get mu-editor running with windows store python
https://docs.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=cmd#enable-long-paths-in-windows-10-version-1607-and-later