mu-editor/mu #2295: Appimage does not run on Arch Linux (Address boundary error)

hoh, 2022-07-16:
### What were you trying to do?

Run Mu_Editor-1.1.1-x86_64.AppImage on a laptop running Arch Linux

### What steps did you take to trigger the issue?

1. Download Mu_Editor-1.1.1-x86_64.AppImage the Download link on https://codewith.mu/en/download (https://github.com/mu-editor/mu/releases/download/v1.1.1/Mu_Editor-1.1.1-x86_64.AppImage)
2. Make the file executable with `chmod +x Mu_Editor-1.1.1-x86_64.AppImage`
3. Run Mu from a terminal using `./Mu_Editor-1.1.1-x86_64.AppImage`
4. Get the following error: `/tmp/.mount_Mu_EdiWDgDGs/usr/bin/python3.8: error while loading shared libraries: libcrypt.so.1: cannot open shared object file: No such file or directory`
5. Install the missing library using `sudo pacman -S libxcrypt-compat`
6. Run Mu from a terminal using `./Mu_Editor-1.1.1-x86_64.AppImage`

### What did you expect to happen?

Mu opens and I can start mu-ing

### What actually happened?

The following error message appeared in the terminal:
```
No settings file found at /home/myself/.local/share/mu/venv.json; skipping
fish: Job 1, './Mu_Editor-1.1.1-x86_64.AppIma…' terminated by signal SIGSEGV (Address boundary error)
```

When looking at `dmesg`, I read the following error message:
```
[140410.054283] python3.8[134164]: segfault at 52b9 ip 00007f3913ee5ed8 sp 00007fff723f8fa0 error 4 in libX11.so.6.4.0[7f3913ed9000+8b000]
[140410.054290] Code: e8 bd f8 ff ff 48 8b 9d 48 12 00 00 48 85 db 74 29 ff 15 cb 5b 11 00 eb 11 66 0f 1f 84 00 00 00 00 00 48 8b 1b 48 85 db 74 10 <48> 39 43 08 75 f2 48 83 c4 08 5b 5d c3 0f 1f 00 48 8b 85 30 0a 00
[140410.054432] audit: type=1701 audit(1657982545.600:1232): auid=1000 uid=1000 gid=1000 ses=11 pid=134164 comm="python3.8" exe="/tmp/.mount_Mu_EdixuOhZu/usr/bin/python3.8" sig=11 res=1
[140410.057756] audit: type=1334 audit(1657982545.603:1233): prog-id=136 op=LOAD
[140410.057821] audit: type=1334 audit(1657982545.603:1234): prog-id=137 op=LOAD
[140410.057824] audit: type=1334 audit(1657982545.603:1235): prog-id=138 op=LOAD
[140410.058588] audit: type=1130 audit(1657982545.606:1236): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@21-134173-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[140410.278299] audit: type=1131 audit(1657982545.823:1237): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@21-134173-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[140410.415272] audit: type=1334 audit(1657982545.963:1238): prog-id=0 op=UNLOAD
[140410.415283] audit: type=1334 audit(1657982545.963:1239): prog-id=0 op=UNLOAD
[140410.415287] audit: type=1334 audit(1657982545.963:1240): prog-id=0 op=UNLOAD
```

This message includes `segfault at (...) in libX11.so.6.4.0`

### Operating System Version

Arch Linux, updated as of 2022-07-16

### Mu Version

1.1.1

### Other Info

The system has an NVIDIA graphic card. 
Hardware: XPS 15 9520 laptop with a i7-12700H and NVIDIA GeForce RTX 3050 Ti.

hoh, 2022-07-16:
@tmontes this is the issue I showed you during the EP sprints.

tmontes, 2022-07-16:
(linking with #2235 that seems to report a similar case under Arch Linux)

carlosperate, 2022-08-16:
Are you using Wayland? Does the X11 seg fault happens when running the appimage with this env variable set?
```
QT_QPA_PLATFORM=wayland Mu_Editor-1.1.1-x86_64.AppImage
```

Or when using the AppImage from this PR build?
https://github.com/mu-editor/mu/actions/runs/2699412265

hoh, 2022-08-16:
I was using Xorg. 

I tested again today: both Appimage (1.1.1 and run [2699412265](https://github.com/mu-editor/mu/actions/runs/2699412265)) work on Wayland and crash on Xorg.

hoh, 2022-08-16:
## System:

KDE Plasma Version 5.25.4
KDE Frameworks Version 5.97.0
Qt Version 5.15.5
Kernel 5.19.1-arch2-1 (64-bit)
Graphics Platform X11
Intel COre i7-12700H
Mesa Intel Graphics
X.Org versiom 21.1.4

carlosperate, 2022-08-16:
Ah, interesting, thanks for the test and the additional information!

I'm not too familiar with debugging linux applications, so I don't really know if this will be useful at all, but would it be worth running ldd to see what it shows?

On Ubuntu 22.04 this is what I get:

```
$ ./Mu_Editor-1.1.1-x86_64.AppImage --appimage-extract
...

$ ldd squashfs-root/usr/bin/python3.8 
	linux-vdso.so.1 (0x00007ffef10c6000)
	/home/carlos/Downloads/mu-test/Mu_Editor-AppImage-x86_64-d7fec50815fa1a68e484e5343572b27a80d359ed/squashfs-root/usr/bin/../lib/libpython3.8.so.1.0 (0x00007fc45b297000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007fc45b24e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fc45b249000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fc45b244000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007fc45b23f000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fc45b156000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fc45b151000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fc45af29000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fc45c664000)
```

hoh, 2022-08-17:
```
user@desktop-pc /t/muu> ldd squashfs-root/usr/bin/python3.8 

        linux-vdso.so.1 => linux-vdso.so.1 (0x00007ffc922f9000)
        /tmp/muu/squashfs-root/usr/bin/../lib/libpython3.8.so.1.0 => /tmp/muu/squashfs-root/usr/bin/../lib/libpython3.8.so.1.0 (0x00007f6248b78000)
        libcrypt.so.1 => /usr/lib/libcrypt.so.1 (0x00007f6248b23000)
        libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f6248b1e000)
        libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f6248b19000)
        libutil.so.1 => /usr/lib/libutil.so.1 (0x00007f6248b14000)
        libm.so.6 => /usr/lib/libm.so.6 (0x00007f6248a2a000)
        librt.so.1 => /usr/lib/librt.so.1 (0x00007f6248a25000)
        libc.so.6 => /usr/lib/libc.so.6 (0x00007f624883e000)
        /lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007f6249f45000)

user@desktop-pc /t/muu> readelf --dynamic squashfs-root/usr/bin/python3.8 

Dynamic section at offset 0x3da8 contains 32 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [$ORIGIN/../lib/libpython3.8.so.1.0]
 0x0000000000000001 (NEEDED)             Shared library: [libcrypt.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libpthread.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libdl.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libutil.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [librt.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000c (INIT)               0x401000
 0x000000000000000d (FINI)               0x4011b4
 0x0000000000000019 (INIT_ARRAY)         0x403d98
 0x000000000000001b (INIT_ARRAYSZ)       8 (bytes)
 0x000000000000001a (FINI_ARRAY)         0x403da0
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x0000000000000004 (HASH)               0x3ff5e0
 0x000000006ffffef5 (GNU_HASH)           0x3ff580
 0x0000000000000005 (STRTAB)             0x3ff2e0
 0x0000000000000006 (SYMTAB)             0x3ff400
 0x000000000000000a (STRSZ)              283 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000015 (DEBUG)              0x0
 0x0000000000000003 (PLTGOT)             0x404000
 0x0000000000000002 (PLTRELSZ)           48 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0x400670
 0x0000000000000007 (RELA)               0x400658
 0x0000000000000008 (RELASZ)             24 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000006ffffffe (VERNEED)            0x400638
 0x000000006fffffff (VERNEEDNUM)         1
 0x000000006ffffff0 (VERSYM)             0x400618
 0x0000000000000000 (NULL)               0x0
```

carlosperate, 2022-11-16:
Hi @hoh,

I don't think this is the exact same issue as https://github.com/mu-editor/mu/issues/2318, but might be related to the packaged Python version, so maybe the same fix would apply.

Would you be able to test if this CI AppImage works on your environment? It'd be really appreciated.

You'll need to be logged into GitHub to be able to click on the download link from this page:
https://github.com/mu-editor/mu/actions/runs/3479069676
[image]

hoh, 2022-11-23:
Hi @carlosperate ,

That version works on my Arch Linux environment.

It would be difficult to know if the fix of #2318 or updates on Arch (rolling release) fixed the issue, but it works 😊.

hoh, 2022-11-23:
Thanks for keeping track and for the update. I believe we can now close this issue.

closed 2022-11-23 by hoh