mu-editor/mu #2303: Cannot run mu-editor developer version on macOS

prcutler, 2022-07-17:
### What were you trying to do?

Build and run the development version of Mu

### What steps did you take to trigger the issue?

I am following the Developer quickstart:  https://mu.readthedocs.io/en/latest/

I understand this may be an issue with PyQt5, but I wanted to share it in case anyone else is trying to build it on macOS.

I have a Python 3.8.9 virtual environment and I was able to clone and build Mu successfully:

`pip install -e ".[dev]"` results in `Successfully installed mu-editor-1.1.1`

But trying to run it results in:
```
mu-editor
Traceback (most recent call last):
  File "/Users/prcutler/workspace/mu/.venv/bin/mu-editor", line 33, in <module>
    sys.exit(load_entry_point('mu-editor', 'console_scripts', 'mu-editor')())
  File "/Users/prcutler/workspace/mu/.venv/bin/mu-editor", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/prcutler/workspace/mu/mu/app.py", line 33, in <module>
    from PyQt5.QtCore import (
ModuleNotFoundError: No module named 'PyQt5'
```

I am unable to build PyQT5 on macOS:

```
pip install pyqt5  
Collecting pyqt5
  Using cached PyQt5-5.15.7.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [29 lines of output]
      Traceback (most recent call last):
        File "/Users/prcutler/workspace/mu/.venv/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 156, in prepare_metadata_for_build_wheel
          hook = backend.prepare_metadata_for_build_wheel
      AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_for_build_wheel'
      
      During handling of the above exception, another exception occurred:
      
      Traceback (most recent call last):
        File "/Users/prcutler/workspace/mu/.venv/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/Users/prcutler/workspace/mu/.venv/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/Users/prcutler/workspace/mu/.venv/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 160, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
        File "/private/var/folders/nc/wv1g_z612glftnlh2hfqr9780000gn/T/pip-build-env-u3tg0gy2/overlay/lib/python3.8/site-packages/sipbuild/api.py", line 46, in build_wheel
          project = AbstractProject.bootstrap('wheel',
        File "/private/var/folders/nc/wv1g_z612glftnlh2hfqr9780000gn/T/pip-build-env-u3tg0gy2/overlay/lib/python3.8/site-packages/sipbuild/abstract_project.py", line 87, in bootstrap
          project.setup(pyproject, tool, tool_description)
        File "/private/var/folders/nc/wv1g_z612glftnlh2hfqr9780000gn/T/pip-build-env-u3tg0gy2/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 584, in setup
          self.apply_user_defaults(tool)
        File "project.py", line 69, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/nc/wv1g_z612glftnlh2hfqr9780000gn/T/pip-build-env-u3tg0gy2/overlay/lib/python3.8/site-packages/pyqtbuild/project.py", line 70, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/nc/wv1g_z612glftnlh2hfqr9780000gn/T/pip-build-env-u3tg0gy2/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 236, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        File "/private/var/folders/nc/wv1g_z612glftnlh2hfqr9780000gn/T/pip-build-env-u3tg0gy2/overlay/lib/python3.8/site-packages/pyqtbuild/builder.py", line 67, in apply_user_defaults
          raise PyProjectOptionException('qmake',
      sipbuild.pyproject.PyProjectOptionException
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

### What did you expect to happen?

The development version of Mu to start

### What actually happened?

See Steps above for error details

### Operating System Version

macOS (M1 / Arm)

### Mu Version

Git 

### Other Info

Log files not available.  If I'm doing something wrong, please let me know.  I'd like to run the development version to help contribute some docs.  Thanks!

carlosperate, 2022-07-17:
Hi @prcutler,
The version of PyQt5 Mu uses doesn't have wheels for Arm, so you'd have to use rosetta for the moment.

Soon we'll realease v1.2.0 and right after that we'll merge PR https://github.com/mu-editor/mu/pull/2279, which should get us much closer (if not completely over the line) to be able to run Mu natively in M1/2.

prcutler, 2022-07-17:
Ok, thank you for letting me know

closed 2022-07-17 by prcutler