mu-editor/mu #2309: New AppImage docker image will be needed for PyQt6

carlosperate, 2022-07-18:


carlosperate, 2022-07-19:
Right, so the Qt binaries from the riverbank wheels come from Qt directly: https://doc.qt.io/qt-6/supported-platforms.html

That means the glibc 2.28 limitation is from there, not the wheel generation.

@tmontes I'll need to update the AppImage docker image to Ubunut 20.04, or I could use Debina 10 Buster, which came out a year earlier so that might be better? Do you have any thoughts or preferences?

tmontes, 2022-07-19:
(then again, if Qt says "Ubuntu 20.04" I wonder if we'll hit any bumps with Debian!... but I'd start with Debian, yes)

carlosperate, 2022-07-19:
My assumption is that it says Ubuntu 20.04 because that's the oldest LTS with the right glibc, let's hope there isn't another reason.

Debian Buster sounds good, as it has 2.28 exactly 👍