mu-editor/mu #2312: Raspberry Pi 400 "Could not find an attached device"

rtyley, 2022-07-19:
### What were you trying to do?

Use Mu on a Raspberry Pi 400 to read the Serial out of a rp2040 CircuitPython device, wanted to see the Serial output

### What steps did you take to trigger the issue?

Press 'Serial' menu button

### What did you expect to happen?

Expected Mu's built in serial monitor to operate

### What actually happened?

Mu gave a "Could not find an attached device" error - the video below shows the issue being recreated. Note that `screen /dev/ttyACM0 115200` worked, and was able to display serial output - but Mu's in-built Serial view failed:

https://user-images.githubusercontent.com/52038/179762346-579a049a-fb49-465a-b4dd-ccfa38de4d89.mov

### Operating System Version

Raspberry Pi OS (raspbian) 11 (bullseye)

### Mu Version

1.0.3

### Other Info

Tested with 2 different CircuitPython devices:

* Keybow 2040 running CircuitPython 7.3.1
* Pimoroni Pico LiPo 16MB

Both devices are read ok with Mu running on my Ubuntu laptop and also on my Mac OS M1 Mac.

This issue seems similar/related to https://github.com/mu-editor/mu/issues/1596 / https://github.com/mu-editor/mu/issues/1543, though to be clear, I'm encountering the problem on a Raspberry Pi 400, rather than a M1 Mac

### Editor Log

2022-07-19 12:04:50,402 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.3

2022-07-19 12:04:50,403 - root:113(run) INFO: uname_result(system='Linux', node='raspberrypi', release='5.15.32-v7l+', version='#1538 SMP Thu Mar 31 19:39:41 BST 2022', machine='armv7l')

2022-07-19 12:04:50,403 - root:114(run) INFO: Python path: ['/usr/share/mu-editor', '/usr/lib/python39.zip', '/usr/lib/python3.9', '/usr/lib/python3.9/lib-dynload', '/usr/local/lib/python3.9/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3.9/dist-packages', '/usr/lib/python3/dist-packages/IPython/extensions']

2022-07-19 12:04:50,403 - root:115(run) INFO: Language code: en_GB

2022-07-19 12:04:50,670 - mu.logic:543(__init__) INFO: Setting up editor.

2022-07-19 12:04:50,670 - mu.logic:560(__init__) DEBUG: Creating directory: /home/adorable/.local/share/mu

2022-07-19 12:04:50,671 - mu.logic:315(get_admin_file_path) DEBUG: Creating admin file: /home/adorable/.local/share/mu/settings.json

2022-07-19 12:04:50,671 - mu.logic:562(__init__) INFO: Settings path: /home/adorable/.local/share/mu/settings.json

2022-07-19 12:04:50,671 - mu.logic:315(get_admin_file_path) DEBUG: Creating admin file: /home/adorable/.local/share/mu/session.json

2022-07-19 12:04:50,672 - mu.logic:563(__init__) INFO: Session path: /home/adorable/.local/share/mu/session.json

2022-07-19 12:04:50,672 - mu.logic:564(__init__) INFO: Log directory: /home/adorable/.cache/mu/log

2022-07-19 12:04:50,672 - mu.logic:565(__init__) INFO: Data directory: /home/adorable/.local/share/mu

2022-07-19 12:04:50,705 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero

2022-07-19 12:04:50,706 - mu.logic:582(setup) DEBUG: Creating directory: /home/adorable/mu_code

2022-07-19 12:04:50,706 - mu.logic:590(setup) DEBUG: Creating directory: /home/adorable/mu_code/images

2022-07-19 12:04:50,707 - mu.logic:597(setup) DEBUG: Creating directory: /home/adorable/mu_code/fonts

2022-07-19 12:04:50,708 - mu.logic:600(setup) DEBUG: Creating directory: /home/adorable/mu_code/sounds

2022-07-19 12:04:50,708 - mu.logic:603(setup) DEBUG: Creating directory: /home/adorable/mu_code/music

2022-07-19 12:04:51,401 - mu.logic:1147(change_mode) INFO: Workspace directory: /home/adorable/mu_code

2022-07-19 12:04:51,404 - mu.logic:625(restore_session) INFO: Restoring session from: /home/adorable/.local/share/mu/session.json

2022-07-19 12:04:51,404 - mu.logic:626(restore_session) DEBUG: {}

2022-07-19 12:04:51,404 - mu.logic:1104(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']

2022-07-19 12:04:55,314 - mu.logic:1110(select_mode) INFO: New mode selected: adafruit

2022-07-19 12:04:55,441 - mu.interface.main:723(show_message) DEBUG: Could not find an attached Adafruit CircuitPython device.

2022-07-19 12:04:55,442 - mu.interface.main:724(show_message) DEBUG: Python files for Adafruit CircuitPython devices are stored on the device. Therefore, to edit these files you need to have the device plugged in. Until you plug in a device, Mu will use the directory found here:

 /home/adorable/mu_code

...to store your code.

2022-07-19 12:04:58,919 - mu.logic:1147(change_mode) INFO: Workspace directory: /home/adorable/mu_code

2022-07-19 12:04:58,984 - mu.logic:678(restore_session) INFO: Starting with blank file.

2022-07-19 12:04:59,077 - mu.logic:1147(change_mode) INFO: Workspace directory: /home/adorable/mu_code

2022-07-19 12:05:02,969 - mu.modes.base:231(find_device) WARNING: Could not find device.

2022-07-19 12:05:02,970 - mu.modes.base:232(find_device) DEBUG: Available ports:

2022-07-19 12:05:02,970 - mu.modes.base:233(find_device) DEBUG: ['PID:0x08c6 VID:0x16d0 PORT:ttyACM0', 'PID:0x0000 VID:0x0000 PORT:ttyAMA0']

2022-07-19 12:05:02,972 - mu.interface.main:723(show_message) DEBUG: Could not find an attached device.

2022-07-19 12:05:02,974 - mu.interface.main:724(show_message) DEBUG: Please make sure the device is plugged into this computer.

It must have a version of MicroPython (or CircuitPython) flashed onto it before the REPL will work.

Finally, press the device's reset button and wait a few seconds before trying again.

2022-07-19 12:05:06,484 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.

2022-07-19 12:07:50,059 - mu.modes.base:231(find_device) WARNING: Could not find device.

2022-07-19 12:07:50,059 - mu.modes.base:232(find_device) DEBUG: Available ports:

2022-07-19 12:07:50,059 - mu.modes.base:233(find_device) DEBUG: ['PID:0x08c6 VID:0x16d0 PORT:ttyACM0', 'PID:0x0000 VID:0x0000 PORT:ttyAMA0']

2022-07-19 12:07:50,062 - mu.interface.main:723(show_message) DEBUG: Could not find an attached device.

2022-07-19 12:07:50,062 - mu.interface.main:724(show_message) DEBUG: Please make sure the device is plugged into this computer.

It must have a version of MicroPython (or CircuitPython) flashed onto it before the REPL will work.

Finally, press the device's reset button and wait a few seconds before trying again.

2022-07-19 12:07:52,401 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.

2022-07-19 12:08:20,621 - mu.logic:1075(show_admin) INFO: Showing logs from /home/adorable/.cache/mu/log/mu.log

2022-07-19 12:14:33,674 - mu.modes.base:231(find_device) WARNING: Could not find device.

2022-07-19 12:14:33,676 - mu.modes.base:232(find_device) DEBUG: Available ports:

2022-07-19 12:14:33,676 - mu.modes.base:233(find_device) DEBUG: ['PID:0x08c6 VID:0x16d0 PORT:ttyACM0', 'PID:0x0000 VID:0x0000 PORT:ttyAMA0']

2022-07-19 12:14:33,682 - mu.interface.main:723(show_message) DEBUG: Could not find an attached device.

2022-07-19 12:14:33,683 - mu.interface.main:724(show_message) DEBUG: Please make sure the device is plugged into this computer.

It must have a version of MicroPython (or CircuitPython) flashed onto it before the REPL will work.

Finally, press the device's reset button and wait a few seconds before trying again.

2022-07-19 12:14:38,932 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.

2022-07-19 12:14:41,503 - mu.modes.base:231(find_device) WARNING: Could not find device.

2022-07-19 12:14:41,503 - mu.modes.base:232(find_device) DEBUG: Available ports:

2022-07-19 12:14:41,504 - mu.modes.base:233(find_device) DEBUG: ['PID:0x08c6 VID:0x16d0 PORT:ttyACM0', 'PID:0x0000 VID:0x0000 PORT:ttyAMA0']

2022-07-19 12:14:41,506 - mu.interface.main:723(show_message) DEBUG: Could not find an attached device.

2022-07-19 12:14:41,506 - mu.interface.main:724(show_message) DEBUG: Please make sure the device is plugged into this computer.

It must have a version of MicroPython (or CircuitPython) flashed onto it before the REPL will work.

Finally, press the device's reset button and wait a few seconds before trying again.

2022-07-19 12:14:42,626 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.

2022-07-19 14:06:34,615 - mu.logic:1075(show_admin) INFO: Showing logs from /home/adorable/.cache/mu/log/mu.log

carlosperate, 2022-07-19:
You are right, the main issue is that debian package is at v1.0.3, which is very old and predates the RP2040 devices, so Mu v1.03, which at the time used USB VID and PIDs, doesn't have the values for this devices in the source code.

Unfortunately due to the current dependencies Mu uses it is not compatible with Python 3.8, so there is no simple way to install it in Raspbian Bullseye, which comes with Python 3.9.

That being said, the main limiting factor for the Python 3.8 max version is PyQt, and to install Mu from source in PiOS you need to use the PyQt5 packages from apt, so you might be able to get it working by following the dev instructions but using the `--ignore-requires-python` flag for the pip install step:
https://mu.readthedocs.io/en/latest/setup.html#raspberry-pi

```
pip install -e ".[dev]" --ignore-requires-python
```

xav-e, 2024-04-07:
> You are right, the main issue is that debian package is at v1.0.3, which is very old and predates the RP2040 devices, so Mu v1.03, which at the time used USB VID and PIDs, doesn't have the values [...]

Hello amigo, I tried this solution but it fails to run the run.py file. I Have collected the log file for the installation process as well as the error message which is the following
`aqua-pi@raspberrypi:~/mu-source $ python run.py
Traceback (most recent call last):
  File "/home/aqua-pi/mu-source/run.py", line 2, in <module>
    from mu.app import run
  File "/home/aqua-pi/mu-source/mu/app.py", line 45, in <module>
    from .virtual_environment import venv, logger as vlogger
  File "/home/aqua-pi/mu-source/mu/virtual_environment.py", line 26, in <module>
    from . import settings
  File "/home/aqua-pi/mu-source/mu/settings.py", line 19, in <module>
    from . import config
  File "/home/aqua-pi/mu-source/mu/config.py", line 3, in <module>
    import platformdirs
ModuleNotFoundError: No module named 'platformdirs'
aqua-pi@raspberrypi:~/mu-source $ `
[mu_editor_installation-log.txt](https://github.com/mu-editor/mu/files/14898198/mu_editor_installation-log.txt)
thanks for your help

earlthesquirrel, 2024-04-25:
Question :  I'm encountering a similar problem on a regular raspberry pi 4.  Latest OS, so past 3.8.  But have Pyenv installed and was able to install Mu-editor via the Raspberry pi Preferences->Recommended Software.  

That installs a version of Mu-editor that runs and can load code from the device that shows up as a CircuitPy device.  

However, I get an error when trying to run serial.   Could not find attached device.     Is there a work around for this yet?And any ETA when Mu will get caught up with more recent Python versions?

MohitBallikar, 2024-06-09:
Hail Mary here, but I'm trying to get Mu to work with a Raspi4 and a PN532 breakout board and am facing a similar issue that OP had. Any fixes or updates on this yet?

PaulskPt, 2024-07-04:
I encounter the same problem on a Raspberry Pi 5B-8GB. Raspberry PI OS: Linux raspi5B-8GB 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64. Mu editor version 1.0.3. Also installed via the Raspberry Pi Preferences->Recommended Software. The device I have connected via USB is an Espressif ESP-BOX S3. It's serial connection is via /dev/ttyACM0. 
Mu editor sees the Circuitpython drive: /media/<user>/CIRCUITPY/
The Logfile of Mu Editor shows this serial device:
2024-07-03 18:04:12,659 - mu.modes.base:233(find_device) DEBUG: ['PID:0x0000 VID:0x0000 PORT:ttyAMA0', 'PID:0x7005 VID:0x303a PORT:ttyACM0', 'PID:0x0000 VID:0x0000 PORT:ttyAMA10'],
however when I press the Serial button, a pop-up window opens with the error: "Could not find an attached device".
I, am member of the group DIALOUT.