mu-editor/mu #2313: ESP32-S3 "waiting for download" after hard reset

hyansuper, 2022-07-19:
### What were you trying to do?

hard reset and back into micropython REPL

### What steps did you take to trigger the issue?

just press reset button on a ESP32-S3.

### What did you expect to happen?

hard reset and back into micropython REPL

### What actually happened?

in the REPL window says "waiting for download"

### Operating System Version

win10

### Mu Version

1.1.1

### Other Info

`idf.py -p [PORT] monitor` works fine after hard reset, althongh the monitor only shows output and can't handle input command.

Another editor called Thonny also can go into micropython REPL after hard reset (without GPIO0 connected to pullup or capacitor).

But on Mu editor, soft reset `machine.reset()` is fine, but not hard reset. I tried GPIO0 connected to a 10k pullup to 3.3v, or add a 0.1uF cap to ground, also tried different combinations (with/without pullup/cap), upon hard reset, the REPL says "waiting for download"

I don't expect this be fixed soon, as it can be related to hardware

[image]
[image]

### Editor Log

MicroPython 364569d-dirty on 2022-06-03; ESP32S3 module (spiram) with ESP32S3
Type "help()" for more information.
>>> 
>>> # now I press reset button
>>> ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download

carlosperate, 2022-07-19:
Is that a message from the bootloader waiting for a tool to send something? In that case how do other tools avoid it or workaround it? I assume the bootloader will just wait if a serial connection is opened while it boots up?

greenoaktree, 2022-12-01:
Same problem on my esp32s3，but esptool write the single file firmware bin file is OK.Wired....

michelegirolami, 2023-11-29:
the same for me. @mrx23dot mentioned to Release GPIO0 from GND. Can you explain more in detail this step? Please .

sonpython, 2024-01-02:
same here

```
rst:0x1 (POWERON_RESET),boot:0x1 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_REO_V2))
waiting for download
```

I found this [`note-on-c15`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html#note-on-c15) from espressif docs: 

```
The component C15 may cause the following issues on earlier ESP32-DevKitC V4 boards:

The board may boot into Download mode

If you output clock on GPIO0, C15 may impact the signal

In case these issues occur, please remove the component. The figure below shows the location of C15 highlighted in yellow.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/_images/esp32-devkitc-c15-location.png
```

michelegirolami, 2024-01-03:
@sonpython  after many tries, I found this procedure to successfully flash a fw into my ESP32 (lora 32 S3):
- plugging the device
- open Arduino IDE
- Select the correct board. In my case ESP32S3
- From the "Tools" menu, select "USB CDC On Boot" to ENABLED
- Flash the fw
as shown with the attached picture... hope this help
[image]

sonpython, 2024-01-03:
I dont see that menu

[image]

fumikrobot, 2024-02-14:
> * gging the device
> * open Arduino IDE
> * Select the correct board. In my case ESP32S3
> * From the "Tools" menu, select "USB CDC On Boot" to ENABLED
> * Flash the fw

Perfect, i can run it after facing same issue. Thank you very much.

iolo, 2024-04-08:
press BOOT whiling pressing RST.

teleger, 2024-04-09:
I also encountered a similar problem, my board is esp32s3, I don't think there is a hardware problem, because when I press the boot key and press the rst button again, I can indicate the existence of serial device, so I found that there is a problem with my burning program, I found that I used the port GPIO19. I checked the board of esp32s3, at least connected to the esp32h2 module on my board, GPIO19 is used as USB, the program I burned occupied the port of GPIO19, I erased all again. The erase instruction is idf.py -p COMx erase-flash, and I re-burned my updated program again with no problem

Krish7980, 2024-12-25:
See guys i know this issue isvery frustrating:
But here is the magic bullet for you:
For ESP32S3 Pico connect the 3V3,GND To positve and negative rails of bredboard AND THE "EN" To GROUND.This will reset the chip and solve your all issues.Guys this is one time solution.Keep the baud rate 115200 or what your port actual board raute is.
You can check your com port baud rate in the device manager windows.This problem is so frustrating but yet so simple.
I doont understand why there are such complicated datasheeets that cant solve our one isssue.
Also you dont need to do everytime it is one time solution...Byeeeeeee.See your output and serial monitor..tell me what is the thig that comes in your pc....bye bye

smithwinston, 2025-01-22:
@Krish7980 -- Thank you!  Your suggestion of shorting EN to GND got me out of this infuriating loop, I had tried pretty much everything.  I still don't know what the cause was, I'm using a Seeed Studio XIAO ESP32C6 (it has the EN and GND on the bottom of the PCB as pads).  I thought maybe the flash was corrupt (I'm working on OTA updates), but I tried erasing it using `espflash erase-flash` as well as loading a stock "hello world" app from the esp-idf-template, but it didn't help, only your suggestion has worked so far!  Thanks again!

antIggl, 2025-04-12:
> See guys i know this issue isvery frustrating:(FOR ARDUINO IDE) But here is the magic bullet for you: For ESP32S3 Pico connect the 3V3,GND To positve and negative rails of bredboard AND THE "EN" To GROUND.This will reset the chip and solve your all issues.Guys this is one time solution.Keep the baud rate 115200 or what your port actual board raute is. You can check your com port baud rate in the device manager windows.This problem is so frustrating but yet so simple. I doont understand why there are such complicated datasheeets that cant solve our one isssue. Also you dont need to do everytime it is one time solution...Byeeeeeee.See your output and serial monitor..tell me what is the thig that comes in your pc....bye bye

Hi.
Thank you for the response.
I have the same issue, but I got confused reading your response.
According to the datasheet, EN pin should be HIGH for the core to be enabled. https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf
Where do you support your response? Can you elaborate more on this.? 
I have designed another board with the same module and never had an issue. On the board I am working now, I have. It's random and very very frustrating.

Krish7980, 2025-04-12:
Hello dear..
just to clarify..i used an ESP32S3 PICO by waveshare and this module was
showing like i would not say random behaviour but hmm like it was showing
values "0xfffffff" continuously in loop in serial monitor of Arduino IDE.SO
i tried a lot pressed reset but still it was showing that...and i guess it
was the issue of bootloader.So yes when i tried to pull EN pin tO GND..it
somehow worked then..but at the time of excitement i did not researched
about it much..and as usal asked ChatGPT at that time and yes as usual it
told me that you are correct.(LOL).So i did not thought much
about that.Today i researched about it on Gemmini and found this:"*Based on
the ESP32-S3-WROOM-1/1U datasheet you provided:*

*No, connecting the EN (Enable) pin to GND (Ground) does NOT perform
a factory reset.*

*Here's what the datasheet indicates about the EN pin:*

   1.

   *Pin Definition (Page 11, Table 3):*
   -

      *Name: EN*
      -

      *No.: 3*
      -

      *Type: I (Input)*
      -

      *Function: "High: on, enables the chip. Low: off, the chip powers
      off. Note: Do not leave the EN pin floating."*
      2.

   *Reset Functionality (Implied on Page 42, Figure 7):*
   -

      *The typical application circuit shows an optional reset button (SW1)
      connected between the EN pin and GND.*
      -

      *Pressing this button pulls the EN pin LOW.*

*Conclusion:*

   -

   *Pulling the EN pin LOW (connecting it to GND) acts as a hardware
   reset or chip disable. It forces the chip to stop execution and restart its
   boot sequence when released (brought back HIGH).*
   -

   *It does not erase the flashed firmware or restore any configuration
   settings to their original factory defaults. That would be a "factory
   reset," which is typically done via software commands or sometimes specific
   GPIO combinations during boot, not simply by toggling the EN pin.*

*Therefore, connecting EN to GND simply resets the chip."-This is the
response from AI...But still i have one dobut why would not pressing the
reset button resolved this issue..i like why it made me to put EN pin to
ground...that i am still wondering..So i hope i could clarigy oyur
response...and yes one more thing i am junior tryinng to learn
electroncis ai in deep level as possible go rabbit holes so please can we
be freinds and supports each other...Bye!!!!!!!-Signig OFF-Krish here from
INDIA*