mu-editor/mu #2315: Mu 1.1.1 on macOS 12.4 (M1):  min window height & width seems to be 50% of monitor. Way too wide on 49" wide screen monitor.

haikusw, 2022-07-21:
### What were you trying to do?

Run Mu for first time.

### What steps did you take to trigger the issue?

Launch Mu and it opens a window that is way way way wider than it needs to be.
Tried to resize it smaller - no can do!

Play with resizing height and width and determine that it appears the minimum width and height were set to be 50% of the screen height and width (a pretty bad assumption) instead of basing it on the window content (toolbar items or whatever).

### What did you expect to happen?

I should be able to resize the window down to as small as I, the user who should be in charge, wish to.

### What actually happened?

On a 49" widescreen monitor this makes for a window with a max width of roughly 2560 pixels wide when the toolbar icons need only 1040 pixels to fit all their content.

From a UX perspective, the programmer should not decide what the minimum "usable" width is.  The user may have a use case that is different than anything you can imagine.  The only real limit should probably be: no smaller than the minimum that still allows the user to expand the window back to a more reasonable size.

One could argue that it should not go smaller than the minimum width necessary to display all the toolbar icons; I would counter by saying that this ignores use cases where the user may wish to make it smaller to the point where it's wide enough to still read the code (which they may have written at only 60 characters per line) for a situation where they want to refer to the code but don't need to interact with the toolbar icons, just as a for-example.

[image]

### Operating System Version

macOS 12.4

### Mu Version

1.1.1

### Editor Log

2022-07-21 13:44:55,491 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.1
2022-07-21 13:44:55,491 - root:270(run) INFO: uname_result(system='Darwin', node='HaikuSilly.domain', release='21.5.0', version='Darwin Kernel Version 21.5.0: Tue Apr 26 21:08:29 PDT 2022; root:xnu-8020.121.3~4/RELEASE_ARM64_T8101', machine='x86_64', processor='i386')
2022-07-21 13:44:55,508 - root:271(run) INFO: Platform: macOS-10.16-x86_64-i386-64bit
2022-07-21 13:44:55,508 - root:272(run) INFO: Python path: ['/', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python38.zip', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/lib-dynload', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/setuptools-57.4.0-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/pip-21.1.3-py3.8.egg']
2022-07-21 13:44:55,508 - root:273(run) INFO: Language code: en_US
2022-07-21 13:44:55,508 - mu.settings:220(load) WARNING: No settings file found at /Users/tyler/Library/Application Support/mu/settings.json; skipping
2022-07-21 13:44:55,736 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2022-07-21 13:44:55,737 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2022-07-21 13:44:55,737 - mu.virtual_environment:685(ensure_path) INFO: Virtual Environment found at: /Users/tyler/Library/Application Support/mu/mu_venv-38-20220721-121304
2022-07-21 13:44:55,737 - mu.virtual_environment:698(ensure_interpreter) INFO: Interpreter found at: /Users/tyler/Library/Application Support/mu/mu_venv-38-20220721-121304/bin/python
2022-07-21 13:44:55,737 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/Users/tyler/Library/Application Support/mu/mu_venv-38-20220721-121304/bin/python', '-c', 'import sys; print("%s%s" % sys.version_info[:2])') with kwargs {'shell': False}
2022-07-21 13:44:55,946 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 38
2022-07-21 13:44:55,946 - mu.virtual_environment:730(ensure_interpreter_version) INFO: Both interpreters at version 38
2022-07-21 13:44:55,947 - mu.virtual_environment:759(ensure_pip) INFO: Pip found at: /Users/tyler/Library/Application Support/mu/mu_venv-38-20220721-121304/bin/pip
2022-07-21 13:44:55,949 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: pgzero
2022-07-21 13:44:55,949 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/Users/tyler/Library/Application Support/mu/mu_venv-38-20220721-121304/bin/python', '-c', 'import pgzero') with kwargs {'shell': False}
2022-07-21 13:44:55,980 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-07-21 13:44:55,980 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: flask
2022-07-21 13:44:55,981 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/Users/tyler/Library/Application Support/mu/mu_venv-38-20220721-121304/bin/python', '-c', 'import flask') with kwargs {'shell': False}
2022-07-21 13:44:56,154 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-07-21 13:44:56,154 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipykernel
2022-07-21 13:44:56,155 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/Users/tyler/Library/Application Support/mu/mu_venv-38-20220721-121304/bin/python', '-c', 'import ipykernel') with kwargs {'shell': False}
2022-07-21 13:44:56,558 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-07-21 13:44:56,558 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipython_genutils
2022-07-21 13:44:56,558 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/Users/tyler/Library/Application Support/mu/mu_venv-38-20220721-121304/bin/python', '-c', 'import ipython_genutils') with kwargs {'shell': False}
2022-07-21 13:44:56,588 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-07-21 13:44:56,588 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: esptool
2022-07-21 13:44:56,588 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/Users/tyler/Library/Application Support/mu/mu_venv-38-20220721-121304/bin/python', '-c', 'import esptool') with kwargs {'shell': False}
2022-07-21 13:44:56,643 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-07-21 13:44:56,643 - mu.virtual_environment:628(ensure_and_create) INFO: Valid virtual environment found at /Users/tyler/Library/Application Support/mu/mu_venv-38-20220721-121304
2022-07-21 13:44:56,643 - mu.settings:169(save) DEBUG: Saving to /Users/tyler/Library/Application Support/mu/venv.json
[... 56 lines cut ...]
pgzero            1.2.1
pickleshare       0.7.5
pip               22.0.3
platformdirs      2.5.1
prompt-toolkit    3.0.28
ptyprocess        0.7.0
pure-eval         0.2.2
pycparser         2.21
pygame            2.1.2
Pygments          2.11.2
pyserial          3.5
python-dateutil   2.8.2
pyzmq             22.3.0
reedsolo          1.5.4
setuptools        60.9.3
six               1.16.0
stack-data        0.2.0
tomli             2.0.1
tornado           6.1
traitlets         5.1.1
typing_extensions 4.1.1
wcwidth           0.2.5
Werkzeug          2.0.3
wheel             0.37.1
2022-07-21 13:45:00,227 - mu.virtual_environment:931(installed_packages) INFO: []

haikusw, 2022-07-21:
Issue is [here](https://github.com/mu-editor/mu/blob/f47782ee999269f24c5231310ff1e30cebd55eaf/mu/interface/main.py#L1155):

haikusw, 2022-07-22:
I fixed this by adding:

```
MIN_WINDOW_WIDTH = 200
MIN_WINDOW_HEIGHT = 200
```

haikusw, 2022-12-25:
I have this fixed.  Once my fix for #2375 gets merged into master I'll update my branch fixing this issue and create a PR.

ntoll, 2022-12-25:
Season's greetings @haikusw - and thank you so much for the gift of your contribution..! :santa:

Looks like @tjguk has merged the things needed for your fix for this issue. If I'm honest, I think the original reason for setting the dimensions (as you found) is lost in the mists of time. Your solution sounds like a good one. :+1:

haikusw, 2022-12-25:
Thank you.  I've merged the updated master into my fix branch for this and tested and it resolves the issue. 
 make check all comes up clean and so I've created a PR.
Cheers.

commit 925c6a1e (2022-12-26): Fixes #2315 (#2377)

* fix #2315.  Untested because can't build on M1 easily (#2316)

* fix tabs vs spaces

* Import new defines

* make tidy cleaned up spaces

* Make minimum match values from test; better minimum

* remove attempt to retain old logic, which realy doesn't make any sense, and just set minimum size

* Revert "remove attempt to retain old logic, which really doesn't make any sense, and just set minimum size"

This reverts commit c70ff4574202543d019b82c42ba11ad4df920238.

Occurred to me that someone might be on a vga or other small monitor and this change would have increased the [... cut]
closed 2022-12-26 by tjguk