mu-editor/mu #2316: ModuleNotFoundError: No module named 'PyQt5' after `pip install -e ".[dev]" as per Quickstart

haikusw, 2022-07-22:
### What were you trying to do?

build mu on macOS

### What steps did you take to trigger the issue?

Followed the "QuickStart" instructions [here](https://mu.readthedocs.io/en/latest/#quickstart)
but created python 3.8.13 (latest 3.8 release pyenv shows available).

> % pyenv install 3.8.13
> % git clone https://github.com/mu-editor/mu.git
> % cd mu
> % pyenv local 3.8.13
> % python --version
> Python 3.8.13
> % python3.8 -m venv .venv
> % source .venv/bin/activate
> (.venv) % python -m pip install --upgrade pip
> (.venv) % pip install -e ".[dev]"

... <snip>
[lots of work installing many packages (no references to PyQt5 anywhere in the output)]
... <snip>
[then see:]

> Installing collected packages: webencodings, wcwidth, snowballstemmer, pytz, pyserial, pure-eval, ptyprocess, pickleshare, mypy-extensions, mccabe, ipython_genutils, iniconfig, executing, distlib, commonmark, backcall, appnope, alabaster, zipp, wheel, urllib3, typing-extensions, traitlets, tornado, tomli, sphinxcontrib-serializinghtml, sphinxcontrib-qthelp, sphinxcontrib-jsmath, sphinxcontrib-htmlhelp, sphinxcontrib-devhelp, sphinxcontrib-applehelp, six, semver, rfc3986, pyzmq, pyparsing, pygments, pyflakes, pycodestyle, py, prompt-toolkit, pluggy, platformdirs, pkginfo, pexpect, pathspec, parso, nudatus, MarkupSafe, imagesize, idna, filelock, docutils, decorator, coverage, click, charset-normalizer, certifi, babel, attrs, adafruit-board-toolkit, virtualenv, rich, requests, python-dateutil, packaging, matplotlib-inline, jupyter-core, Jinja2, jedi, importlib-metadata, flake8, bleach, black, asttokens, stack-data, sphinx, requests-toolbelt, readme-renderer, qtpy, pytest, keyring, jupyter-client, twine, pytest-timeout, pytest-random-order, pytest-faulthandler, pytest-cov, ipython, ipykernel, qtconsole, mu-editor

>   Running setup.py develop for mu-editor
> Successfully installed Jinja2-3.1.2 MarkupSafe-2.1.1 adafruit-board-toolkit-1.1.0 alabaster-0.7.12 appnope-0.1.3 asttokens-2.0.5 attrs-21.4.0 babel-2.10.3 backcall-0.2.0 black-21.12b0 bleach-5.0.1 certifi-2022.6.15 charset-normalizer-2.1.0 click-8.0.4 commonmark-0.9.1 coverage-6.4.2 decorator-5.1.1 distlib-0.3.5 docutils-0.18.1 executing-0.8.3 filelock-3.7.1 flake8-4.0.1 idna-3.3 imagesize-1.4.1 importlib-metadata-4.12.0 iniconfig-1.1.1 ipykernel-5.5.6 ipython-8.4.0 ipython_genutils-0.2.0 jedi-0.18.1 jupyter-client-6.1.12 jupyter-core-4.11.1 keyring-23.7.0 matplotlib-inline-0.1.3 mccabe-0.6.1 mu-editor-1.1.1 mypy-extensions-0.4.3 nudatus-0.0.5 packaging-21.3 parso-0.8.3 pathspec-0.9.0 pexpect-4.8.0 pickleshare-0.7.5 pkginfo-1.8.3 platformdirs-2.5.2 pluggy-1.0.0 prompt-toolkit-3.0.30 ptyprocess-0.7.0 pure-eval-0.2.2 py-1.11.0 pycodestyle-2.8.0 pyflakes-2.4.0 pygments-2.12.0 pyparsing-3.0.9 pyserial-3.5 pytest-7.1.2 pytest-cov-3.0.0 pytest-faulthandler-2.0.1 pytest-random-order-1.0.4 pytest-timeout-2.1.0 python-dateutil-2.8.2 pytz-2022.1 pyzmq-23.2.0 qtconsole-4.7.7 qtpy-2.1.0 readme-renderer-35.0 requests-2.28.1 requests-toolbelt-0.9.1 rfc3986-2.0.0 rich-12.5.1 semver-2.13.0 six-1.16.0 snowballstemmer-2.2.0 sphinx-5.0.2 sphinxcontrib-applehelp-1.0.2 sphinxcontrib-devhelp-1.0.2 sphinxcontrib-htmlhelp-2.0.0 sphinxcontrib-jsmath-1.0.1 sphinxcontrib-qthelp-1.0.3 sphinxcontrib-serializinghtml-1.1.5 stack-data-0.3.0 tomli-1.2.3 tornado-6.2 traitlets-5.3.0 twine-4.0.1 typing-extensions-4.3.0 urllib3-1.26.10 virtualenv-20.15.1 wcwidth-0.2.5 webencodings-0.5.1 wheel-0.37.1 zipp-3.8.1

If I then do either `mu-editor` or `make check` I get an error saying "E   ModuleNotFoundError: No module named 'PyQt5'" 

(Detailed output in the "What actually happened?" section below)

I see "PyQt5==5.13.2" in the `install_requires = [` list so I'm not seeing what the issue is.

I did note that Windows and Raspberry Pi developer setup include an `apt get` comment to fetch PyQt5 so maybe something analogous needs to be added to the documentation for this to work on macOS also?

### What did you expect to happen?

run checks successfully, run app successfully.

### What actually happened?

The QuickStart steps seemed to work untilI try to run mu with:

```% mu-editor```

it fails to find the QtPy5 module:

> Traceback (most recent call last):
>   File "/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu/.venv/bin/mu-editor", line 33, in <module>
>     sys.exit(load_entry_point('mu-editor', 'console_scripts', 'mu-editor')())
>   File "/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu/.venv/bin/mu-editor", line 25, in importlib_load_entry_point
>     return next(matches).load()
>   File "/Users/haikusw/.pyenv/versions/3.8.13/lib/python3.8/importlib/metadata.py", line 77, in load
>     module = import_module(match.group('module'))
>   File "/Users/haikusw/.pyenv/versions/3.8.13/lib/python3.8/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 843, in exec_module
>   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
>   File "/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu/mu/app.py", line 33, in <module>
>     from PyQt5.QtCore import (
**> ModuleNotFoundError: No module named 'PyQt5'**

So I tried:
```(.venv) % make check```
as per the directions but that fails for seemingly the same reason:

> rm -rf build
> rm -rf dist
> rm -rf .coverage
> rm -rf .eggs
> rm -rf *.egg-info
> rm -rf docs/_build
> rm -rf .pytest_cache
> rm -rf lib
> rm -rf *.mp4
> rm -rf .git/avatar/*
> rm -rf venv-pup
> find . \( -name '*.py[co]' -o -name dropin.cache \) -delete
> find . \( -name '*.bak' -o -name dropin.cache \) -delete
> find . \( -name '*.tgz' -o -name dropin.cache \) -delete
> find . | grep -E "(__pycache__)" | xargs rm -rf
> rm -f ./mu/locale/messages.pot
> rm -f ./mu/wheels/*.zip
> python make.py black
> 
> Clean
> 
> black
> All done! ✨ 🍰 ✨
> 1 file would be left unchanged.
> All done! ✨ 🍰 ✨
> 1 file would be left unchanged.
> All done! ✨ 🍰 ✨
> 52 files would be left unchanged.
> All done! ✨ 🍰 ✨
> 2 files would be left unchanged.
> All done! ✨ 🍰 ✨
> 38 files would be left unchanged.
> All done! ✨ 🍰 ✨
> 7 files would be left unchanged.
> 
> flake8
> export LANG=en_GB.utf8
> pytest -v --random-order --cov-config .coveragerc --cov-report term-missing --cov=mu tests/
> ImportError while loading conftest '/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu/tests/conftest.py'.
> tests/conftest.py:4: in <module>
>     from PyQt5.QtWidgets import QApplication
**> E   ModuleNotFoundError: No module named 'PyQt5'**
> make: *** [coverage] Error 4

### Operating System Version

macOS 12.4

### Mu Version

1.1.1

### Other Info

exact code I'm on:

`% git branch --list -v`

> * master f47782e Merge pull request #2311 from tonybaloney/patch-1

`% git status`
> On branch master
> Your branch is up to date with 'origin/master'.
> 
> Untracked files:
>   (use "git add <file>..." to include in what will be committed)
> 	.python-version
> 
> nothing added to commit but untracked files present (use "git add" to track)
>

carlosperate, 2022-07-22:
Hi @haikusw,

If you are using an M1/M2 mac you'll need to use rosetta, as at the moment there are no Arm wheels for the dependencies that Mu needs.

Future versions will be able to update this dependencies, but we'll need to wait until the next release before we take the step to update them.

haikusw, 2022-07-22:
Thank you for the reply.  Yes, it's a November 2020 13" M1 MacBook Pro.
I don't know what "wheels" are (yet) or how to make them, but it's a good thread for me to go learn about next :)

I'm curious which dependencies are missing?  Initially it seemed many were missing, but it's been ~18 months since the M1 laptops were released and most things I've run into have been updated.  `brew` seems to show PyQt@5 available and it just installed with all dependencies under arm fine.  Does seem to require python 3.9 which is above mu's setup.py requirement though.

I'll start by giving `arch -x86_64 /System/Applications/Utilities/Terminal.app/Contents/MacOS/Terminal` a try (I wasn't aware you could run two copies of terminal - one under arm and one under Rosetta - at the same time, but it seems to work).

Speaking of dependencies needing updating, doing this again hits the very long time to resolve `pip install -e ".[dev]"` (took > 3 hours last time)….

Lot of entries of the form:

> INFO: This is taking longer than usual. You might need to provide the dependency resolver with stricter constraints to reduce runtime. See https://pip.pypa.io/warnings/backtracking for guidance. If you want to abort this run, press Ctrl + C.

It seems `pip install -e ".[dev]"` is checking a lot of quite old versions[^1] (which takes a long time (deep python call chain maxes cpu out at 100%)).  This suggests that many of the dependencies listed in setup.py seem like they are pretty out of date.  I see "FIXME"s indicating post 1.1 release things can go away, etc.  Maybe that can get fixed up when the wheels are updated.

All this just to test a 4 line PR for mu to fix #2315. <sigh>

----

[^1]:  For example, pytest is checking every version since version 4.6, of which there are a _many_:  
   Collecting pytest>=4.6  
     Downloading pytest-5.2.3-py3-none-any.whl (227 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 227.4/227.4 kB 8.7 MB/s eta 0:00:00  
     Downloading pytest-5.2.2-py3-none-any.whl (227 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 227.3/227.3 kB 7.5 MB/s eta 0:00:00  
     Downloading pytest-5.2.1-py3-none-any.whl (226 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 226.8/226.8 kB 7.9 MB/s eta 0:00:00  
     Downloading pytest-5.2.0-py3-none-any.whl (226 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 226.7/226.7 kB 7.7 MB/s eta 0:00:00  
     Downloading pytest-5.1.3-py3-none-any.whl (224 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 224.1/224.1 kB 7.8 MB/s eta 0:00:00  
     Downloading pytest-5.1.2-py3-none-any.whl (224 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 224.2/224.2 kB 7.7 MB/s eta 0:00:00  
     Downloading pytest-5.1.1-py3-none-any.whl (223 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 223.6/223.6 kB 7.3 MB/s eta 0:00:00  
     Downloading pytest-5.1.0-py3-none-any.whl (223 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 223.6/223.6 kB 8.6 MB/s eta 0:00:00  
     Downloading pytest-5.0.1-py3-none-any.whl (221 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 221.8/221.8 kB 6.6 MB/s eta 0:00:00  
   Collecting importlib-metadata>=0.12  
     Using cached importlib_metadata-4.12.0-py3-none-any.whl (21 kB)  
   Collecting pytest>=4.6  
     Downloading pytest-5.0.0-py3-none-any.whl (221 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 221.8/221.8 kB 7.1 MB/s eta 0:00:00  
     Downloading pytest-4.6.11-py2.py3-none-any.whl (231 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 231.6/231.6 kB 7.0 MB/s eta 0:00:00  
     Downloading pytest-4.6.10-py2.py3-none-any.whl (231 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 231.6/231.6 kB 8.4 MB/s eta 0:00:00  
     Downloading pytest-4.6.9-py2.py3-none-any.whl (231 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 231.1/231.1 kB 8.8 MB/s eta 0:00:00  
     Downloading pytest-4.6.8-py2.py3-none-any.whl (230 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 230.5/230.5 kB 7.6 MB/s eta 0:00:00  
     Downloading pytest-4.6.7-py2.py3-none-any.whl (230 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 230.4/230.4 kB 7.2 MB/s eta 0:00:00  
     Downloading pytest-4.6.6-py2.py3-none-any.whl (230 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 230.2/230.2 kB 7.2 MB/s eta 0:00:00  
     Downloading pytest-4.6.5-py2.py3-none-any.whl (230 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 230.1/230.1 kB 7.2 MB/s eta 0:00:00  
     Downloading pytest-4.6.4-py2.py3-none-any.whl (229 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 229.6/229.6 kB 7.9 MB/s eta 0:00:00  
     Downloading pytest-4.6.3-py2.py3-none-any.whl (229 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 229.6/229.6 kB 9.3 MB/s eta 0:00:00  
     Downloading pytest-4.6.2-py2.py3-none-any.whl (229 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 229.5/229.5 kB 8.4 MB/s eta 0:00:00  
     Downloading pytest-4.6.1-py2.py3-none-any.whl (229 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 229.9/229.9 kB 9.6 MB/s eta 0:00:00  
     Downloading pytest-4.6.0-py2.py3-none-any.whl (229 kB)  
        ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 229.8/229.8 kB 7.7 MB/s eta 0:00:00.  
  .  
  Likewise platformdirs and many others.

haikusw, 2022-07-22:
@carlosperate hey, so I tried using rosetta as you suggested (by launching another terminal via `arch`) and I'm still getting the same error. 
I'm using python 3.8.13 (latest pre 3.9). 

Here's the console log generated showing my commands and the output as I followed the QuickStart instructions again.  You can see that I confirmed it was running under rosetta with the `arch` command and confirmed the version of python that was being used inside the venv.

Same exact error, unfortunately.  Any ideas?

```
haikusw@HaikuSilly mu_dev % echo `arch`
i386
haikusw@HaikuSilly mu_dev % git clone https://github.com/mu-editor/mu.git mu2
Cloning into 'mu2'...
remote: Enumerating objects: 13673, done.
remote: Counting objects: 100% (12/12), done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 13673 (delta 3), reused 7 (delta 0), pack-reused 13661
Receiving objects: 100% (13673/13673), 22.31 MiB | 5.58 MiB/s, done.
Resolving deltas: 100% (9555/9555), done.
haikusw@HaikuSilly mu_dev % cd mu2
haikusw@HaikuSilly mu2 % ls
AUTHORS.rst          ROADMAP.rst          pytest.ini
CHANGES.rst          conf/                requirements.txt
CODE_OF_CONDUCT.rst  debian/              run.py*
CONTRIBUTING.rst     docs/                setup.cfg
LICENSE              make.cmd             setup.py
MANIFEST.in          make.py              tests/
Makefile             mu/                  todo.txt
README.rst           package/             utils/
haikusw@HaikuSilly mu2 % pyenv versions
  system
  2.7.18
  3.8.13
  3.9.5
* 3.9.7 (set by /Users/haikusw/.python-version)
haikusw@HaikuSilly mu2 % pyenv local 3.8.13
haikusw@HaikuSilly mu2 % pyenv versions
  system
  2.7.18
* 3.8.13 (set by /Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu2/.python-version)
  3.9.5
  3.9.7
haikusw@HaikuSilly mu2 % python -m venv .venv
haikusw@HaikuSilly mu2 % source .venv/bin/activate
(.venv) haikusw@HaikuSilly mu2 % python -m pip install --upgrade pip
Requirement already satisfied: pip in ./.venv/lib/python3.8/site-packages (22.0.4)
Collecting pip
  Using cached pip-22.2-py3-none-any.whl (2.0 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 22.0.4
    Uninstalling pip-22.0.4:
      Successfully uninstalled pip-22.0.4
Successfully installed pip-22.2
(.venv) haikusw@HaikuSilly mu2 % python --version
Python 3.8.13
(.venv) haikusw@HaikuSilly mu2 % pip install -e ".[dev]"
Obtaining file:///Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu2
  Preparing metadata (setup.py) ... done
Collecting jupyter-client<6.2,>=4.1
  Using cached jupyter_client-6.1.12-py3-none-any.whl (112 kB)
Collecting ipykernel<6,>=4.1
  Using cached ipykernel-5.5.6-py3-none-any.whl (121 kB)
Collecting ipython_genutils>=0.2.0
  Using cached ipython_genutils-0.2.0-py2.py3-none-any.whl (26 kB)
Collecting qtconsole==4.7.7
  Using cached qtconsole-4.7.7-py2.py3-none-any.whl (118 kB)
Collecting adafruit-board-toolkit~=1.1
  Using cached adafruit_board_toolkit-1.1.0-py3-none-any.whl
Collecting pyserial~=3.5
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
[... 170 lines cut ...]
  Using cached executing-0.8.3-py2.py3-none-any.whl (16 kB)
Installing collected packages: webencodings, wcwidth, snowballstemmer, pytz, pyserial, pure-eval, ptyprocess, pickleshare, mypy-extensions, mccabe, ipython_genutils, iniconfig, executing, distlib, commonmark, backcall, appnope, alabaster, zipp, wheel, urllib3, typing-extensions, traitlets, tornado, tomli, sphinxcontrib-serializinghtml, sphinxcontrib-qthelp, sphinxcontrib-jsmath, sphinxcontrib-htmlhelp, sphinxcontrib-devhelp, sphinxcontrib-applehelp, six, semver, rfc3986, pyzmq, pyparsing, pygments, pyflakes, pycodestyle, py, prompt-toolkit, pluggy, platformdirs, pkginfo, pexpect, pathspec, parso, nudatus, MarkupSafe, imagesize, idna, filelock, docutils, decorator, coverage, click, charset-normalizer, certifi, babel, attrs, adafruit-board-toolkit, virtualenv, rich, requests, python-dateutil, packaging, matplotlib-inline, jupyter-core, Jinja2, jedi, importlib-metadata, flake8, bleach, black, asttokens, stack-data, sphinx, requests-toolbelt, readme-renderer, qtpy, pytest, keyring, jupyter-client, twine, pytest-timeout, pytest-random-order, pytest-faulthandler, pytest-cov, ipython, ipykernel, qtconsole, mu-editor
  Running setup.py develop for mu-editor
Successfully installed Jinja2-3.1.2 MarkupSafe-2.1.1 adafruit-board-toolkit-1.1.0 alabaster-0.7.12 appnope-0.1.3 asttokens-2.0.5 attrs-21.4.0 babel-2.10.3 backcall-0.2.0 black-21.12b0 bleach-5.0.1 certifi-2022.6.15 charset-normalizer-2.1.0 click-8.0.4 commonmark-0.9.1 coverage-6.4.2 decorator-5.1.1 distlib-0.3.5 docutils-0.18.1 executing-0.8.3 filelock-3.7.1 flake8-4.0.1 idna-3.3 imagesize-1.4.1 importlib-metadata-4.12.0 iniconfig-1.1.1 ipykernel-5.5.6 ipython-8.4.0 ipython_genutils-0.2.0 jedi-0.18.1 jupyter-client-6.1.12 jupyter-core-4.11.1 keyring-23.7.0 matplotlib-inline-0.1.3 mccabe-0.6.1 mu-editor-1.1.1 mypy-extensions-0.4.3 nudatus-0.0.5 packaging-21.3 parso-0.8.3 pathspec-0.9.0 pexpect-4.8.0 pickleshare-0.7.5 pkginfo-1.8.3 platformdirs-2.5.2 pluggy-1.0.0 prompt-toolkit-3.0.30 ptyprocess-0.7.0 pure-eval-0.2.2 py-1.11.0 pycodestyle-2.8.0 pyflakes-2.4.0 pygments-2.12.0 pyparsing-3.0.9 pyserial-3.5 pytest-7.1.2 pytest-cov-3.0.0 pytest-faulthandler-2.0.1 pytest-random-order-1.0.4 pytest-timeout-2.1.0 python-dateutil-2.8.2 pytz-2022.1 pyzmq-23.2.0 qtconsole-4.7.7 qtpy-2.1.0 readme-renderer-35.0 requests-2.28.1 requests-toolbelt-0.9.1 rfc39 [... cut ...]
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu2/mu/app.py", line 33, in <module>
    from PyQt5.QtCore import (
ModuleNotFoundError: No module named 'PyQt5'

```

haikusw, 2022-07-22:
Decided to try again with python 3.9.5 (and modify setup.py to accept python <=3.9.5) to see if PyQt5 seeming to require 3.9 or later might be contributing.  But that made the pip install -e ".[dev]"  become impossible again - hours later it died with an exception message:  "**pip._vendor.resolvelib.resolvers.ResolutionTooDeep: 2000000**". 

Here's the tail end of that log:

> % pip install -e ".[dev]"
> ...
> INFO: pip is looking at multiple versions of jupyter-client to determine which version is compatible with other requirements. This could take a while.
> Collecting jupyter-client<6.2,>=4.1
>   Downloading jupyter_client-6.1.11-py3-none-any.whl (108 kB)
>      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 108.7/108.7 kB 2.5 MB/s eta 0:00:00
>   Downloading jupyter_client-6.1.10-py3-none-any.whl (108 kB)
>      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 108.7/108.7 kB 3.2 MB/s eta 0:00:00
> Collecting jedi<=0.17.2
>   Downloading jedi-0.17.2-py2.py3-none-any.whl (1.4 MB)
>      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.4/1.4 MB 26.1 MB/s eta 0:00:00
> Collecting jupyter-client<6.2,>=4.1
>   Downloading jupyter_client-6.1.9-py3-none-any.whl (108 kB)
>      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 108.7/108.7 kB 3.5 MB/s eta 0:00:00
> ERROR: Exception:
> Traceback (most recent call last):
>   File "/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu/venv/lib/python3.9/site-packages/pip/_internal/cli/base_command.py", line 167, in exc_logging_wrapper
>     status = run_func(*args)
>   File "/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu/venv/lib/python3.9/site-packages/pip/_internal/cli/req_command.py", line 247, in wrapper
>     return func(self, options, args)
>   File "/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu/venv/lib/python3.9/site-packages/pip/_internal/commands/install.py", line 369, in run
>     requirement_set = resolver.resolve(
>   File "/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu/venv/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 92, in resolve
>     result = self._result = resolver.resolve(
>   File "/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu/venv/lib/python3.9/site-packages/pip/_vendor/resolvelib/resolvers.py", line 481, in resolve
>     state = resolution.resolve(requirements, max_rounds=max_rounds)
>   File "/Users/haikusw/Desktop/RaspberryPi_stuff/mu_dev/mu/venv/lib/python3.9/site-packages/pip/_vendor/resolvelib/resolvers.py", line 392, in resolve
>     raise ResolutionTooDeep(max_rounds)
> **pip._vendor.resolvelib.resolvers.ResolutionTooDeep: 2000000**
>

haikusw, 2022-07-22:
Yah, ok. I give up.

> % brew install QT    
> Error: Cannot install under Rosetta 2 in ARM default prefix (/opt/homebrew)!
> To rerun under ARM use:
>     arch -arm64 brew install ...
> To install under x86_64, install Homebrew into /usr/local.

I'd have to reinstall my entire homebrew installation for x86 and keep it and the arm one straight on the path (at least).

x86 Macs are basically history at this point (only one model still being made I think?).

This needs to get fixed so you can build on an M1 as a native ARM build properly; too much to fight as is.

That's beyond my current python skills.  Maybe I'll come back to it once I've acquired more, but hopefully someone in the core group who understands all the history of the project and python interdependencies can take a look at it.

carlosperate, 2022-11-11:
Are you sure you were running `pip install mu-editor` using rosetta? The output of pip shows that it didn't try to install `PyQt5` at all, which happens when the platform is `arm64`, and under rosetta it should have been `x86_64`.

If you run this command under rosetta what do you get?
```
python3 -c 'import platform as p; print(p.machine(), p.architecture())'
```

haikusw, 2022-11-11:
Hi, yah is.  No, I'm probably not sure.  I had to move on.  I'll be happy to contribute fixes for the app when it can be built natively on ARM in a straightforward way.

Hopefully someone who can build it successfully will test the fix I contributed (referenced above) and merge it in.

haikusw, 2022-12-25:
I got an x86 mac and have verified my fix for the issue I was trying to fix by building there so this issue isn't blocked me any longer.  

It would be nice to be able to easily build on an M1 macOS box, but setting up a parallel brew install for x86 took a ton of work and didn't end up working for me (not 100% sure why).

carlosperate, 2023-01-13:
Thanks for the update @haikusw! 
Assuming this is working on rosetta I will close this issue, as we are not yet supporting M1 (that is something we'll be able to do soon once we can update the Mu dependencies).

closed 2023-01-13 by carlosperate

haikusw, 2023-01-13:
I was not able to get a parallel brew setup working in rosetta on my M1 machine in a reasonable amount of time.  Something always seemed to cross over and use the ARM version of a tool.  So I can't verify rosetta working; sorry.

psyched to hear about native M1 support though.  Working on Mu would be a lot more enjoyable on a 9 years newer Mac :)