mu-editor/mu #2318: Debian Bullseye / Lubuntu 22.04 AppImage "Segmentation fault"

cpina, 2022-07-23:
### What were you trying to do?

Use Mu-Editor

### What steps did you take to trigger the issue?

I followed the steps from https://codewith.mu/en/howto/1.1/install_linux

### What did you expect to happen?

mu-editor would run

### What actually happened?

I got a segmentation falt

### Operating System Version

Debian Bullseye 11.4 and Lubuntu 22.04

### Mu Version

1.1.1

### Other Info

```
carles@pinux:~$ ./Mu_Editor-1.1.1-x86_64.AppImage
No settings file found at /home/carles/.local/share/mu/venv.json; skipping
Segmentation fault
carles@pinux:~$ QT_QPA_PLATFORM=wayland ./Mu_Editor-1.1.1-x86_64.AppImage
No settings file found at /home/carles/.local/share/mu/venv.json; skipping
Failed to create wl_display (No such file or directory)
qt.qpa.plugin: Could not load the Qt platform plugin "wayland" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted
carles@pinux:~$ 
```

For the Debian: `apt install mu-editor` worked. For the LUbuntu I then had to apply the patch from https://github.com/mu-editor/mu/issues/2228

I'm happy to follow up any debugging for the Debian Bullseye. I will not have access to the LUbuntu anymore.

Both the Debian and LUbuntu installations use X11. Debian with lxde and LUbuntu with lxqt.

As said: no work around needed for me, reporting only so the projects know. Thanks very much for writing mu-editor! :-)

rp10007, 2022-08-01:
Same for regular Ubuntu 22.04, with same error message. This is a breaking problem for me. I will roll back to a previous version.

carlosperate, 2022-08-02:
The installation instructions were updated a couple of weeks ago:
https://codewith.mu/en/howto/1.1/install_linux

Have you tried `sudo apt install libfuse2`?

cpina, 2022-08-02:
`libfuse2` was already installed:
```
carles@pinux:~$ dpkg -l | grep -i libfuse2
ii  libfuse2:amd64                                2.9.9-5                                amd64        Filesystem in Userspace (library)
carles@pinux:~$ ./Mu_Editor-1.1.1-x86_64.AppImage 
No settings file found at /home/carles/.local/share/mu/venv.json; skipping
Violació de segment
carles@pinux:~$ 
```

Or just in case:
```
carles@pinux:~$ LANG=C bash 
carles@pinux:~$ dpkg -l | grep -i libfuse2
ii  libfuse2:amd64                                2.9.9-5                                amd64        Filesystem in Userspace (library)
carles@pinux:~$ ./Mu_Editor-1.1.1-x86_64.AppImage
No settings file found at /home/carles/.local/share/mu/venv.json; skipping
Segmentation fault
carles@pinux:~$ 
```

dhalbert, 2022-11-15:
I am seeing the same issue on Ubuntu 22.04 with the new 1.2.0 AppImage. libfuse2 is installed. Same errors as in the first post and in https://github.com/mu-editor/mu/issues/2318#issuecomment-1202342703,

carlosperate, 2022-11-15:
Do you get the same error with:

```
QT_QPA_PLATFORM=xcb ./Mu_Editor....
```

dhalbert, 2022-11-15:
> Do you get the same error with:
> 
> ```
> QT_QPA_PLATFORM=xcb ./Mu_Editor....
> ```

I do. But `XDG_SESSION_TYPE` is `x11`, not `wayland` (as asked by @ntoll). I am using Cinnamon on Ubuntu 22.04. If I switch to the standard Ubuntu gnome desktop, it works. So perhaps the OP is also on `x11`, not `wayland`, with Lubuntu.

dhalbert, 2022-11-15:
Looks like it did not get to that `logging.info()`:

```
$ MU_LOG_TO_STDOUT=1 QT_QPA_PLATFORM=xcb bin/MuEditor-Linux-1.2.0-x86_64.AppImage 
2022-11-15 11:42:16,368 - root:283(run) INFO: 

-----------------

Starting Mu 1.2.0
2022-11-15 11:42:16,368 - root:284(run) INFO: uname_result(system='Linux', node='cod', release='5.15.0-53-generic', version='#59-Ubuntu SMP Mon Oct 17 18:53:30 UTC 2022', machine='x86_64', processor='x86_64')
2022-11-15 11:42:16,369 - root:285(run) INFO: Platform: Linux-5.15.0-53-generic-x86_64-with-glibc2.2.5
2022-11-15 11:42:16,369 - root:286(run) INFO: Python path: ['/tmp/.mount_MuEditJhxMXT/usr/lib/python38.zip', '/tmp/.mount_MuEditJhxMXT/usr/lib/python3.8', '/tmp/.mount_MuEditJhxMXT/usr/lib/python3.8/lib-dynload', '/tmp/.mount_MuEditJhxMXT/usr/lib/python3.8/site-packages']
2022-11-15 11:42:16,369 - root:287(run) INFO: Language code: en_US
2022-11-15 11:42:16,369 - mu.settings:220(load) WARNING: No settings file found at /home/halbert/.local/share/mu/settings.json; skipping
Segmentation fault (core dumped)
```

Here is a set of strace logs: `sudo strace -f -o strace.log -f bin/MuEditor-Linux-1.2.0-x86_64.AppImage`. I had to run as root to avoid a FUSE problem.

[strace.zip](https://github.com/mu-editor/mu/files/10014590/strace.zip)

carlosperate, 2022-11-15:
Okay, thanks everyone for all the info, I've been looking into this and been able to replicate in an Ubuntu 22.04 VM, but only when configured to run with X11.
22.04 with Wayland and 20.04 with X11 works.

I believe the issue is somewhat related to the Python build we are shipping, 3.8.12 from https://github.com/indygreg/python-build-standalone

To test it we can [download it](https://github.com/indygreg/python-build-standalone/releases/download/20211017/cpython-3.8.12-x86_64-unknown-linux-gnu-pgo-20211017T1616.tar.zst), uncompress it, install PyQt5:
```
./python/install/bin/python3 -m pip install PyQt5
```
And run this code in the REPL:
```
>>> from PyQt5.QtWidgets import QApplication, QWidget
>>> import sys
>>> app = QApplication(sys.argv)
Segmentation fault (core dumped)
```

However, doing the same with [3.8.13](https://github.com/indygreg/python-build-standalone/releases/download/20220630/cpython-3.8.13+20220630-x86_64-unknown-linux-gnu-pgo-full.tar.zst) works.

We'll need to update the packaged Python and this should be fixed for the next release.

ntoll, 2022-11-16:
Bravo @carlosperate - this is really great work..! :tada:

carryonrewardless, 2022-11-16:
In my case I have python 3.10.6 and X11 with 22.04 Budgie.  Sounds like I will need to install python 3.8.x.

carlosperate, 2022-11-16:
I've created a test AppImage in CI, and a quick test on my VM seems to work.
If anybody could download it and test if it works in their environment that'd be great!

You'll need to be logged into GitHub to be able to click on the download link from this page:
https://github.com/mu-editor/mu/actions/runs/3479069676
[image]

ntoll, 2022-11-16:
Works for me on Ubuntu 20.02 with standard X11 and Wayland.

dhalbert, 2022-11-16:
Worked for me on Ubuntu 22.04 running Cinnamon desktop (uses x11, not Wayland). Thanks!!

carlosperate, 2023-01-10:
Okay, I've merged PR https://github.com/mu-editor/mu/pull/2371 which will have fixed this issue for the next Mu release, thanks everyone for the report and the testing!

closed 2023-01-10 by carlosperate