mu-editor/mu #2324: UnicodeEncodeError: 'gbk' codec can't encode character

winnerpig329, 2022-08-09:
### What were you trying to do?

running the program

### What steps did you take to trigger the issue?

when the progamme using Chinese character
then click "run" button

### What did you expect to happen?

programme running correctly

### What actually happened?

UnicodeEncodeError: 'gbk' codec can't encode character

[image]

### Operating System Version

win10

### Mu Version

1.1.1

tjguk, 2022-08-09:
This is a Python issue rather than a Mu one as such :)
Can you try putting the following line at the top of your file please?

# -*- coding: UTF-8 -*-

tjguk, 2022-08-09:
(Please note that the previous version was messed up by the markdown renderer. cf here for some examples https://peps.python.org/pep-0263/#examples