mu-editor/mu #2325: Mu Editor and the Tinybit library doesn't work on a Mac

jdsjdk, 2022-08-10:
### What were you trying to do?

I am trying to get the Tinybit library to work with the Mu editor on a Mac.

### What steps did you take to trigger the issue?

So this is what is supposed to happen - and it does on a Window Machine:
If you go here:  http://www.yahboom.net/study/Tiny:bit and then you click on the sections:
2. Preparation before class --> 2.Import Tiny:bit library

Again, on Windows this works just fine.

Here's what happens on a Mac:
1. I take the Tinybit library: "Tinybit-micropythonV1.hex" and upload it to the mounted robot image on my mac
2. Once it's done installing, I verify that it works in the Mu Editor by open it up, and then hitting the REPL button in the Mu editor
       2.1.  I see that the text "MicroPython for Tinybit V1.3 Modified by the Yahboom Team" displays correctly
       2.2.  I finish checking my code and making sure there's no errors, hit the REPL button again, and then I hit the 
               "Flash" button

Once this is done, the "Tinybit-micropythonV1.hex" file and library gets overwritten on the mac and POOF, all of a sudden the syntax `import tinybit` is no longer recognized by the robot anymore, and none of the Tinybit commands work anymore.  Sad.  Now my students cannot fully program their robots the way that they would like!

**PLEASE NOTE**

I have already spoken to the Yahboom Support team about this issue, and they do not know how to solve this problem.  They recommended that I reach out to you guys to assist.  Thanks again.

### What did you expect to happen?

For the Tinybit library to work on a Mac, and not get overwritten.

### What actually happened?

Every time I try to re-upload the Tinybit library file to the Microbit on a mac, it gets overwritten on within the Mu Editor once I hit the "Flash" button, and the MicroPython for Tinybit library gets overwritten once I flash the robot with my code.  It no longer recognizes `import tinybit` syntax, and thus the robot will not work properly on the mac.

### Operating System Version

Me and my students are on different OS versions of Mac, and it happens with everyone.

### Mu Version

Current version 1.1.1

### Other Info

You can collaborate with the Yahboom team to see what can be done to fix this issue on WhatsApp:

+86 186 8237 8128
+86 131 0654 1306

You can also email them:  support@yahboom.com

They have told me that this works fine on Windows, but they cannot get the "Tinybit-micropythonV1.hex" library to properly import on the Mac.

Please help!  Getting this to work for my students would be a HUGE help for them!  Thanks again!

carlosperate, 2022-08-10:
Hi @jdsjdk 

If Yahboom has built their own MicroPython runtime, have you tried adding the path to that hex file in the Mu micro:bit settings?

[image]

jdsjdk, 2022-08-13:
I did actually, and for the file "Tinybit-micropythonV1.hex" it said, "cannot find file" or something to that effect.  If you could provide some instructions so that the Mu editor can find the file that would be great.  I might have an idea on how to do it, but the more assistance the merrier.  Thanks again for all of your help.

jdsjdk, 2022-08-30:
Hello @carlosperate, 
I wanted to follow up with a screen shot as well.  Please have a look at the following error (see attached).

[image]

jdsjdk, 2022-08-30:
Hello @carlosperate, please take a look at this screenshot for a better understanding of the error, here have a look:

[image]

Please keep me posted if anything can be done.  A friendly reminder, the fall semester is coming up, and new student's will be depending on this working.  Anything you can do to help these burgeoning minds would be greatly appreciated, thanks again for all of your help!

carlosperate, 2022-08-30:
Hi @jdsjdk,

That error basically indicates that the path to the hex file added to the settings does not exists, or that Mu cannot find it (maybe due to file permissions).
The file permissions is unlikely, assuming Mu has been able to save files into your `~/mu_code` directory.
Could you double check that the file path is correct?
If you are 100% sure the path is correct, can you open the hex file with Mu (like it was a Python script)? Does it give you any errors that way?

jdsjdk, 2022-09-29:
Hello @carlosperate, let me get to this task this weekend, and I'll look into it.  More than likely I'll need your help on **where** to put the .hex file in the `~/mu_code` location.  
Currently I have been pretty busy with other projects.  Hence why the delay.  Thanks again.

jdsjdk, 2023-01-06:
Hello @carlosperate, sometime this week I'd like to see if I can fully get this bug fixed.  I now have longer term goals of somehow integrating robotics with AI.  The first step for me is to get the robotics programming working again with python.  This has now turned into a personal project.  
:-)

occbee, 2023-07-27:
Did anyone find a fix to this problem? As I am  having the same issue

jdsjdk, 2023-08-01:
> Did anyone find a fix to this problem? As I am having the same issue

While I have been busy with other projects, has anyone been able to come up with a more concise way of fixing this issue for Mac?  The more AI explodes everywhere, the more that we need to be able to make robotics ubiquitous and easier to use across not only PC, but Mac as well.  Thanks.