mu-editor/mu #2329: Won't run on Mac OSX 10.12

arxionic, 2022-08-14:
### What were you trying to do?

Start the editor for use with Circuit Python and Adafruit hardware

### What steps did you take to trigger the issue?

Downloaded v1.1.1 dmg through Circuit Python links.  Mounted the dmg and copied the program to my Applications directory. Double clicked to run the Mu editor program.

### What did you expect to happen?

I was kind of hoping the program would run. ;-)

### What actually happened?

Program start failed with error report:
ate
    self.install_baseline_packages()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 885, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: numpy-1.22.2-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform. 

### Operating System Version

OSX 10.12 (Sierra)

### Mu Version

1.1.1

### Other Info

/Users/xxxxx/Library/Logs/mu/mu.log
[mu.log](https://github.com/mu-editor/mu/files/9332630/mu.log)

### Editor Log

/Users/xxxxx/Library/Logs/mu/mu.log
[mu.log](https://github.com/mu-editor/mu/files/9332630/mu.log)

carlosperate, 2022-08-15:
Hi @davidabarter, thank you for the report!

I believe you are encountering this issue: https://github.com/mu-editor/mu/issues/2120

Until we release a version of Mu with a fix this is a possible workaround:

Run the following command on a terminal:

```
rm /Applications/Mu\ Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/1.1.1.zip
```

Then launch Mu again.

If you could let us know if this worked or not that would help us a lot as well. Thanks!

arxionic, 2022-08-16:
Thank you very much for the very prompt direct response to
my issue.

Removing the nested 1.1.1.zip package resolved the issue
and the program appears to be running properly.

I will share this fix with the Adafruit forum where I also reported
the issue, and I'll include copious kudos for your support! ☺

in peace,
aaron

carlosperate, 2022-08-16:
Thanks for the update, I'm glad it now works! 🎉

closed 2022-08-16 by carlosperate