mu-editor/mu #2330: mu-editor will not run on mint 21 xfce

wizard0072, 2022-08-15:
Have install the package that is available in mint 21 and keep getting same error

Traceback (most recent call last):
  File "/usr/bin/mu-editor", line 33, in <module>
    sys.exit(load_entry_point('mu-editor==1.0.3', 'console_scripts', 'mu-editor')())
  File "/usr/share/mu-editor/mu/app.py", line 148, in run
    editor_window.setup(editor.debug_toggle_breakpoint, editor.theme)
  File "/usr/share/mu-editor/mu/interface/main.py", line 825, in setup
    self.autosize_window()
  File "/usr/share/mu-editor/mu/interface/main.py", line 774, in autosize_window
    self.move((screen.width() - size.width()) / 2,
TypeError: arguments did not match any overloaded call:
  move(self, QPoint): argument 1 has unexpected type 'float'
  move(self, int, int): argument 1 has unexpected type 'float'

also i download and tried the aMu_Editor-1.1.1-x86_64.AppImage which gives the following output

No settings file found at /home/george/.local/share/mu/venv.json; skipping
Segmentation fault (core dumped)

any solution to this?

carlosperate, 2022-08-16:
Hi @wizard0072,
The issue with the deb package is listed here: https://github.com/mu-editor/mu/issues/2228

For the AppImage, does it work if it is run like this?
```
QT_QPA_PLATFORM=wayland Mu_Editor-1.1.1-x86_64.AppImage
```
https://github.com/mu-editor/mu/actions/runs/2699412265

wizard0072, 2022-08-17:
Failed to create wl_display (No such file or directory)
qt.qpa.plugin: Could not load the Qt platform plugin "wayland" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted (core dumped)

wizard0072, 2022-08-17:
Tried one from https://github.com/mu-editor/mu/actions/runs/2699412265

file mu-editor-Linux-d7fec50815fa1a68e484e5343572b27a80d359e.zip extracted it all

1st try
No settings file found at /home/$/.local/share/mu/venv.json; skipping
Segmentation fault (core dumped)

2nd try

QT_QPA_PLATFORM=wayland ./Mu_Editor-1.1.1-x86_64.AppImage
No settings file found at /home/george/.$/share/mu/venv.json; skipping
Failed to create wl_display (No such file or directory)
qt.qpa.plugin: Could not load the Qt platform plugin "wayland" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted (core dumped)

drohm, 2022-08-23:
I'm running LM21 Cinnamon and getting the same error as well.

abixadamj, 2022-08-24:
@tmontes - I know you do the best with AppImage.
I have the same situation: Ubuntu Mate 22.04

```
adasiek@pythonista:~/01_aplikacje$ ll
razem 826900
drwxrwxr-x  2 adasiek adasiek         8 sie 24 06:22 ./
drwxr-x--- 39 adasiek adasiek        57 sie 24 05:36 ../
-rwxr-xr-x  1 adasiek adasiek  95980512 kwi 22 15:18 balenaEtcher-1.7.9-x64.AppImage*
-rwxr-xr-x  1 adasiek adasiek 174187712 lip 19 19:28 Mu_Editor-1.1.1-github-2699412265-x86_64.AppImage*
-rwxrwxr-x  1 adasiek adasiek 171697344 sie 24 06:16 Mu_Editor-1.1.1-official-x86_64.AppImage*
-rw-rw-r--  1 adasiek adasiek 174192640 lip 19 17:28 Mu_Editor-AppImage-x86_64-d7fec50815fa1a68e484e5343572b27a80d359ed.tar
-rw-rw-r--  1 adasiek adasiek 173126007 sie 24 06:21 mu-editor-Linux-d7fec50815fa1a68e484e5343572b27a80d359ed.zip
-rwxrwxr-x  1 adasiek adasiek  57430439 sie 21 20:25 pcloud*
adasiek@pythonista:~/01_aplikacje$ QT_QPA_PLATFORM=wayland ./Mu_Editor-1.1.1-github-2699412265-x86_64.AppImage 
No settings file found at /home/adasiek/.local/share/mu/venv.json; skipping
Failed to create wl_display (No such file or directory)
qt.qpa.plugin: Could not load the Qt platform plugin "wayland" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Przerwane (zrzut pamięci) # CORE DUMPED
adasiek@pythonista:~/01_aplikacje$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

```
Deep inspection (via strace) shows:
```
prctl(PR_CAPBSET_READ, 0x2a /* CAP_??? */) = -1 EINVAL (Zły argument)
prctl(PR_CAPBSET_READ, 0x29 /* CAP_??? */) = -1 EINVAL (Zły argument)
munmap(0x7ff6b51ec000, 79215)           = 0
socket(AF_UNIX, SOCK_STREAM|SOCK_CLOEXEC, 0) = 7
connect(7, {sa_family=AF_UNIX, sun_path="/run/user/1000/wayland-0"}, 27) = -1 ENOENT (Nie ma takiego pliku ani katalogu)
close(7)                                = 0
write(2, "Failed to create wl_display (No "..., 56Failed to create wl_display (No such file or directory)
) = 56
write(2, "qt.qpa.plugin: Could not load th"..., 95qt.qpa.plugin: Could not load the Qt platform plugin "wayland" in "" even though it was found.
) = 95
write(2, "This application failed to start"..., 305This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
```
I have some QT5 libs in system...
```
adasiek@pythonista:~/01_aplikacje$ dpkg -l | grep qt5
ii  libqt5core5a:amd64                       5.15.3+dfsg-2ubuntu0.1                  amd64        Qt 5 core module
ii  libqt5dbus5:amd64                        5.15.3+dfsg-2ubuntu0.1                  amd64        Qt 5 D-Bus module
ii  libqt5designer5:amd64                    5.15.3-1                                amd64        Qt 5 designer module
ii  libqt5gui5:amd64                         5.15.3+dfsg-2ubuntu0.1                  amd64        Qt 5 GUI module
ii  libqt5help5:amd64                        5.15.3-1                                amd64        Qt 5 help module
ii  libqt5multimedia5:amd64                  5.15.3-1                                amd64        Qt 5 Multimedia module
ii  libqt5network5:amd64                     5.15.3+dfsg-2ubuntu0.1                  amd64        Qt 5 network module
ii  libqt5printsupport5:amd64                5.15.3+dfsg-2ubuntu0.1                  amd64        Qt 5 print support module
ii  libqt5qml5:amd64                         5.15.3+dfsg-1                           amd64        Qt 5 QML module
ii  libqt5qmlmodels5:amd64                   5.15.3+dfsg-1                           amd64        Qt 5 QML Models library
ii  libqt5qmlworkerscript5:amd64             5.15.3+dfsg-1                           amd64        Qt 5 QML Worker Script library
ii  libqt5quick5:amd64                       5.15.3+dfsg-1                           amd64        Qt 5 Quick library
ii  libqt5quickcontrols2-5:amd64             5.15.3+dfsg-1                           amd64        Qt 5 Quick Controls 2 library
ii  libqt5quicktemplates2-5:amd64            5.15.3+dfsg-1                           amd64        Qt 5 Quick Templates 2 library
ii  libqt5sql5:amd64                         5.15.3+dfsg-2ubuntu0.1                  amd64        Qt 5 SQL module
ii  libqt5sql5-sqlite:amd64                  5.15.3+dfsg-2ubuntu0.1                  amd64        Qt 5 SQLite 3 database driver
ii  libqt5svg5:amd64                         5.15.3-1                                amd64        Qt 5 SVG module
ii  libqt5test5:amd64                        5.15.3+dfsg-2ubuntu0.1                  amd64        Qt 5 test module
ii  libqt5widgets5:amd64                     5.15.3+dfsg-2ubuntu0.1                  amd64        Qt 5 widgets module
ii  libqt5xml5:amd64                         5.15.3+dfsg-2ubuntu0.1                  amd64        Qt 5 XML module
ii  python3-dbus.mainloop.pyqt5              5.15.6+dfsg-1ubuntu3                    amd64        D-Bus Qt main loop support for Python 3
ii  python3-pyqt5                            5.15.6+dfsg-1ubuntu3                    amd64        Python 3 bindings for Qt5
ii  python3-pyqt5.sip                        12.9.1-1build1                          amd64        runtime module for Python extensions using SIP
[... cut ...]
             bus info: pci@0000:00:02.0
             logical name: /dev/fb0
             version: 09
             width: 64 bits
             clock: 33MHz
             capabilities: vga_controller bus_master cap_list rom fb
             configuration: depth=32 driver=i915 latency=0 resolution=1920,1080
             resources: irq:138 memory:de000000-deffffff memory:80000000-bfffffff ioport:f000(size=64) memory:c0000-dffff
```

What can I check else? On the same machine with Ubuntu 20.04 everything was OK.