mu-editor/mu #2331: error while loading shared libraries: libcrypt.so.1 on latest Arch Linux

ericfont, 2022-08-16:


carlosperate, 2022-08-16:
Thanks for the report @ericfont. I've moved the issue from the https://github.com/mu-editor/docker-mu-appimage/ issue tracker to the mu-editor/mu, as that's were we are looking into Mu issues.

carlosperate, 2022-08-16:
If you install `libxcrypt-compat` package do you get the same error?

ericfont, 2022-08-16:
After installing `libxcrypt-compat`, I no longer get the same error, however I now get a segfault:

```
./Mu_Editor-1.1.1-x86_64.AppImage 
No settings file found at /home/e/.local/share/mu/venv.json; skipping
Segmentation fault (core dumped)
```

carlosperate, 2022-08-16:
Thanks for the quick test! Okay, we are getting one step closer.

Are you using Wayland or X11? Does the X11 seg fault happens when running the appimage with this env variable set?
```
QT_QPA_PLATFORM=wayland Mu_Editor-1.1.1-x86_64.AppImage
```

Or when using the AppImage from this PR build?
https://github.com/mu-editor/mu/actions/runs/2699412265

ericfont, 2022-08-17:
I was on X11.  Here is output when setting the env variable:

```
QT_QPA_PLATFORM=wayland ./Mu_Editor-1.1.1-x86_64.AppImage
No settings file found at /home/e/.local/share/mu/venv.json; skipping
Failed to create wl_display (No such file or directory)
qt.qpa.plugin: Could not load the Qt platform plugin "wayland" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted (core dumped)
```

ericfont, 2022-08-17:
here is output from PR build 2699412265 on X11:

```
QT_QPA_PLATFORM=X11 ./Mu_Editor-1.1.1-x86_64.AppImage
No settings file found at /home/e/.local/share/mu/venv.json; skipping
qt.qpa.plugin: Could not find the Qt platform plugin "x11" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted (core dumped)
```

ericfont, 2022-08-17:
FYI, I also have wayland installed on this same computer.  Earlier I ran with an X11 plasma desktop.  However if I now relogin and start in wayland plasma desktop, both of the Mu-editor app images seem to work (it at least opens at least to the python editor...haven't tested the program further).

carlosperate, 2022-08-17:
And if you run this?

```
QT_QPA_PLATFORM=xcb ./Mu_Editor-1.1.1-x86_64.AppImage
```

I assume that'd likely throw the same seg fault at before?

ericfont, 2022-08-17:
I get a segfault with `=xcb`, on both the official AppImage and run 2699412265's AppImage.  Both when booted into X11 or wayland.

carlosperate, 2022-08-17:
Okay, so it looks like the seg happens on X11, but it works on Wayland (with the env variable, or without it if using the PR AppImage, which sets the flag internally).

We'll need to look into why that might be happening in Arch, as it seems to work fine in X11 in Ubuntu 20.04 and 22.04.

Thanks for all the testing!

carlosperate, 2022-11-16:
I think maybe you are seeing this issue as well:

I've created a test AppImage in CI, and it seems to work for Ubuntu 22.04 with X11. Would you be able to test it to see if it works on your environment as well? It'd be really appreciated.

You'll need to be logged into GitHub to be able to click on the download link from this page:
https://github.com/mu-editor/mu/actions/runs/3479069676
[image]

ericfont, 2022-11-16:
Unfortunately I deleted my arch linux boot and so can't test.  Since you say you were able to reproduce this issue in an Ubuntu 22.04 VM configured to run with X11, and that your new build doesn't have that issue on that VM, then I am confident that arch linux X11 is fixed as well.

closed 2022-11-16 by ericfont

agilum, 2022-11-18:
Hi carlosperate, Your created mu-editor AppImage  (Mu_Editor-1.2.0-x86_64.AppImage) runs on Manjaro (Arch based)  while the official MuEditor-Linux-1.2.0-x86_64.AppImage does not. Segmentation  fault. libxcrypt-compat is installed.

Rgds

agilum, 2022-11-20:
> I think maybe you are seeing this issue as well: #2318
>
> I've created a test AppImage in CI, and it seems to work for Ubuntu 22.04 with X11. Would you be able to test it to see if it works on your [...]