mu-editor/mu #2336: Mu pointing to a previous Python version

ximeng1986, 2022-08-28:
### What were you trying to do?

Trying to use Python 3.10

### What steps did you take to trigger the issue?

I realized that MU is still pointing to Python version 3.8 and maybe because of conflicts Pygame is not working properly

### What did you expect to happen?

Pygame to work

### What actually happened?

Pygame did not work

### Operating System Version

MacOS 12.5.1

### Mu Version

The latest

carlosperate, 2022-09-06:
Hi @ximeng1986,

If Mu is downloaded from the https://codewith.mu website, it comes with its own Python integrated, so it won't be using any other Python installation you might have in the computer.

Right now Mu ships with Python 3.8, and pygame should be installed on first run.

What doesn't work for you exactly?

rakshasha-medhi, 2023-02-06:
@carlosperate other than codewith.mu what alternative do we have? 
I'm also having the same issue: I got Python 3.10.4 installed but Mu is using 3.8.12
I want to use the Python in my system to avoid having to reinstall the libraries.

xbecas, 2023-06-19:
Hi @ximeng1986 and @rakshasha-medhi.

As @carlosperate said Sept 6 2022, "right now Mu ships with Python 3.8, and pygame should be installed on first run". Pygame is indeed installed with Mu, and there is a specific mode for it. Did you try it?