mu-editor/mu #2337: Moving window error, provided "float" values should be "int".

acjackman86, 2022-09-08:
### What were you trying to do?

Start the application.

### What steps did you take to trigger the issue?

Starting the application never happened due to this issue.

### What did you expect to happen?

For the application to start.

### What actually happened?

It never started. I got an error while trying to run it in the terminal, returning a type error that the window's move function was receiving float values instead of integers. The problem was solved by replacing lines 774 and 775 in main.py with these values:

self.move(int((screen.width() - size.width()) / 2),
                 int((screen.height() - size.height()) / 2))

After saving main.py, the Mu Editor I was using functioned as normal.

### Operating System Version

Ubuntu 22.04

### Mu Version

1.1.1

### Other Info

Just the small edit in the main.py file made everything run smoothly.

### Editor Log

2022-09-06 19:22:17,565 - root:271(run) INFO: 

-----------------

Starting Mu 1.1.1
2022-09-06 19:22:17,566 - root:272(run) INFO: uname_result(system='Linux', node='lil-mate-top', release='5.15.0-47-generic', version='#51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022', machine='x86_64', processor='x86_64')
2022-09-06 19:22:17,571 - root:273(run) INFO: Platform: Linux-5.15.0-47-generic-x86_64-with-glibc2.2.5
2022-09-06 19:22:17,571 - root:274(run) INFO: Python path: ['/tmp/.mount_Mu_EdiDZwRKR/usr/lib/python38.zip', '/tmp/.mount_Mu_EdiDZwRKR/usr/lib/python3.8', '/tmp/.mount_Mu_EdiDZwRKR/usr/lib/python3.8/lib-dynload', '/tmp/.mount_Mu_EdiDZwRKR/usr/lib/python3.8/site-packages']
2022-09-06 19:22:17,571 - root:275(run) INFO: Language code: en_US
2022-09-06 19:22:17,572 - mu.settings:220(load) WARNING: No settings file found at /home/ajay/.local/share/mu/settings.json; skipping
2022-09-06 19:22:17,573 - mu.settings:220(load) WARNING: No settings file found at /home/ajay/.local/share/mu/session.json; skipping
2022-09-06 19:40:01,534 - root:271(run) INFO: 

-----------------

Starting Mu 1.1.1
2022-09-06 19:40:01,557 - root:272(run) INFO: uname_result(system='Linux', node='lil-mate-top', release='5.15.0-47-generic', version='#51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022', machine='x86_64', processor='x86_64')
2022-09-06 19:40:01,569 - root:273(run) INFO: Platform: Linux-5.15.0-47-generic-x86_64-with-glibc2.2.5
2022-09-06 19:40:01,569 - root:274(run) INFO: Python path: ['/tmp/.mount_Mu_EdiInx2aU/usr/lib/python38.zip', '/tmp/.mount_Mu_EdiInx2aU/usr/lib/python3.8', '/tmp/.mount_Mu_EdiInx2aU/usr/lib/python3.8/lib-dynload', '/tmp/.mount_Mu_EdiInx2aU/usr/lib/python3.8/site-packages']
2022-09-06 19:40:01,569 - root:275(run) INFO: Language code: en_US
2022-09-06 19:40:01,570 - mu.settings:220(load) WARNING: No settings file found at /home/ajay/.local/share/mu/settings.json; skipping
2022-09-06 19:40:01,571 - mu.settings:220(load) WARNING: No settings file found at /home/ajay/.local/share/mu/session.json; skipping
2022-09-06 19:40:29,264 - root:271(run) INFO: 

-----------------

Starting Mu 1.1.1
2022-09-06 19:40:29,286 - root:272(run) INFO: uname_result(system='Linux', node='lil-mate-top', release='5.15.0-47-generic', version='#51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022', machine='x86_64', processor='x86_64')
2022-09-06 19:40:29,290 - root:273(run) INFO: Platform: Linux-5.15.0-47-generic-x86_64-with-glibc2.2.5
2022-09-06 19:40:29,291 - root:274(run) INFO: Python path: ['/tmp/.mount_Mu_EdidMQS9I/usr/lib/python38.zip', '/tmp/.mount_Mu_EdidMQS9I/usr/lib/python3.8', '/tmp/.mount_Mu_EdidMQS9I/usr/lib/python3.8/lib-dynload', '/tmp/.mount_Mu_EdidMQS9I/usr/lib/python3.8/site-packages']
2022-09-06 19:40:29,291 - root:275(run) INFO: Language code: en_US
2022-09-06 19:40:29,292 - mu.settings:220(load) WARNING: No settings file found at /home/ajay/.local/share/mu/settings.json; skipping
2022-09-06 19:40:29,292 - mu.settings:220(load) WARNING: No settings file found at /home/ajay/.local/share/mu/session.json; skipping
2022-09-06 19:40:29,536 - root:271(run) INFO: 

[... 468 lines cut ...]
        else:
            color_indext = 0
            pass
    if cpx.button_b:
        #cpx.play_file("EasyButton-A.wav")
        for how_bright in stages_fadein:
            cpx.pixels.brightness = how_bright
            time.sleep(0.1)

        cpx.pixels[0:10] = captured_color * 10
        cpx.pixels.brightness = 1

        time.sleep(300)
        for how_bright in stages_fadeout:
            cpx.pixels.brightness = how_bright
            time.sleep(0.1)

    cpx.pixels.brightness = 0
2022-09-06 21:09:14,793 - mu.logic:647(restore_session) INFO: Loaded files.
2022-09-06 21:09:14,793 - mu.logic:650(restore_session) INFO: User defined environment variables: []
2022-09-06 21:09:14,793 - mu.logic:654(restore_session) INFO: Minify scripts on micro:bit? False
2022-09-06 21:09:14,850 - mu.logic:1147(change_mode) INFO: Workspace directory: /media/ajay/CIRCUITPY
2022-09-06 21:11:13,442 - mu.logic:1051(quit) DEBUG: Session: {'theme': 'day', 'mode': 'adafruit', 'paths': ['/media/ajay/CIRCUITPY/code.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2022-09-06 21:11:13,442 - mu.logic:1052(quit) DEBUG: Saving session to: /home/ajay/.local/share/mu/session.json
2022-09-06 21:11:13,443 - mu.logic:1066(quit) INFO: Quitting.

superpotato9, 2022-09-11:
thanks you saved me so much time with this

carlosperate, 2023-01-06:
Thanks for the report @acjackman86!
This is covered in https://github.com/mu-editor/mu/issues/2228 already, so I'll close this to keep things organises in a single ticket 👍

closed 2023-01-06 by carlosperate