mu-editor/mu #2350: mu-editor on Ubuntu 22.04 WITHOUT Wayland

agutierrm, 2022-11-09:
Hi there, we are using mu-editor on our Linux lab on University. Some of our computers runs x11 wihtout Wayland,
because they are using nvidia drivers. 

So, is possible to run mu-editor 1.1.1 WITHOUT Wayland? For me it's impossible to run. And, also I have discovered that pip install mu-editor doesnt works:

`pip install mu-editor                                         
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement mu-editor (from versions: 1.0.0b12, 1.0.0b13, 1.0.0b14, 1.0.0b15, 1.0.0b16, 1.0.0b17, 1.0.0rc1, 1.0.0, 1.0.1, 1.0.2, 1.0.3)
ERROR: No matching distribution found for mu-editor`

This is normal? Thank you so much.

carlosperate, 2022-11-10:
How have you tried to run the AppImage? Just double clicking on it?

Have you run them from the command line? do they throw any error?

Usually the problem we encounter is launching the AppImage **with** wayland.

Can you try running it with these two environemntal variables to see if they run?
```
QT_QPA_PLATFORM=xcb ./Mu_Editor-1.1.1-x86_64.AppImage
```
```
QT_QPA_PLATFORM=wayland ./Mu_Editor-1.1.1-x86_64.AppImage
```

agutierrm, 2022-11-12:
Hello @carlosperate , thanks for your answer. 

I am running Ubuntu 22.04 LTS with nvidia driver, not using Wayland (x11). XDG_SESSION_TYPE contents "x11" value.

So, Im running : ./Mu_Editor-1.1.1-x86_64.AppImage A window appears asking for "Only run" or "Integrate and run" (I have translated from spanish, maybe with english language is different). If I click on "Only run" nothing happens. This window dissapears. No message in terminal.

If I run with: QT_QPA_PLATFORM=xcb ./Mu_Editor-1.1.1-x86_64.AppImage the behaviour is the same. Nothing happens. 

:-(

KGOrphanides, 2022-11-17:
I'm also affected by this, running Pop!_OS 22.04. Although I'm on AMD graphics hardware, I have a specific setup on my primary workstation that precludes using Wayland right now. 

Given the popularity of xorg, I presume that someone must have a viable workaround? 
*looks hopefully into the void*

dhalbert, 2022-11-17:
Related: #2318.

carlosperate, 2022-11-17:
Yes, thanks @dhalbert!

@agutierrm @KGOrphanides if you could check if this test AppImage works in your environment that'd be great!

You'll need to be logged into GitHub to be able to click on the download link from this page:
https://github.com/mu-editor/mu/actions/runs/3479069676
[image]

ntoll, 2022-11-17:
Bravo @carlosperate and hurrah @agutierrm ... best of luck with your students..! :tada:

KGOrphanides, 2022-11-17:
Working here, too. Thank you, @carlosperate !

carlosperate, 2023-01-13:
Thanks everyone! I'll close this issue as resolved and the fix will be shipped in the next release 👍

closed 2023-01-13 by carlosperate

RasheedAZ, 2024-03-10:
> How have you tried to run the AppImage? Just double clicking on it?
>
> Have you run them from the command line? do they throw any error?
>
> Usually the problem we encounter is launching the AppImage [...]

``It's not working for me either. Tried both variables. Pop!_OS 22.04

No settings file found at /home/az/.local/share/mu/venv.json; skipping
Segmentation fault (core dumped)``