mu-editor/mu #2355: Mu editor in 'circuit python mode' should list the new '.env'  file

kllsamui, 2022-11-15:
### What were you trying to do?

edit new secrets file of circuit python 8.0.0 beta 4

### What steps did you take to trigger the issue?

using 
[ LOAD ] file open dialog
but not see that file '.env' 
i can force it by type '.env'

while [new][save] [+ other] works

### What did you expect to happen?

that in circuit python mode a '.env' file is listed

### What actually happened?

not see it listed

### Operating System Version

win 10 and raspi linux

### Mu Version

0.9 linux 1.1 and 1.2 win 10

carlosperate, 2022-11-15:
Thanks for the report @kllsamui!
Do you have any links to the docs providing a bit more info about the `.env` file?

kllsamui, 2022-11-15:
https://docs.circuitpython.org/en/latest/docs/environment.html

ntoll, 2022-11-16:
OK. I've been able to recreate this. There are two aspects to the problem:

1. The dialog displayed by the operating system (not Mu) that's used by users to select a file to load, may by default not display hidden files on unix-like operating systems. I'm using Gnome with Linux and if I right click on the dialog I can set a flag to view such files (although this is off by default).
2. Mu limits the types of files it knows about depending on the mode. In this case, it doesn't know about `.env` files. I'll create a PR to fix this and reference this issue so you can see when the fix lands in master. It'll be in the next point release. :+1:

kllsamui, 2022-12-22:
just read
CP800b6
updated above document say:
old ".env" is now "settings.toml"
( will test now )

Neradoc, 2022-12-29:
Yeah we pivoted from `.env` to `settings.toml` because invisible files are a nightmare to manage especially in a beginner environment (but it's true in general, I don't setup my OS to show invisible files by default, that would be super annoying).
So I think we would add `.toml` to the file_extensions for the CP mode ?

kllsamui, 2023-01-08:
@Neradoc 
but even there is a MU v1.2.x version what support settings.toml in CP MODE,
on a RPI i still have only a **MU 1.0.3**  what completely refuses editing anything besides *.py files
( installed by RPI / recommended software / )

how we get that 3 involved parties together to have a updated version then **also on RPI ?**
( 32bit and 64bit Raspberry Pi OS)

kllsamui, 2023-02-11:
testing with the now CP800 and a PICO_W board
and the by CP800 provided ( empty, 0 long ) 'settings.toml' file 
the situation is messy!

on a RPI small desktop image only preinstalled editor:
- nano
- geany
using that to edit the 'settings.toml' file ( directly on the board ) leads to a bricked PICO

the user installed MU 1.0.3 on RPI refuses to open a '.toml' file ( any not .py )

________________

on a win PC user installed MU 1.2.0
can usually edit that file by typing the name 'settings.toml' ( and that name is remembered there in a ?recent? list )

BUT the by CP800 created 0 long file on the CIRCUITPY drive
win MU 1.2.0 refuses to open WITHOUT WARNING
so i have to create one and copy / overwrite it to the board.
________________
also interesting
i quote:
```@dhalbert
Another idea (which I haven't tried) would be to edit settings.toml as settings.py with Mu, and then rename it to settings.toml from the command line or from the file browser.
````
________________
all this trouble makes me think if i should recommend to my readers 
to ignore the CP800 settings.toml file / feature
and use a settings.py file like in the 'old' times.

________________
may i ask if there is a MU revision in work for the Raspberry Pi?

DJDevon3, 2023-06-01:
Some if not all of it could be avoided if there was a drop down option for "all file types" which is a common thing in many programs.  With Mu it will only view a select list of explicit file types. 

[image]