mu-editor/mu #2361: Mu Editor Fails to Load and Run

drnow3, 2022-12-04:
I have a new Lenovo laptop running Windows 10, and I have downloaded MuEditor-win64-1.2.0.  After running the install program, the Mu Editor program fails to run.  I would like to use this program but I am frustrated by its not running.

For reference, I have attached the error file.
[mu.log](https://github.com/mu-editor/mu/files/10148489/mu.log)

Suggestions

frodeseverin, 2022-12-07:
I had the same problem, described here: #378

xbecas, 2023-06-12:
@frodeseverin - Thanks for pointing out a solution.

@drnow3 - did you manage to run Mu?

frodeseverin, 2023-06-12:
Yes, it seems to have been corrected in the Microsoft Endpoint signatures now.

The procedure shown in 
https://github.com/mu-editor/crash-reports/issues/378
is a workaround, not a solution.