mu-editor/mu #2370: 1.20 Appimage Can't paste from external clipboard

rp10007, 2022-12-13:
### What were you trying to do?

paste example code from circuit python web docs

### What steps did you take to trigger the issue?

Opened Appimage, verified communication with QTPy SAMD21.
Opened code.py
Selected code on web page and copied it.
Returned to mu-editor
Tried to paste code, using both menu command and keyboard shortcut(s)

### What did you expect to happen?

pasted code to appear in editor window (or in REPL window)

### What actually happened?

Nothing. Verified that cut&paste works within mu by cutting some repl text and pasting it successfully. Also verified that text had been copied from web page by opening another editor and pasting successfully. (Code copied from that editor window also did not paste into mu.)

### Operating System Version

Ubuntu 22.04 Wayland

### Mu Version

1.2.0 x86 Appimage

### Other Info

Even with python 3.8 installed from deadsnakes, there does not appear to be a straightforward way to install mu using pip3. If there were, that could serve as a workaround.

carlosperate, 2022-12-13:
Hi @rp10007, thanks for the report.

While we look into this, you can follow the instructions from the "Install Mu via PyPI and pipx" section from this comment, which should help you get Mu installed from PyPI:
- https://github.com/mu-editor/mu/issues/2228

@ntoll could you try the AppImage in Linux to see if the clipboard works?

rp10007, 2022-12-14:
Thanks, the pipx installation instructions worked fine (and just for pedantry, copy/paste from external works fine in that installation).