mu-editor/mu #2398: Mu 1.20 doesn't work on MacOS High Sierra 10.13 due to zmq dependency

mrz944, 2023-02-16:
### What were you trying to do?

Run .app

### What steps did you take to trigger the issue?

Run Mu Editor for the first time

### Operating System Version

MacOS High Sierra 10.13.6

### Mu Version

1.2.0

### Other Info

When I tried to run Mu Editor for the first time app crashed with no error dialog.
I tried running `./Applications/Mu\ Editor.app/Contents/MacOS/Mu\ Editor` directly from the console and got an error caused by non compatible `libsodium.23.dylib` used by zmq.

After I manually downgraded entire `Mu\ Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq` catalog with zmq 20.0.0, everything works flawless.

carlosperate, 2023-02-20:
Thanks for the repor @mrz944!

We've identified this in issue https://github.com/mu-editor/mu/issues/2368 and should be fixed with PR https://github.com/mu-editor/mu/pull/2371.

The latest builds from CI should work correctly and the fix will be present in the next release, so I'll close this one as resolved (although not yet release).

Thanks!

closed 2023-02-20 by carlosperate

lagomorph, 2023-05-24:
I believe I'm getting this issue with the latest release 1.2.0 and macOS 10.13.6.

```
Traceback (most recent call last):
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/__main__.py", line 1, in <module>
    from mu.app import run
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 48, in <module>
    from .modes import (
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/modes/__init__.py", line 1, in <module>
    from .python3 import PythonMode
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/modes/python3.py", line 27, in <module>
    from qtconsole.manager import QtKernelManager
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/qtconsole/manager.py", line 9, in <module>
    from jupyter_client import KernelManager
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/jupyter_client/__init__.py", line 4, in <module>
    from .connect import *
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/jupyter_client/connect.py", line 21, in <module>
    import zmq
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/__init__.py", line 103, in <module>
    from zmq import backend
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/__init__.py", line 31, in <module>
    raise original_error from None
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/__init__.py", line 26, in <module>
    _ns = select_backend(first)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/select.py", line 31, in select_backend
    mod = import_module(name)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/cython/__init__.py", line 6, in <module>
    from . import (
ImportError: dlopen(/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/cython/_device.cpython-38-darwin.so, 2): Symbol not found: ____chkstk_darwin
  Referenced from: /Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/cython/../../.dylibs/libsodium.23.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/zmq/backend/cython/../../.dylibs/libsodium.23.dylib
```