mu-editor/mu #2402: i can't download mu 64 bits

nazha7, 2023-02-25:


carlosperate, 2023-02-28:
Hi @nazha7,
I'll the links in https://codewith.mu/en/download seem to be working. How did you try to download Mu?

calli299, 2023-05-05:
I cant download either 
[image]

ntoll, 2023-05-05:
@calli299 this appears to be because your computer is managed / limited in some way to only allow official downloads. Please speak to your system administrator to fix this.

calli299, 2023-05-05:
I am the admin of my pc and I don't know where to go to fix it

ntoll, 2023-05-05:
It's probably your virus checking software.

calli299, 2023-05-05:
I turned it off, still the same

calli299, 2023-05-05:
Thank you, I have it installed

xbecas, 2023-06-12:
Closing - all ended well.

Enjoy Mu and please share suggestions and bugs you may find.

closed 2023-06-12 by xbecas