mu-editor/mu #2403: Need to eject drive.

wflynn, 2023-02-26:
### What were you trying to do?

Edit code and save to make it run

### What steps did you take to trigger the issue?

Connect Circuit Playground Express board.  CPython Drive shows up.  Com3 shows up.
No REPL output...  this is wrong.
Write code.  Hit save.  It saves but does not run.  Still no REPL output.
Eject CPython Drive.  REPL starts working, code is running.  CTL-C and CTL-D work.
Modify code.  Can't save since drive is ejected.  Reset board.  Drive shows up.  REPL stops working.  Edit and save code.  Repeat.
Try 100000000 things including chatting with people on Discord.  Nothing works.
Move board to laptop.  Run Mu there.  All works fine.
Move board back to desktop.  Problem returns.
Something on my PC is messing up Mu.

Also, I had an FTDI cable plugged in to something that is not anything to do with any of this.  When I restart the board, Mu sees the board on Com3 (or whatever) go away and then goes and finds the FTDI cable on COM29 (or whatever) and says "Hey, I just found an ESP Micropython board, wanna switch?".  But it's not an anything.  The thing the FTDI cable is attached to isn't even powered. 

This is all shown in the attached log.

### What did you expect to happen?

It should save and run

### What actually happened?

It couldn't save since the drive was ejected

### Operating System Version

Win 10 on both machines

### Mu Version

1.2.0

### Other Info

See this link for more info:  https://discord.com/channels/327254708534116352/1079202175084941312
Also, this video shows early attempts at figuring out what's going on:  https://www.dropbox.com/s/tsoy8awd7alkaeh/2023-02-25%2016-58-00.mkv?dl=0

### Editor Log

See attached log
[Mu Log.txt](https://github.com/mu-editor/mu/files/10832760/Mu.Log.txt)

wflynn, 2023-02-27:
OK so it turns out there was a driver conflict.  Boot to safe mode, no issue.  Went through everything and it seems that Acronis 2016 was the isuse.  Uninstalled and now everything works fine.  What's up with that??

dhalbert, 2023-02-28:
https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#code-dot-py-restarts-constantly-3108374 lists one problem with Acronis, which might not have been yours. There are several other disk utility programs listed on that page that cause various kinds of troubles.

carlosperate, 2023-02-28:
Glad you were able to resolve it @wflynn, and thanks for the link @dhalbert!
I'll close this issue as resolved then.

closed 2023-02-28 by carlosperate

wflynn, 2023-02-28:
Thank you.  I guess in the 800 pages of stuff I read I missed that.  Ah well.  That’s why we’re paid the big bucks.

Bill

William E. Flynn Jr
Endeavor Consulting Group, LLC
203-514-2455

This document may contain privileged or confidential information which you must not read, copy, store or disclose unless it is intended for you or your organization or you have received prior permission. If received in error, please delete and contact the sender. Unless stated to the contrary, any opinions or comments are personal to the writer and do not necessarily represent the official view of the Company. While we have taken every precaution to minimize the risk of computer viruses, we cannot accept any liability for such viruses and you should carry out your own virus checks before opening any attachments contained in this document.