mu-editor/mu #2407: "Unknown sequence number while appending request" error when using Tkinter

dconover23, 2023-03-08:
### What were you trying to do?

I'm a new Python programmer and am attempting to create a Tkinter application. However, whenever I try to run it, I get the error described in the title.

### What steps did you take to trigger the issue?

1. Create a Tkinter application and attempt to add any widgets (entries, labels, etc.)
2. Run it.
3. Get the following error:

"[xcb] Unknown sequence number while appending request
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
python: ../../src/xcb_io.c:157: append_pending_request: Assertion `!xcb_xlib_unknown_seq_number' failed.

---------- FINISHED ----------
exit code: 6 status: 1"

### What did you expect to happen?

The application should open a window with the relevant widgets.

### What actually happened?

The application crashed with the error listed above.

### Operating System Version

Ubuntu 22.04 LTS

### Mu Version

1.2.0

### Editor Log

2023-03-08 08:39:13,545 - mu.settings:169(save) DEBUG: Saving to /home/bunny/.local/share/mu/session.json
2023-03-08 08:39:13,547 - mu.logic:1430(quit) INFO: Quitting.

2023-03-08 08:39:13,547 - mu.settings:169(save) DEBUG: Saving to /home/bunny/.local/share/mu/session.json
2023-03-08 08:39:13,548 - mu.settings:169(save) DEBUG: Saving to /home/bunny/.local/share/mu/venv.json
2023-03-08 08:44:18,286 - root:283(run) INFO: 

-----------------

Starting Mu 1.2.0
2023-03-08 08:44:18,286 - root:284(run) INFO: uname_result(system='Linux', node='bunny-Latitude-E6530', release='5.19.0-35-generic', version='#36~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb 17 15:17:25 UTC 2', machine='x86_64', processor='x86_64')
2023-03-08 08:44:18,288 - root:285(run) INFO: Platform: Linux-5.19.0-35-generic-x86_64-with-glibc2.2.5
2023-03-08 08:44:18,288 - root:286(run) INFO: Python path: ['/tmp/.mount_Mu_EdiyxMJdQ/usr/lib/python38.zip', '/tmp/.mount_Mu_EdiyxMJdQ/usr/lib/python3.8', '/tmp/.mount_Mu_EdiyxMJdQ/usr/lib/python3.8/lib-dynload', '/tmp/.mount_Mu_EdiyxMJdQ/usr/lib/python3.8/site-packages']
2023-03-08 08:44:18,288 - root:287(run) INFO: Language code: en_US
2023-03-08 08:44:18,289 - mu.settings:220(load) WARNING: No settings file found at /home/bunny/.local/share/mu/settings.json; skipping
2023-03-08 08:44:18,289 - root:312(run) INFO: Wayland detected, setting QT_QPA_PLATFORM=wayland
2023-03-08 08:44:18,696 - mu.virtual_environment:619(ensure_and_create) INFO: Added log handler.
2023-03-08 08:44:18,696 - mu.virtual_environment:630(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2023-03-08 08:44:18,697 - mu.virtual_environment:719(ensure_path) INFO: Virtual Environment found at: /home/bunny/.local/share/mu/mu_venv-38-20230307-205056
2023-03-08 08:44:18,697 - mu.virtual_environment:740(ensure_interpreter) INFO: No interpreter found at: /home/bunny/.local/share/mu/mu_venv-38-20230307-205056/bin/python
2023-03-08 08:44:18,697 - mu.virtual_environment:750(ensure_interpreter) INFO: Symlinked /home/bunny/.local/share/mu/mu_venv-38-20230307-205056/bin/python to AppImage's /tmp/.mount_Mu_EdiyxMJdQ/usr/bin/python3.8
2023-03-08 08:44:18,698 - mu.virtual_environment:463(run_subprocess) INFO: Running ('/home/bunny/.local/share/mu/mu_venv-38-20230307-205056/bin/python', '-c', 'import sys; print("%s%s" % sys.version_info[:2])') with kwargs {'shell': False}
2023-03-08 08:44:18,754 - mu.virtual_environment:475(run_subprocess) DEBUG: Process returned 0; output: 38
2023-03-08 08:44:18,754 - mu.virtual_environment:786(ensure_interpreter_version) INFO: Both interpreters at version 38
2023-03-08 08:44:18,754 - mu.virtual_environment:815(ensure_pip) INFO: Pip found at: /home/bunny/.local/share/mu/mu_venv-38-20230307-205056/bin/pip
2023-03-08 08:44:18,754 - mu.virtual_environment:798(ensure_key_modules) DEBUG: Verifying import of: pgzero
2023-03-08 08:44:18,754 - mu.virtual_environment:463(run_subprocess) INFO: Running ('/home/bunny/.local/share/mu/mu_venv-38-20230307-205056/bin/python', '-c', 'import pgzero') with kwargs {'shell': False}
2023-03-08 08:44:18,801 - mu.virtual_environment:475(run_subprocess) DEBUG: Process returned 0; output: No output received.
2023-03-08 08:44:18,801 - mu.virtual_environment:798(ensure_key_modules) DEBUG: Verifying import of: flask
2023-03-08 08:44:18,801 - mu.virtual_environment:463(run_subprocess) INFO: Running ('/home/bunny/.local/share/mu/mu_venv-38-20230307-205056/bin/python', '-c', 'import flask') with kwargs {'shell': False}
2023-03-08 08:44:19,144 - mu.virtual_environment:475(run_subprocess) DEBUG: Process returned 0; output: No output received.
2023-03-08 08:44:19,145 - mu.virtual_environment:798(ensure_key_modules) DEBUG: Verifying import of: ipykernel
2023-03-08 08:44:19,145 - mu.virtual_environment:463(run_subprocess) INFO: Running ('/home/bunny/.local/share/mu/mu_venv-38-20230307-205056/bin/python', '-c', 'import ipykernel') with kwargs {'shell': False}
2023-03-08 08:44:19,751 - mu.virtual_environment:475(run_subprocess) DEBUG: Process returned 0; output: No output received.
[... 439 lines cut ...]
numpy              1.23.4
parso              0.8.3
pexpect            4.8.0
pgzero             1.2.1
pickleshare        0.7.5
pip                22.3.1
platformdirs       2.5.4
prompt-toolkit     3.0.32
ptyprocess         0.7.0
pure-eval          0.2.2
pygame             2.1.2
Pygments           2.13.0
python-dateutil    2.8.2
pyzmq              24.0.1
requests           2.28.2
setuptools         65.5.1
six                1.16.0
stack-data         0.6.1
tornado            6.2
traitlets          5.5.0
urllib3            1.26.14
wcwidth            0.2.5
Werkzeug           2.2.2
wheel              0.38.4
2023-03-08 08:57:17,640 - mu.virtual_environment:993(installed_packages) INFO: ['certifi', 'charset-normalizer', 'idna', 'requests', 'urllib3']

carlosperate, 2023-03-13:
Hi @dconover23 

Can you post a minimal example Python script to trigger this error? Emphasis in minimal, please strip the code to the bare minimun that will cause the error, otherwise it'll be really hard for us look into it.

dconover23, 2023-03-13:
@carlosperate 

I couldn't find a way to post .py files on Github so here it is in plaintext:

import tkinter as tk

window = tk.Tk()

window.title("Crash")

# This line causes a crash
entry = tk.Entry(window)

carlosperate, 2023-03-13:
Thanks @dconover23! ( I hope you don't mind I edited your message to format the code with markdown).

The programme doesn't crash for me in macOS, next step would be to test it in Ubuntu 22.04.
Are you running Mu via the AppImage? Or did you installed it a different way?

dconover23, 2023-03-13:
@carlosperate 

It's being run using the AppImage, yeah.