mu-editor/mu #2412: Where is the 1.2.1 installer file ?

CrackXT, 2023-04-02:
### What were you trying to do?

I wanted to use the latest version.

### What steps did you take to trigger the issue?

looked where the latest version is linked if necessary

### What did you expect to happen?

1.2.1

### What actually happened?

There are several hints that version 1.2.1 is the current one and has (actually?) been released. The release date actually seems to have been 09.01.2023.

https://mu.readthedocs.io/en/latest/changes.html

[image]

But you can't download it and there seems to be no build (.exe,...) available.

### Operating System Version

W11

### Mu Version

1.2.0

### Other Info

Did they forget to create the corresponding builds or is this just an internal test version (alpha, beta, prerelease)?

lagomorph, 2023-05-24:
I am also curious about this release build since neither 1.1.1 nor 1.2.0 seems to work on macOS 10.13.6.

ProgramminCat, 2023-08-25:
Download Mu via your distributions package manager to get the latest version.

CrackXT, 2023-10-22:
According to the latest status, this version was never published... and there is probably soon the 1.2.2.
[image]
Topic closed

closed 2023-10-22 by CrackXT