mu-editor/mu #2423: PyGame mode issues in macOS 10.12 Sierra

carlosperate, 2023-06-27:


carlosperate, 2023-06-27:
```
pygame 2.5.0 (SDL 2.28.0, Python 3.8.13)
Hello from the pygame community. https://www.pygame.org/contribute.html
/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pgzero/runner.py:2: RuntimeWarning: use mixer: dlopen(/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/mixer.cpython-38-darwin.so, 2): Library not loaded: @loader_path/libmp3lame.0.dylib
  Referenced from: /Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/.dylibs/libsndfile.1.0.34.dylib
  Reason: no suitable image found.  Did find:
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/.dylibs/libmp3lame.0.dylib: cannot load 'libmp3lame.0.dylib' (load command 0x80000034 is unknown)
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/.dylibs/libmp3lame.0.dylib: cannot load 'libmp3lame.0.dylib' (load command 0x80000034 is unknown)
(ImportError: dlopen(/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/mixer.cpython-38-darwin.so, 2): Library not loaded: @loader_path/libmp3lame.0.dylib
  Referenced from: /Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/.dylibs/libsndfile.1.0.34.dylib
  Reason: no suitable image found.  Did find:
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/.dylibs/libmp3lame.0.dylib: cannot load 'libmp3lame.0.dylib' (load command 0x80000034 is unknown)
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/.dylibs/libmp3lame.0.dylib: cannot load 'libmp3lame.0.dylib' (load command 0x80000034 is unknown))
  pygame.mixer.pre_init(frequency=22050, size=-16, channels=2)
Traceback (most recent call last):
  File "/Users/carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Users/carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pgzero/__main__.py", line 1, in <module>
    from pgzero.runner import main
  File "/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pgzero/runner.py", line 2, in <module>
    pygame.mixer.pre_init(frequency=22050, size=-16, channels=2)
  File "/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/__init__.py", line 70, in __getattr__
    raise NotImplementedError(missing_msg)
NotImplementedError: mixer module not available (ImportError: dlopen(/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/mixer.cpython-38-darwin.so, 2): Library not loaded: @loader_path/libmp3lame.0.dylib
  Referenced from: /Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/.dylibs/libsndfile.1.0.34.dylib
  Reason: no suitable image found.  Did find:
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/.dylibs/libmp3lame.0.dylib: cannot load 'libmp3lame.0.dylib' (load command 0x80000034 is unknown)
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-100218/lib/python3.8/site-packages/pygame/.dylibs/libmp3lame.0.dylib: cannot load 'libmp3lame.0.dylib' (load command 0x80000034 is unknown))

---------- FINISHED ----------
exit code: 1 status: 0
```

carlosperate, 2023-06-27:
Going back to pygame version 2.0.3 works.

carlosperate, 2023-06-27:
```
$ /Users/carlos/Library/Application\ Support/mu/mu_venv-38-20230627-155005/bin/pip install "pygame==2.1.3"
Collecting pygame==2.1.3
  Downloading pygame-2.1.3-cp38-cp38-macosx_10_9_x86_64.whl (12.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 12.8/12.8 MB 5.6 MB/s eta 0:00:00
Installing collected packages: pygame
  Attempting uninstall: pygame
    Found existing installation: pygame 2.5.0
    Uninstalling pygame-2.5.0:
      Successfully uninstalled pygame-2.5.0
Successfully installed pygame-2.1.3

[notice] A new release of pip is available: 23.1 -> 23.1.2
[notice] To update, run: /Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/bin/python -m pip install --upgrade pip
carloss-Mac:~ carlos$ /Users/carlos/Library/Application\ Support/mu/mu_venv-38-20230627-155005/bin/python -m pgzero
pygame 2.1.3 (SDL 2.0.22, Python 3.8.13)
Hello from the pygame community. https://www.pygame.org/contribute.html
/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pgzero/runner.py:2: RuntimeWarning: use mixer: dlopen(/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/mixer.cpython-38-darwin.so, 2): Library not loaded: @loader_path/libomp.dylib
  Referenced from: /Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/.dylibs/libfluidsynth.3.1.1.dylib
  Reason: no suitable image found.  Did find:
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/.dylibs/libomp.dylib: cannot load 'libomp.dylib' (load command 0x80000034 is unknown)
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/.dylibs/libomp.dylib: cannot load 'libomp.dylib' (load command 0x80000034 is unknown)
(ImportError: dlopen(/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/mixer.cpython-38-darwin.so, 2): Library not loaded: @loader_path/libomp.dylib
  Referenced from: /Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/.dylibs/libfluidsynth.3.1.1.dylib
  Reason: no suitable image found.  Did find:
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/.dylibs/libomp.dylib: cannot load 'libomp.dylib' (load command 0x80000034 is unknown)
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/.dylibs/libomp.dylib: cannot load 'libomp.dylib' (load command 0x80000034 is unknown))
  pygame.mixer.pre_init(frequency=22050, size=-16, channels=2)
Traceback (most recent call last):
  File "/Users/carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Users/carlos/Downloads/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pgzero/__main__.py", line 1, in <module>
    from pgzero.runner import main
  File "/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pgzero/runner.py", line 2, in <module>
    pygame.mixer.pre_init(frequency=22050, size=-16, channels=2)
  File "/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/__init__.py", line 70, in __getattr__
    raise NotImplementedError(missing_msg)
NotImplementedError: mixer module not available (ImportError: dlopen(/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/mixer.cpython-38-darwin.so, 2): Library not loaded: @loader_path/libomp.dylib
  Referenced from: /Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/.dylibs/libfluidsynth.3.1.1.dylib
  Reason: no suitable image found.  Did find:
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/.dylibs/libomp.dylib: cannot load 'libomp.dylib' (load command 0x80000034 is unknown)
	/Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/lib/python3.8/site-packages/pygame/.dylibs/libomp.dylib: cannot load 'libomp.dylib' (load command 0x80000034 is unknown))
carloss-Mac:~ carlos$ /Users/carlos/Library/Application\ Support/mu/mu_venv-38-20230627-155005/bin/pip install "pygame==2.1.2"
Collecting pygame==2.1.2
  Downloading pygame-2.1.2-cp38-cp38-macosx_10_9_x86_64.whl (8.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.9/8.9 MB 7.0 MB/s eta 0:00:00
Installing collected packages: pygame
  Attempting uninstall: pygame
    Found existing installation: pygame 2.1.3
    Uninstalling pygame-2.1.3:
      Successfully uninstalled pygame-2.1.3
Successfully installed pygame-2.1.2

[notice] A new release of pip is available: 23.1 -> 23.1.2
[notice] To update, run: /Users/carlos/Library/Application Support/mu/mu_venv-38-20230627-155005/bin/python -m pip install --upgrade pip
carloss-Mac:~ carlos$ /Users/carlos/Library/Application\ Support/mu/mu_venv-38-20230627-155005/bin/python -m pgzero
pygame 2.1.2 (SDL 2.0.18, Python 3.8.13)
Hello from the pygame community. https://www.pygame.org/contribute.html
[... cut ...]
	0 : <CFString 0x7fffa27167d8 [0x7fffa26ab8e0]>{contents = "InputSourceKind"} = <CFString 0x7fffa275bd18 [0x7fffa26ab8e0]>{contents = "Keyboard Layout"}
	1 : <CFString 0x7fffa2747e58 [0x7fffa26ab8e0]>{contents = "KeyboardLayout ID"} = <CFNumber 0xe9b9b2ba1f934ee3 [0x7fffa26ab8e0]>{value = +2, type = kCFNumberSInt64Type}
	9 : <CFString 0x7fffa27114d8 [0x7fffa26ab8e0]>{contents = "KeyboardLayout Name"} = British
}
Usage: __main__.py [options]

__main__.py: error: You must specify which module to run.
```

carlosperate, 2023-06-27:
- https://github.com/pygame/pygame/issues/3941

illume, 2023-07-02:
Hey hey.

Is this whilst using vmware?

Also I wonder what the mac model is?
(wondering if it is CPU instruction related)

carlosperate, 2023-07-03:
Hi René! Yes, this is using virtualised Intel macOS with VMware using an i7 Macbook Pro 2020.
Unfortunately I don't have any mac with macOS lower than 12 Monterey, so tests in older macOS versions are done with virtual machines.

illume, 2023-09-09:
Hey hey.

pygame==2.5.1 fixes this issue.

At least it does work on the 10.11, 10.12, and 10.13 tested on.

Hopefully it works for you too?